esphome/issues #746: MHZ19B - Reading UART fails if log level is VERY_VERBOSE

borpin, 2019-10-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io / Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

WEMOS D1 Mini Rev 3.1.0
**Affected component:**

https://esphome.io/components/sensor/mhz19.html

**Description of problem:**
When the log level is set to VERY_VERBOSE the reading of the UART fails for both HW and SW UARTs.

**Problem-relevant YAML-configuration entries:**
```yaml
logger:
  log_level: VERY_VERBOSE
  hardware_uart: UART0_SWAP

# Example configuration entry
uart:
  rx_pin: RX
  tx_pin: TX
  baud_rate: 9600
  id: uart_co2 

sensor:
  - platform: mhz19
    co2:
      name: "MH-Z19 CO2 Value"
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 30s
    uart_id: uart_co2
```

**Logs (if applicable):**

Log set to VERBOSE Using HW UART
```
[20:08:26][I][app:101]: esphome version 1.14.0-dev compiled on Oct 18 2019, 20:06:42
[20:08:26][C][wifi:409]: WiFi:
[20:08:26][C][wifi:277]:   SSID: [redacted]
[20:08:26][C][wifi:278]:   IP Address: 192.168.7.60
[20:08:26][C][wifi:280]:   BSSID: [redacted]
[20:08:26][C][wifi:281]:   Hostname: 'co2_4'
[20:08:27][C][wifi:285]:   Signal strength: -48 dB ▂▄▆█
[20:08:27][V][wifi:287]:   Priority: 0.0
[20:08:27][C][wifi:289]:   Channel: 1
[20:08:27][C][wifi:290]:   Subnet: 255.255.255.0
[20:08:27][C][wifi:291]:   Gateway: 192.168.7.25
[20:08:27][C][wifi:292]:   DNS1: 192.168.7.29
[20:08:27][C][wifi:293]:   DNS2: (IP unset)
[20:08:27][C][uart:124]: UART Bus:
[20:08:27][C][uart:126]:   TX Pin: GPIO1
[20:08:27][C][uart:129]:   RX Pin: GPIO3
[20:08:27][C][uart:131]:   Baud Rate: 9600 baud
[20:08:27][C][uart:133]:   Using hardware serial interface.
[20:08:27][C][logger:175]: Logger:
[20:08:27][C][logger:176]:   Level: VERBOSE
[20:08:27][C][logger:177]:   Log Baud Rate: 115200
[20:08:27][C][logger:178]:   Hardware UART: UART0_SWAP
[20:08:27][C][mhz19:094]: MH-Z19:
[20:08:27][C][mhz19:095]:   CO2 'MH-Z19 CO2 Value'
[20:08:27][C][mhz19:095]:     Unit of Measurement: 'ppm'
[20:08:27][C][mhz19:095]:     Accuracy Decimals: 0
[20:08:28][C][mhz19:095]:     Icon: 'mdi:periodic-table-co2'
[20:08:28][C][mhz19:096]:   Temperature 'MH-Z19 Temperature'
[20:08:28][C][mhz19:096]:     Unit of Measurement: '°C'
[20:08:28][C][mhz19:096]:     Accuracy Decimals: 0
[... 132 lines cut ...]
[20:15:50][VV][api.service:035]: send_ping_response: PingResponse {}
[20:15:55][VV][api.service:250]: on_ping_request: PingRequest {}
[20:15:55][VV][api.service:035]: send_ping_response: PingResponse {}
[20:16:00][VV][api.service:250]: on_ping_request: PingRequest {}
[20:16:00][VV][api.service:035]: send_ping_response: PingResponse {}
[20:16:02][VV][scheduler:124]: Running interval 'update' with interval=30000 last_execution=32160 (now=62160)
[20:16:02][VV][uart:155]:     Wrote 0b11111111 (0xFF)
[20:16:02][VV][uart:155]:     Wrote 0b00000001 (0x01)
[20:16:02][VV][uart:155]:     Wrote 0b10000110 (0x86)
[20:16:02][VV][uart:155]:     Wrote 0b00000000 (0x00)
[20:16:02][VV][uart:155]:     Wrote 0b00000000 (0x00)
[20:16:02][VV][uart:155]:     Wrote 0b00000000 (0x00)
[20:16:02][VV][uart:155]:     Wrote 0b00000000 (0x00)
[20:16:02][VV][uart:155]:     Wrote 0b00000000 (0x00)
[20:16:02][VV][uart:145]:     Wrote 0b01111001 (0x79)
[20:16:02][VV][uart:226]:     Flushing...
[20:16:02][E][uart:211]: Reading from UART timed out at byte 0!
[20:16:02][W][mhz19:034]: Reading data from MHZ19 failed!
[20:16:03][V][app:076]: A component took a long time in a loop() cycle (0.8 s).
[20:16:03][V][app:077]: Components should block for at most 20-30ms in loop().
[20:16:03][V][app:078]: This will become a warning soon.

```

**Additional information and things you've tried:**

OttoWinter, 2019-10-20:
Yeah that's because the MHZ19 has some pretty strict timing requirements. Sending the log messages is taking up a bunch of time and making the MHZ19 timeout. There's not really a way to fix it either unfortunately.

closed 2019-10-20 by OttoWinter

borpin, 2019-10-20:
@OttoWinter Can this be added to the docs please?