esphome/issues #749: Hass.io ingress wizard 404

OttoWinter, 2019-10-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hass.io

**Affected component:**

After the user completes setup wizard in the dashboard view, the dashboard page returns 404s for everything.

Reason: The wizard creates a POST request to wizard.html, when that completes successfully it returns a 301 to ./?begin=true, which the browser follows correctly.

However, for some reason the browser appears to loose the relative URL in the process and any following relative URL loads fail for some reason.

Requests before error:
http://hassio.local:8123/api/hassio_ingress/tnlvR6i-N4GeIZaPFRTUJPyT6_TAZMW27N50opVLWYI/ping

Requests after completing wizard (stylesheet and JS load fine though!):
http://hassio.local:8123/ping

Reloading the page fixes the problem, but it's still very bad for onboarding.

jonathanadams, 2020-04-29:
It is likely this has been fixed by esphome/esphome#1025

glmnet, 2020-04-30:
Thanks @jonathanadams

closed 2020-04-30 by glmnet