esphome/issues #752: Unable to compile with API enabled

syphernl, 2019-10-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Pip installed version of dev.zip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (Blitzwolf BW-SHP6)

**Affected component:**

https://esphome.io/components/api

**Description of problem:**
As soon as I enable `api:` in my ESPHome config file, it fails. This happens both on dev and on 1.13.6, although the failure is a bit different.

This is affecting new devices but also existing one's.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  plug_name: blitz02
  # Higher value gives lower watt readout
  current_res: "0.00290"
  # Lower value gives lower voltage readout
  voltage_div: "940"

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp8285

wifi:
  ssid: 'XXXXXXXXXXXXXXXXXX'
  password: 'XXXXXXXXXXXXXXXXX'
  use_address: 'XXXXXXXXXXXXXXX'

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: 'XXXXXXXXXXXXXXXXXXXXXX'
    
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO13
      inverted: True
    name: "${plug_name}_button"
    on_press:
      - switch.toggle: switch1

output:
  - platform: gpio
    pin: GPIO15
    id: relay
  - platform: gpio
    pin: GPIO00
    inverted: True
    id: led1

status_led:
  pin:
    number: GPIO02
    inverted: True

switch:
- platform: template
[... 475 lines cut ...]
src/esphome/components/api/basic_messages.cpp: In member function 'virtual bool esphome::api::HelloRequest::decode_length_delimited(uint32_t, const uint8_t*, size_t)':
src/esphome/components/api/basic_messages.cpp:11:48: error: 'as_string' was not declared in this scope
       this->client_info_ = as_string(value, len);
                                                ^
src/esphome/components/api/basic_messages.cpp: In member function 'virtual bool esphome::api::ConnectRequest::decode_length_delimited(uint32_t, const uint8_t*, size_t)':
src/esphome/components/api/basic_messages.cpp:25:45: error: 'as_string' was not declared in this scope
       this->password_ = as_string(value, len);
                                             ^
src/esphome/components/api/basic_messages.cpp: In member function 'virtual bool esphome::api::DisconnectRequest::decode_length_delimited(uint32_t, const uint8_t*, size_t)':
src/esphome/components/api/basic_messages.cpp:40:43: error: 'as_string' was not declared in this scope
       this->reason_ = as_string(value, len);
                                           ^
src/esphome/components/api/basic_messages.cpp: At global scope:
src/esphome/components/api/basic_messages.cpp:48:32: error: variable or field 'encode' declared void
 void DisconnectRequest::encode(APIBuffer &buffer) {
                                ^
*** [.pioenvs/blitz02/src/esphome/components/api/command_messages.cpp.o] Error 1
src/esphome/components/api/basic_messages.cpp:48:32: error: 'APIBuffer' was not declared in this scope
src/esphome/components/api/basic_messages.cpp:48:43: error: 'buffer' was not declared in this scope
 void DisconnectRequest::encode(APIBuffer &buffer) {
                                           ^
*** [.pioenvs/blitz02/src/esphome/components/api/basic_messages.cpp.o] Error 1
```

**Additional information and things you've tried:**

OttoWinter, 2019-10-20:
Please look for the folder `blitz02` in your config directory. Delete it and recompile. Something appears to have messed up the compile file system (potentially switching between versions in a weird way?)

Alex9779, 2019-10-21:
I had a similar problem a few days ago, cleaning the compile and caching directories didn't help, I had to remove the pip installation completely with `pip2 freeze | xargs pip2 uninstall -y` and then reinstalled the desired  version...

OttoWinter, 2019-10-21:
Hmm, it might be pip not removing old files from the install directory under certain circumstances.

Could you post the commands you installed esphome with and how you perfomed the upgrade? I can't reproduce it here.

Also, please post the output of the following commands:

```
$ python2 -c "from esphome.components import api;import os;print(list(os.listdir(os.path.dirname(api.__file__))))"
$ pip2 --version
```

(replace 2 with 3 if you used python 3)

syphernl, 2019-10-22:
Looks like I experienced the same issue @Alex9779 mentioned. After removing all versions of esphome and installing it again, compiling worked fine.

OttoWinter, 2019-10-22:
Reopening this issue, I want to know what the reason is so that the 1.14 rollout will be as smooth as possible.

Alex9779, 2019-10-23:
I just tried something and I broke it again.
I was on dev with `pip2 install https://github.com/esphome/esphome/archive/dev.zip`.
The I did `pip2 install esphome` which seems to break it, going back to dev with above command still leaves it broken but with different compilation errors...

Alex9779, 2019-10-23:
seems going back is the problem, did everything now with pip3 since you added a deprecation warning for python2 but same results, going back from dev to normal breaks it, cleaning builds does not help when going back to dev, have to remove everything then reinstall the desired version, clean the build and then it compiles fine...

OttoWinter, 2019-10-24:
Ok, looked into this. It appears to be an issue in setuptools. Basically, when installing via the URL it will be saved in a different way internally (egg designator if I understand correctly).

Since it only affects dev-to-pypi updates (and presumable vice versa), I think this is low priority. And I can't think of any workarounds to this issue that we can do other than changing the code structure a lot (which won't happen for a low priority issue).

So: For anyone experiencing this, uninstall esphome (`pip2/3 uninstall esphome`) and reinstall it again.

closed 2019-10-24 by OttoWinter