esphome/issues #753: Cover is not safe - enabling switches need to be decoupled

swifty99, 2019-10-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

not relevant
**ESP (ESP32/ESP8266, Board/Sonoff):**

not relevant
**Affected component:**

cover time_based

**Description of problem:**
cover activation switches are not blocked at each other. If a moving shade is operating and the ui requests opposite both switches are operated the same time. Bad things could happen.

a temporary solution is to make the switches turn off each other:
`switch:
  - platform: gpio
    id: "kueche_rollo_relais_open"
    on_turn_on:
    - switch.turn_off: kueche_rollo_relais_close
    pin:
      pcf8574: pcf8574_hub
      number: 2
      mode: OUTPUT
      inverted: True
      
  - platform: gpio
    id: "kueche_rollo_relais_close"
    on_turn_on:
    - switch.turn_off: kueche_rollo_relais_open
    pin:
      pcf8574: pcf8574_hub
      number: 3
      mode: OUTPUT
      inverted: True

      
cover:
  - platform: time_based
    name: "kueche tuer rollo"

    open_action:
      - switch.turn_on: kueche_rollo_relais_open
    open_duration: 2.1min

    close_action:
      - switch.turn_on: kueche_rollo_relais_close
    close_duration: 2min

    stop_action:
      - switch.turn_off: kueche_rollo_relais_open
      - switch.turn_off: kueche_rollo_relais_close  `

however this still has some drawbacks. Most motors have a high induction currents. This implementation switches direction immediatly. Resulting currents may destroy relais or motor.

Possible solution:
- the cover component should look for active switches if opposite direction is requested
- if true turn them off and wait some 100ms
- reverse the motor.

I could help to improve, wrote some code for similar integrations. However I don't even know how to accces the cover.cpp on my home assistant machine :-(

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-10-20:
Please see `interlock` for gpio switches.

But I've also had problems with having to wait before activating the other switch in the interlock group - so I created a small patch that would add that feature.

swifty99, 2019-10-20:
true, interlocking is also working.

However, I think this feature should not be optional for covers. At least in the cover ducumentation the interlocking should be highly recommended.

Thx for writing this aweful great SW

OttoWinter, 2019-10-20:
It is there? https://esphome.io/cookbook/endstop-cover.html

swifty99, 2019-10-20:
yes,
however this did not apply to me, so I was only looking here:
[https://esphome.io/components/cover/time_based.html](url)

closed 2019-10-20 by OttoWinter