esphome/issues #754: Weird Result from if/else Action

Rafalow, 2019-10-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io Add-on
**ESP (ESP32/ESP8266, Board/Sonoff):**
Adafruit Huzzah ESP8266

**Affected component:**
 https://esphome.io/guides/automations.html#if-action

**Description of problem:**
Fan (or blower) switch is turning off despite if/else code that should stop this. Eventually it does work (see debug below), but I haven't been able to determine a pattern on how it will behave. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp_thermostat
  platform: ESP8266
  board: huzzah
switch:
  - platform: gpio
    pin: 12
    id: blower
    internal: true
  - platform: gpio
    pin: 14
    name: "Heat"
    id: heat
    interlock: cool
    on_turn_on:
     then:
      - switch.turn_on: blower
  - platform: gpio
    pin: 13
    name: "Cool"
    id: cool
    interlock: heat
    on_turn_on:
     then:
      - switch.turn_on: blower
sensor:
  - platform: dht
    pin: 2
    temperature:
      name: "Temperature"
      id: temp
      on_value:
       then:
        - if:
           condition:
            and:
             - switch.is_off: cool
             - switch.is_off: heat
           then:
            - delay: 60s
            - switch.turn_off: blower
           else:
            - switch.turn_on: blower
    humidity:
      name: "Humidity"
    model: DHT11
    update_interval: 15s
```

**Logs (if applicable):**

```
[11:46:52][D][climate:010]: 'Thermostat Test' - Setting
[11:46:52][D][climate:020]:   Target Temperature Low: 22.78
[11:46:52][D][climate:023]:   Target Temperature High: 25.89
[11:46:52][D][switch:021]: 'Heat' Turning ON.
[11:46:52][D][switch:045]: 'Heat': Sending state ON
[11:46:52][D][switch:021]: 'blower' Turning ON.
[11:46:52][D][switch:045]: 'blower': Sending state ON
[11:46:52][D][climate:172]: 'Thermostat Test' - Sending state:
[11:46:52][D][climate:175]:   Mode: AUTO
[11:46:52][D][climate:177]:   Current Temperature: 22.00°C
[11:46:52][D][climate:181]:   Target Temperature: Low: 22.78°C High: 25.89°C
[11:46:52][D][climate:172]: 'Thermostat Test' - Sending state:
[11:46:52][D][climate:175]:   Mode: AUTO
[11:46:52][D][climate:177]:   Current Temperature: 22.00°C
[11:46:52][D][climate:181]:   Target Temperature: Low: 22.78°C High: 25.89°C
[11:47:00][D][dht:048]: Got Temperature=22.0°C Humidity=71.0%
[11:47:00][D][sensor:092]: 'Temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[11:47:00][D][switch:021]: 'blower' Turning ON.
[11:47:00][D][climate:172]: 'Thermostat Test' - Sending state:
[11:47:00][D][climate:175]:   Mode: AUTO
[11:47:00][D][climate:177]:   Current Temperature: 22.00°C
[11:47:00][D][climate:181]:   Target Temperature: Low: 22.78°C High: 25.89°C
[11:47:00][D][sensor:092]: 'Humidity': Sending state 71.00000 % with 0 decimals of accuracy
[11:47:00][D][switch:025]: 'blower' Turning OFF.
[11:47:00][D][switch:045]: 'blower': Sending state OFF
[11:47:16][D][dht:048]: Got Temperature=22.0°C Humidity=71.0%
[11:47:16][D][sensor:092]: 'Temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[11:47:16][D][switch:021]: 'blower' Turning ON.
[11:47:16][D][switch:045]: 'blower': Sending state ON
[11:47:16][D][climate:172]: 'Thermostat Test' - Sending state:
[11:47:16][D][climate:175]:   Mode: AUTO
[11:47:16][D][climate:177]:   Current Temperature: 22.00°C
[11:47:16][D][climate:181]:   Target Temperature: Low: 22.78°C High: 25.89°C
[11:47:16][D][sensor:092]: 'Humidity': Sending state 71.00000 % with 0 decimals of accuracy
[11:47:16][D][switch:025]: 'blower' Turning OFF.
[11:47:16][D][switch:045]: 'blower': Sending state OFF
[11:47:30][D][dht:048]: Got Temperature=22.0°C Humidity=71.0%
[11:47:30][D][sensor:092]: 'Temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[11:47:30][D][switch:021]: 'blower' Turning ON.
[11:47:30][D][switch:045]: 'blower': Sending state ON
[11:47:30][D][climate:172]: 'Thermostat Test' - Sending state:
[11:47:30][D][climate:175]:   Mode: AUTO
[11:47:30][D][climate:177]:   Current Temperature: 22.00°C
[11:47:30][D][climate:181]:   Target Temperature: Low: 22.78°C High: 25.89°C
[11:47:31][D][sensor:092]: 'Humidity': Sending state 71.00000 % with 0 decimals of accuracy
[11:47:31][D][switch:025]: 'blower' Turning OFF.
[11:47:31][D][switch:045]: 'blower': Sending state OFF
[11:47:45][D][dht:048]: Got Temperature=22.0°C Humidity=71.0%
[11:47:45][D][sensor:092]: 'Temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[11:47:45][D][switch:021]: 'blower' Turning ON.
[11:47:45][D][switch:045]: 'blower': Sending state ON
[11:47:45][D][climate:172]: 'Thermostat Test' - Sending state:
[11:47:45][D][climate:175]:   Mode: AUTO
[11:47:45][D][climate:177]:   Current Temperature: 22.00°C
[11:47:45][D][climate:181]:   Target Temperature: Low: 22.78°C High: 25.89°C
[11:47:45][D][sensor:092]: 'Humidity': Sending state 71.00000 % with 0 decimals of accuracy
[11:47:45][D][switch:025]: 'blower' Turning OFF.
[11:47:45][D][switch:045]: 'blower': Sending state OFF
[11:48:00][D][dht:048]: Got Temperature=22.0°C Humidity=71.0%
[11:48:00][D][sensor:092]: 'Temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[11:48:00][D][switch:021]: 'blower' Turning ON.
[11:48:00][D][switch:045]: 'blower': Sending state ON
[11:48:00][D][climate:172]: 'Thermostat Test' - Sending state:
[11:48:00][D][climate:175]:   Mode: AUTO
[11:48:00][D][climate:177]:   Current Temperature: 22.00°C
[11:48:00][D][climate:181]:   Target Temperature: Low: 22.78°C High: 25.89°C
[11:48:00][D][sensor:092]: 'Humidity': Sending state 71.00000 % with 0 decimals of accuracy
[11:48:16][D][dht:048]: Got Temperature=22.0°C Humidity=71.0%
[11:48:16][D][sensor:092]: 'Temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[11:48:16][D][switch:021]: 'blower' Turning ON.
[11:48:16][D][climate:172]: 'Thermostat Test' - Sending state:
[11:48:16][D][climate:175]:   Mode: AUTO
[11:48:16][D][climate:177]:   Current Temperature: 22.00°C
[11:48:16][D][climate:181]:   Target Temperature: Low: 22.78°C High: 25.89°C

```

**Additional information and things you've tried:**
The fan work perfectly on cool without the need for the "else" part in the action. However, heat has been acting weird with the fan. I added the else statement first, and then tried changing the switch id to blower (on the off chance something in esphome was reading the fan part differently then I intended). 

I am new to HA and esphome...so it's 99% likely to be my fault, but I'm not sure where I messed up. The yaml seems straight forward with easy to follow documentation. Thanks so much for a great tool!

glmnet, 2019-10-20:
I believe your issue might be that there are several “instances” of the script running. So the condition was met 60 seconds ago, now it is not. But it was met so now it should turn fan off. 

Try using for condition, or re checking the condition after the 60s delay. 

https://next.esphome.io/guides/automations.html#for-condition

Rafalow, 2019-10-20:
Thank you glmnet.  I agree there are many instances running that may preclude the fan from coming on correctly, and the code can be optimized, but I believe this is separate from the root issue.  When the heat is turned on by itself on the first action after power up, the fan is not turning on correctly.  In that case, there are no 60 second timers running.

Thanks again!

Rafalow, 2019-10-20:
Nevermind...just realized you are correct. I had gone from idle to heat. I moved the trigger to the heat and cool switches and that worked! Thanks again!

Still a bit strange that is worked for cool. Anyway, yaml code below for anyone doing something similar...

switch:
  - platform: gpio
    pin: 12
    id: blower
    internal: true
  - platform: gpio
    pin: 14
    name: "Heat"
    id: heat
    interlock: cool
    on_turn_off:
      then:
        - delay: 60s
        - if:
           condition:
            and:
             - switch.is_off: cool
             - switch.is_off: heat
           then:
            - switch.turn_off: blower
    on_turn_on:
     then:
      - switch.turn_on: blower
  - platform: gpio
    pin: 13
    name: "Cool"
    id: cool
    interlock: heat
    on_turn_off:
      then:
        - delay: 60s
        - if:
           condition:
            and:
             - switch.is_off: cool
             - switch.is_off: heat
           then:
            - switch.turn_off: blower
    on_turn_on:
     then:
      - switch.turn_on: blower

closed 2019-10-20 by Rafalow