esphome/issues #755: PCF8574 initilization toggles output at boot (if inversed logic)

swifty99, 2019-10-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

not relevant
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemes D1, probably not relevant
**Affected component:**

PCF8574Component

**Description of problem:**
As the PCF8574 has limited capabilities relays may only be used which are low active. At least in other setups this successfully prevents unexpected activation in uncertain conditions.  If low active relays are used logic needs to be inverted. With esphome all configured relays become active for about 100ms after boot.

For me it looks like the PCF8574 setup function may write initialization values which are not yet inverted.

I will try some workarounds when I get the development environment running. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
i2c:
  sda: D2
  scl: D1
  scan: True
  id: bus_a
  
  
pcf8574:
  - id: 'pcf8574_hub'
    address: 0x38

# Individual outputs
switch:
  - platform: gpio
    name: "relais_standby"
    pin:
      pcf8574: pcf8574_hub
      number: 2
      mode: OUTPUT
      inverted: True
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

At least I know it can be avoided, I wrote a driver for this thing for other environments. Might help out.

swifty99, 2019-10-21:
commit looks good, will try to test it.

S-Przybylski, 2019-10-27:
Dear @OttoWinter 
dear @swifty99 
today i checked the current docker beta 1.14.0b4 and found out, that the PCF8574 input mode doesn't work anymore if a input is set to gnd either in input mode (with do not work before) nor in input_pullup mode (which worked). Inputs which are not set to gnd (pulled up to vcc) while booting are working.

See related #667 

So i assume that this change affects the input mode.

S-Przybylski, 2019-10-27:
Dear @OttoWinter 
dear @swifty99 
I have investigated the problem in more in detail and saw that the dedicated PR isn't responsible for that! (it's not included in the latest docker beta - sorry). 
I found another change which is responsible for the input issue: It's #715 (PR #744) 

I reverted that commit locally and tested it out including the change which was done by this issue #750 (Don't update PCF8574 on boot). 

If i set the mode to INPUT_PULLUP, the input works as designed. 
I can't test the output mode, because my current pcf8574 modules are fixed to input mode only.

Could you please revert the PR#744 ?

S-Przybylski, 2019-11-01:
Dear @OttoWinter 
dear @swifty99 
dear @amishv
last week i bought two additional pcf8574 output boards to investigate a little bit more in detail:
Using the 1.14.0b5 with two pcf8574 input boards and two pcf8574 output boards i saw that the input doesn't work either for input_pullup nor input. If a pin is wired to gnd while the esp starts, the pin do not reflect any update.

I also noticed if i use an older pcf8574.cpp version, that the outputs flashed a short time while starting. I made a small change that helps to avoid the flashing and initialies the pcf8574 in case of a gnd signal of a pin correctly:

1. I adapt as a workaround the calcualation of the port mask (this is not sufficient!) &-->| [input_pullup does not exists]
2. I write the configuration only in case of input/input_pullup mode (that solves the issue of avoiding the scattering and the correct initalisation)   -->    this->write_gpio_();
@OttoWinter please put this little change into the pcf8574.cpp?

````
void PCF8574Component::pin_mode(uint8_t pin, uint8_t mode) {
  switch (mode) {
    case PCF8574_INPUT:
      this->ddr_mask_ &= ~(1 << pin);
      this->port_mask_ |= ~(1 << pin);
      this->write_gpio_();
      break;
    case PCF8574_INPUT_PULLUP:
      this->ddr_mask_ &= ~(1 << pin);
      this->port_mask_ |= (1 << pin);
      this->write_gpio_();
      break;
    case PCF8574_OUTPUT:
      this->ddr_mask_ |= (1 << pin);
      this->port_mask_ &= ~(1 << pin);
      break;
    default:
      break;
  }
````

S-Przybylski, 2019-11-02:
Today i tested to new release: 1.14.0:
I doesn't work in input/input_pullup mode while a pin is set to gnd!

closed 2019-11-02 by OttoWinter