esphome/issues #757: secrets should not be expanded in files where

MichaelRegan, 2019-10-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on a Raspberry Pi 3
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 nodeMCU
**Affected component:**

All files that allow customization

**Description of problem:**
While looking through the generated directories and files (for backing up relevant files to Github), I noticed passwords and wifi information that is stored in my secrets.yaml file is expanded in clear text in a number of locations. This is a concern because there is the ability to extend these files with custom code. If these were updated with custom code it would be a challenge to keep them backed up and managed.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-10-21:
Secrets *must* be expanded somewhere because the C++ compiler needs to know their values - there's no way to not print them somewhere in the generated code.

That's why esphome automatically generates a .gitignore file in the configuration directory that automatically excludes all auto-generated files and secrets.yaml.

> If these were updated with custom code it would be a challenge to keep them backed up and managed.

You're not really supposed to modify main.cpp anymore (you can, but it's discouraged). Please see the custom sensor guide - there's a mechanism of including custom files in the top-level `esphome:` configuration.

closed 2019-10-21 by OttoWinter

MichaelRegan, 2019-10-21:
Yeah, I get it. Perhaps this could be addressed in the documentation or the secrets could be included through a header file. I'm sorry if it is already in the docs and I missed it.

OttoWinter, 2019-10-21:
> Perhaps this could be addressed in the documentation

The documentation has removed (almost) all references to editing main.cpp - I went to see again and there was one left and I will remove it.

> the secrets could be included through a header file

the problem is that while the data passes through the validation process, the information is lost as to what data is secret and what not because of some internal reasons. The system can not know what is secret and what not after the validation process.