esphome/issues #758: ESPhome RF issue works on all but a group of 3 RF Sockets

stredwizard, 2019-10-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker on Ubuntu Server 18.04

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266-12F

**Affected component:**

https://esphome.io/components/remote_transmitter.html#remote-transmitter-rc-switch-protocol

**Description of problem:**

I have various RF devices around the house which all used to be controlled by OMG, since switching the node over to ESPhome, there are three RF mains sockets that won't play ball with ESPhome, in the example yaml below the first template switch is a RF wall mounted light switch which works perfectly, the second is one of the mains sockets, I have also attached the code I was using with OMG within homeassistant, the decimal codes were converted to binary and aside from the decimal RF value being 1 decimal shorter I cannot see why they would work with OMG but not esphome?

I could of course switch back to OMG, but I like the web interface for ESPhome to control everything, check online statuses and make simple changes without removing the node from its install location and to finally do away with MQTT.

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: template
    name: Boy's
    id: switch_boys
    optimistic: true
    assumed_state: true
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '101010101010101010101110'
        protocol:
          pulse_length: 170
        repeat: 10
    turn_off_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '101010101010101011110110'
        protocol:
          pulse_length: 170
        repeat: 10
  - platform: template
    name: RF Plug 2
    id: switch_rfplug2
    optimistic: true
    assumed_state: true
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '101010101010111000011'
        protocol:
          pulse_length: 170
        repeat: 10
    turn_off_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '101010101010111001100'
        protocol:
          pulse_length: 170
        repeat: 10
```

**Logs (if applicable):**

```

```

**Additional information and things you've tried:**

```yaml
#light RF Plug 2:
- platform: mqtt
  name: "Lounge Floor Lamp"
  command_topic: "home/commands/MQTTto433/PLSL_170 -m"
  state_topic: "home/433toMQTT"
  qos: 1
  payload_on: 1398211
  payload_off: 1398220
  optimistic: false
  retain: true
#light Boys room:
- platform: mqtt
  name: "Boys Light"
  command_topic: "home/commands/MQTTto433/PLSL_170 -m"
  state_topic: "home/433toMQTT"
  qos: 1
  payload_on: 1398210
  payload_off: 1398201
  optimistic: false
  retain: true
```

OttoWinter, 2019-10-21:
The reason is probably that you have an error in the conversion.

For the first one the pulse is `101010101010101011110110` (24 bits), but for the second one you have `101010101010111000011` (21 bits).

Please try `000101010101010111000011`

stredwizard, 2019-10-21:
thankyou so much for your quick reply and i'm pleased to say solution, tested on one of the plugs and working great.

closed 2019-10-21 by stredwizard