esphome/issues #761: ESP32 boot does not boot with latest dev branch

Lumpusz, 2019-10-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

git clone of dev branch
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

compiled binary

**Description of problem:**
After uploading the binary with ESPHome-Flasher-1.1.0-windows-x64.exe, board does not boot.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_sensor
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "xx"
  password: "yy"

esp32_ble_tracker:
  scan_interval: 10s

sensor:
  - platform: ble_rssi
    mac_address: AC:37:43:77:5F:4C
    name: "Test MAC"

api:

ota:

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310

```

**Additional information and things you've tried:**

mkdir tmp
cd tmp
git clone https://github.com/esphome/esphome
cp ../test_sensor.yaml .
pip install ./esphome
python -m esphome test_sensor.yaml compile
upload using serial flasher

I tried removing the BT parts, and just have wifi, but I got the same error

OttoWinter, 2019-10-21:
Please post what board you're using exactly - or is it just a bare wrover module (but then also post which version of that). And just to clarify - it did work before, right?

I wonder if it might be the bootloader shipped with esphomeflasher, I thought I read that esp-idf's bootloaders will always be backwards-compatible though.

Lumpusz, 2019-10-21:
ESP32 devkitv1 with a ESP WROOM 32

It still works if I upload a binary built with the latest stable release through serial, but the one I build on top of latest dev, fails.

WouterJN, 2019-10-22:
I have exactly the same board and it also doesn't boot. Which board type did you select and how did you flash the ESP32?

Lumpusz, 2019-10-22:
@WouterJN I built the firmware with the esphome cli, and flashed it with esphome flasher.

OttoWinter, 2019-10-22:
Can someone confirm this is working when flashing directly from esphome?

I'm currently battling some issues with my computer - an OS update forced the siliconlabs driver to be updated but the latest version has a bug, so I can't test any uploading from esphome-flasher right now.

Lumpusz, 2019-10-22:
No, it does not work. I uploaded a binary with the flasher through serial, that was built with 1.13.6, so the board came back online, then I ran below command and did an OTA upload:
```
python -m esphome test_sensor.yaml run --upload-port xxx.xxx.xxx.xxx
```
...and the board does not boot.

OttoWinter, 2019-10-22:
Ok, that's good to know - so it's not ESPHome-Flasher (which was my first thought).

That means it's either 1) the esp-idf bootloader or 2) something in ESPHome (though I don't think so because that would likely show at least *some* log output like "Logger initialized" before crashing).

1) can be verified by uploading through ESPHome via USB (so like what you did @Lumpusz just not via OTA but over USB).

Lumpusz, 2019-10-22:
```
$ python -m esphome test_sensor.yaml run --upload-port /dev/ttyUSB0
INFO Reading configuration test_sensor.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d test_sensor
Processing test_sensor (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.11.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <Update> 1.0
Retrieving maximum program size .pioenvs/test_sensor/firmware.elf
Checking size .pioenvs/test_sensor/firmware.elf
DATA:    [=         ]  12.7% (used 41744 bytes from 327680 bytes)
PROGRAM: [=====     ]  49.6% (used 813150 bytes from 1638400 bytes)
================================================================================ [SUCCESS] Took 8.41 seconds ================================================================================
INFO Successfully compiled program.
INFO Running:  platformio run -d test_sensor -t upload --upload-port /dev/ttyUSB0
Processing test_sensor (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.11.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size .pioenvs/test_sensor/firmware.elf
Checking size .pioenvs/test_sensor/firmware.elf
DATA:    [=         ]  12.7% (used 41744 bytes from 327680 bytes)
PROGRAM: [=====     ]  49.6% (used 813150 bytes from 1638400 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, ftdi, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading .pioenvs/test_sensor/firmware.bin
Serial port /dev/ttyUSB0
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15872 bytes to 10319...
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.2 seconds (effective 535.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 4901.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 21939.3 kbit/s)...
Hash of data verified.
Compressed 813264 bytes to 453725...
Wrote 813264 bytes (453725 compressed) at 0x00010000 in 11.1 seconds (effective 583.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=============================================================================== [SUCCESS] Took 23.84 seconds ===============================================================================
INFO Successfully uploaded program.
ERROR Logger is not configured!

OttoWinter, 2019-10-22:
@Lumpusz The log output I see looks fine. It also appears like you have no `logger:` entry in your configuration. Did you delete that on purpose? If so, what for and try with reading it back. I will try without logger once I get this setup to work again to see if that's the culprit.

WouterJN, 2019-10-22:
Before this I was using esptool.py for the flashing and the ESP32 never booted (I tried many different settings). Just now I tried to flash directly from esphome and that worked!

Lumpusz, 2019-10-22:
@OttoWinter I also noticed the missing logger, I put it back and it came back online. So it seems logger is mandatory for the time being :). BTW is it supposed to be mandatory?

OttoWinter, 2019-10-22:
@Lumpusz Well it's not supposed to be required, but I don't actually test that because if there's an issue it's pretty hard to diagnose (which I will have to do now :( ).

Looking at the git history there have also not been any changes to logger that could have caused this. Maybe it's just related to an esp-idf update. Looking at your initial log in the original issue, that now also doesn't mean anything - if at all it shows that the system is *not* crashing (then those messages would be repeated over and over).

Lumpusz, 2019-10-22:
@OttoWinter yes, debugging without logger seems like a nightmare. Let me know if I can help with testing, etc.

OttoWinter, 2019-10-24:
ESPHome-Flasher has been updated to 1.2.0 which should resolve the issue.

I also verified that it works, tried a config without logger and it connected successfully.

closed 2019-10-24 by OttoWinter

thibmaek, 2019-10-24:
Is this also fixed in the esphome python CLI tool? Was experiencing the same issue there