esphome/issues #764: ESPHome with Sonoff Basic problem

onlize, 2019-10-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 Sonoff Basic

**Affected component:**
ESP8266 Sonoff Basic

https://esphome.io/devices/sonoff_basic.html

**Description of problem:**
I had Sonoff Basic running Tasmota, and I decided to update it to ESPHome.  I configured it like it is recommended here: https://esphome.io/devices/sonoff_basic.html

I configured ESPHome firmware and uploaded to the device. First, I tried to upload it using Tasmota Web Interface, but my device did not connect and did not pick up IP address.

I double checked and my WiFi information was correct, so I decided to use FlashESP8266.exe. I used it multiple times before and I uploaded ESPHome file successfully, but I still have the same problem.

If I flush Tasmota firmware back, everything works, so it is something with the bin file that was created by ESPHome web portal. The validation says that file is good. How can I troubleshoot and fix it? 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sonoff_basic_1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "my_ssid"
  password: "my_password"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "AlarmDecoder Sonoff Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "AlarmDecoder Sonoff Relay"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
Reading other posts there was something about logger, but I do not see how I can read logs if I cannot connect to this device.
```

**Additional information and things you've tried:**

OttoWinter, 2019-10-22:
Connect the device to Hass.io as you did when you flashed the device. While having selected the device in the dropdown menu, click logs to view the serial logs from the device. They will help pinpoint the issue.

onlize, 2019-10-22:
Sorry, but I do not understand. The device was never able to connect to Hass.io. It could not connect to my WiFi, so it does not have IP address. Hass.io does not see it.

I created file using Hass.io. After that, I uploaded file using FlashESP8266.exe. I also tried to do the same using NodeMCU-PyFlasher-4.0-x64.exe and ESPHome-Flasher-1.1.0-windows-x64.exe with exactly the same results. It says that file was uploaded successfully, but it cannot connect to get an IP address and Hass.io does not see it. I double checked and SSID information is correct in the .bin file that was created.

OttoWinter, 2019-10-22:
I don't mean Home Assistant here - yes the device has no internet connection.

But you can still get logs from it by opening the serial port. See the "View Logs" button in ESPHome-Flasher. That will only get the raw logs though. If it fails with a crashdump, then you will need to decode the crash log with the ESPHome addon, which you would do by connecting the device to the HAss.io machine via USB and viewing the logs through the dashboard VIEW LOGS button while it's connected via USB.

onlize, 2019-10-22:
Thank you for explaining. I will try it, but can you please tell me how exactly you want me to do this. Should I upload the firmware, reboot the device and, after that, click on the "View Logs' button? Should I configure logs to DEBUG in the .bin file?

onlize, 2019-10-22:
I uploaded firmware and started logs. I posted the output. It is interesting. My SSID is activewt, it says that 'AP Not Found', but it finds this SSID later. It repeats the same message again and again. Any ideas?

```
[15:30:37][I][wifi:164]: WiFi Connecting to 'activewt'...
[15:30:43][W][wifi_esp8266:354]: Event: Disconnected ssid='activewt' bssid=00:00:00:00:00:00 reason='AP Not Found'
[15:30:43][W][wifi:400]: Error while connecting to network.
[15:30:43][W][wifi:431]: Restarting WiFi adapter...
[15:30:43][I][wifi:164]: WiFi Connecting to 'activewt'...
[15:30:49][W][wifi_esp8266:354]: Event: Disconnected ssid='activewt' bssid=00:00:00:00:00:00 reason='AP Not Found'
[15:30:49][W][wifi:400]: Error while connecting to network.
[15:30:49][W][wifi:431]: Restarting WiFi adapter...
[15:30:54][D][wifi:272]: Starting scan...
[15:31:01][D][wifi:287]: Found networks:
[15:31:01][I][wifi:322]: - 'activewt' (70:3A:CB:26:65:FD) ▂▄▆█
[15:31:01][D][wifi:323]:     Channel: 6
[15:31:01][D][wifi:324]:     RSSI: -38 dB
[15:31:01][I][wifi:322]: - 'activewt' (70:3A:CB:26:8B:B7) ▂▄▆█
[15:31:01][D][wifi:323]:     Channel: 11
[15:31:01][D][wifi:324]:     RSSI: -61 dB
```

OttoWinter, 2019-10-22:
Ok, then this is likely an instance of https://github.com/esphome/issues/issues/455

onlize, 2019-10-22:
It does not look like there is a solution. #455 has multiple different issues, but how to fix it? Is there anything I can do?

onlize, 2019-10-22:
Interesting.... Someone recommended to set logging to Verbose. I changed it and my device was able to connect. Thank you for your help.

OttoWinter, 2019-10-23:
Ok, that makes it an instance of #455, I will merge this issue into that discussion.

closed 2019-10-23 by OttoWinter