esphome/issues #765: Show logs in :dev no longer showing any logs

digiblur, 2019-10-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker of :dev and :latest

**ESP (ESP32/ESP8266, Board/Sonoff):**

Any device

**Affected component:**

Show Logs button in the GUI

**Description of problem:**

As of recently in the past week or two of upgrades the dev docker stopped showing any logs.  I can put the same device YAML in both containers config folders and only the prod one shows any logs for the device.  Everything is still working but it doesn't show any logs.  I can see the logs coming across in MQTT explorer without any issue on the device/debug topic.  

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

digiblur, 2019-10-23:
And a snip from the :latest docker show logs of the same device 

[image]

OttoWinter, 2019-10-23:
Huh, interesting. Nothing has changed in the log MQTT client in 8 months: https://github.com/esphome/esphome/blob/dev/esphome/mqtt.py

Only thing I can think of is the recent upgrade of python from 2 to 3 in the docker container. But the time frame doesn't really match up there. The switch was committed 3 days ago (https://github.com/esphome/esphome/commit/823ae7d1aaa170130827706f16c22776dd287e9b#diff-ebacf6f6ae4ee68078bb16454b23247d). Could the error have appeared ~3 days ago?

digiblur, 2019-10-23:
Let me turn back the tags on the container and see exactly when it broke. Stay tuned....

digiblur, 2019-10-23:
Kept rolling back version tags and it started working again on 1.13.6

digiblur, 2019-10-23:
Next one up is 1.14.0b1 and nothing in between listed on docker hub. Rolled to it and it stopped working.

OttoWinter, 2019-10-24:
I can reproduce the issue now.

Does not happen with my dev py3 install.

Happens with `esphome/esphome:dev` docker image. From some quick debugging it looks like it actually receives the messages, but for some reason doesn't print them ¯\_(ツ)_/¯

OttoWinter, 2019-10-24:
Ah ok turns out I was debugging wrong. It does also happen on my local dev install.

Found the error. Apparently the eclipse mqtt library for python is built in a strange way. Any exception raised in the message callback (in this case a unicode error) is silently ignored, with no log output at all ¯\_(ツ)_/¯

closed 2019-10-24 by OttoWinter

digiblur, 2019-10-24:
Awesome! Thanks for looking into this and correcting it.