esphome/issues #770: witty cloud LDR

lpt2007, 2019-10-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Witty cloud ESP8266 nodemcu
**Affected component:**

**Description of problem:**

I try to use witty cloud LDR

[image]

Somethink is wrong with filters:

```
    filters:
      - lambda: (x * 3.3 / 10000.0) * 2000000.0
```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```
sensor:
  - platform: adc
    pin: A0
    name: Illuminance
    unit_of_measurement: lux
    filters:
      - lambda: (x * 3.3 / 10000.0) * 2000000.0
```
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-10-24:
1. Please post the error you get. I can't read minds and I won't try each example someone posts just because they didn't post the log output.

2. Try `lambda: 'return (x * 3.3 / 10000.0) * 2000000.0;'`

closed 2019-10-24 by OttoWinter

lpt2007, 2019-10-26:
Thanks now is working.