esphome/issues #771: WARNING Disconnected from API: Timeout while waiting for message response!

KarlosHit, 2019-10-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
I have 2 relays controlled with an ESP32 and a relay controlled with an ESP01.
Constantly, the three relays are disconnected from the HA API, giving a message of: "changed to unavailable", and immediately afterwards they are reconnected. It happens randomly, but constantly.

I thought it would be a disconnection from the API of HA, or from MQTT, so I did a "show logs" in ESPHome, and found that it gives many flaws of: "WARNING Disconnected from API: Timeout while waiting for message response!".

However, I have been able to verify that when I have "show logs" open in ESPHome, the disconnection in HA stops, even though in "show logs" it still occasionally reports the error "WARNING Disconnected from API: Timeout while waiting for message response! " and in Home Assistant the message "changed to unavaiable" does not appear, but the relays go off without any order. The message is: "RelayX turned off".

**Problem-relevant YAML-configuration entries:**
```yaml
# Definición de variables:
substitutions:
  _ssid: '****'
  _passwifi: '****'
  _passota: '****'
  _usermosquitto: '*****'
  _passmosquitto: '****'

esphome:
  name: sensordht11
  platform: ESP32
#  board: esp-wrover-kit
#  board: nodemcu-32s
  board: esp32doit-devkit-v1

wifi:
  ssid: $_ssid
  password: $_passwifi
  manual_ip:
    static_ip: 192.168.31.173
    gateway: 192.168.31.1
    subnet: 255.255.255.0
  
i2c:
  sda: 21
  scl: 22
  scan: true

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: $_passota

ota:
  password: $_passota

web_server:
  port: 80

mqtt:
  broker: 192.168.31.xx
  username: $_usermosquitto
  password: $_passmosquitto

sensor:
  - platform: dht
    pin: 
      number: GPIO4
    model: DHT22
    temperature:
      name: "Temperatura DHT22"
    humidity:
      name: "Humedad Relativa DHT22"
    update_interval: 60s
  - platform: bh1750
    name: "BH1750 Illuminance"
    address: 0x23
    update_interval: 60s
    
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO27
    name: "PIR Sensor"
    device_class: motion
    
switch:
  - platform: gpio
    pin: GPIO26
    name: "Relé 3"
    inverted: yes
  - platform: gpio
    pin: GPIO25
    name: "Relé 2"
    inverted: yes
```

**Logs (if applicable):**

```
![Registro HA](https://user-images.githubusercontent.com/48394829/67603834-4a87df00-f77a-11e9-911a-ebf86e6b8e04.jpg)

![Show logs](https://user-images.githubusercontent.com/48394829/67603836-4e1b6600-f77a-11e9-886b-f6988ee012a8.jpg)

```

**Additional information and things you've tried:**

ng-galien, 2019-10-26:
You have this kind off message when the ESP crashes. You change logger to VERY_VERBOSE  to find which component fails.

OttoWinter, 2019-10-26:
Adding to what @ng-galien said, please connect the device via USB while the error is happening. If the device is crashing that's the only way to get logs of the crash happening (because during the crash there's obviously no network connection)

KarlosHit, 2019-10-27:
I will, but it is happening with 2 microcontrollers (esp01 and esp32) the funny thing is that only the relays. Is it posible that home assistant fail?

KarlosHit, 2019-10-27:
Maybe we have something here:

**Relay fail in Home Assistant:**
[image]

**Log ESPHome:**
[02:44:43][D][sensor:092]: 'BH1750 Illuminance': Sending state 59.16666 lx with 1 decimals of accuracy
[02:44:43][V][mqtt:372]: Publish(topic='sensordht11/sensor/bh1750_illuminance/state' payload='59.2' retain=1)
[02:44:43][VV][component:128]: Running interval '':0 with interval=10000 last_execution=583888 (now=593893)
[02:44:47][VV][api:598]: on_ping_request_
[02:44:52][VV][api:598]: on_ping_request_
[02:44:53][VV][component:128]: Running interval '':0 with interval=10000 last_execution=593888 (now=603899)
[02:44:57][VV][api:598]: on_ping_request_
[02:45:02][VV][api:598]: on_ping_request_
[02:45:03][VV][component:128]: Running interval '':0 with interval=10000 last_execution=603888 (now=613889)
[02:45:07][VV][api:598]: on_ping_request_
[02:45:12][VV][api:598]: on_ping_request_
[02:45:13][VV][component:128]: Running interval '':0 with interval=10000 last_execution=613888 (now=623895)
[02:45:17][VV][api:598]: on_ping_request_
[02:45:22][VV][api:598]: on_ping_request_
[02:45:23][VV][component:128]: Running interval '':0 with interval=10000 last_execution=623888 (now=633901)
[02:45:27][VV][api:598]: on_ping_request_
[02:45:32][VV][api:598]: on_ping_request_
[02:45:33][VV][component:128]: Running interval '':0 with interval=10000 last_execution=633888 (now=643890)
[02:45:37][VV][api:598]: on_ping_request_
[02:45:41][VV][component:128]: Running interval 'update':0 with interval=60000 last_execution=591590 (now=651606)
[02:45:41][VV][dht:147]: Data: Hum=0b0000001001010010, Temp=0b0000000011011011, Checksum=0b00101111
[02:45:41][D][dht:048]: Got Temperature=21.9°C Humidity=59.4%
[02:45:41][V][sensor:013]: 'Temperatura DHT22': Received new state 21.900000
[02:45:41][D][sensor:092]: 'Temperatura DHT22': Sending state 21.90000 °C with 1 decimals of accuracy
[02:45:41][V][mqtt:372]: Publish(topic='sensordht11/sensor/temperatura_dht22/state' payload='21.9' retain=1)
[02:45:41][V][sensor:013]: 'Humedad Relativa DHT22': Received new state 59.400002
[02:45:41][D][sensor:092]: 'Humedad Relativa DHT22': Sending state 59.40000 % with 0 decimals of accuracy
[02:45:41][V][mqtt:372]: Publish(topic='sensordht11/sensor/humedad_relativa_dht22/state' payload='59' retain=1)
[02:45:42][VV][api:598]: on_ping_request_
[02:45:42][VV][component:128]: Running interval 'update':0 with interval=60000 last_execution=593370 (now=653386)
[02:45:42][VV][i2c:056]: Beginning Transmission to 0x23:
[02:45:42][VV][i2c:093]:     Writing 0b00100001 (0x21)
[02:45:42][VV][i2c:061]:     Transmission ended. Status code: 0x00
[02:45:42][VV][component:075]: set_timeout(name='illuminance', timeout=180)
[02:45:43][VV][component:128]: Running timeout 'illuminance':1 with interval=180 last_execution=653402 (now=653586)
[02:45:43][VV][i2c:083]: Requesting 2 bytes from 0x23:
[02:45:43][VV][i2c:126]:     Received 0b1000000011111111 (0x80FF)
[02:45:43][D][bh1750.sensor:071]: 'BH1750 Illuminance': Got illuminance=27519.2lx
[02:45:43][V][sensor:013]: 'BH1750 Illuminance': Received new state 27519.166016
[02:45:43][D][sensor:092]: 'BH1750 Illuminance': Sending state 27519.16602 lx with 1 decimals of accuracy
[02:45:43][V][mqtt:372]: Publish(topic='sensordht11/sensor/bh1750_illuminance/state' payload='27519.2' retain=1)
[02:45:43][VV][component:128]: Running interval '':0 with interval=10000 last_execution=643888 (now=653903)
[02:45:47][VV][api:598]: on_ping_request_
[02:45:52][VV][api:598]: on_ping_request_
[02:45:53][VV][component:128]: Running interval '':0 with interval=10000 last_execution=653888 (now=663893)
[02:45:57][VV][api:598]: on_ping_request_
[02:46:02][VV][api:598]: on_ping_request_
[02:46:03][VV][component:128]: Running interval '':0 with interval=10000 last_execution=663888 (now=673901)
[02:46:07][VV][api:598]: on_ping_request_
[02:46:12][VV][api:598]: on_ping_request_
[02:46:13][VV][component:128]: Running interval '':0 with interval=10000 last_execution=673888 (now=683891)
[02:46:17][VV][api:598]: on_ping_request_
[02:46:23][VV][api:598]: on_ping_request_
[02:46:23][VV][component:128]: Running interval '':0 with interval=10000 last_execution=683888 (now=693897)
[02:46:28][VV][api:598]: on_ping_request_
[02:46:33][VV][api:598]: on_ping_request_
[02:46:33][VV][component:128]: Running interval '':0 with interval=10000 last_execution=693888 (now=703903)
[02:46:38][VV][api:598]: on_ping_request_
[02:46:41][VV][component:128]: Running interval 'update':0 with interval=60000 last_execution=651590 (now=711603)
[02:46:41][VV][dht:147]: Data: Hum=0b0000001001010010, Temp=0b0000000011011100, Checksum=0b00110000
[02:46:41][D][dht:048]: Got Temperature=22.0°C Humidity=59.4%

OttoWinter, 2019-10-27:
Well I don't see anything that would point to an issue in the logs.

`on_ping_request` is a normal message transmitted every couple of seconds to keep the connection alive, and does not have to do with this.

What's more, the fact that `on_ping_request_` continues being shown during and after the fault means that the connection is fully alive and well. Otherwise these messages would not show and an error would be printed in the logs.

Two things to check next:

 1. It appears you're using both MQTT and API. Please say which one is used for communication with Home Assistant.
 2. Also have a look in your Home Assistant logs. If HA and the ESP are connected to each other via MQTT, it looks like the HA instance might be loosing the connection to the MQTT broker.

KarlosHit, 2019-10-27:
I am integrating with MQTT, I will remove the API from the ESP to see if the problem is fixed. After reporting the result. Thank you!

KarlosHit, 2019-10-27:
I removed API from my two ESPs, it still gives me frequent "unavaiable" errors.
In case it can help, when I pressed to change the state of all the relays at once (they are in the same card of entities of HA), the error occurred at that moment, and all the devices of the ESP32 were without communication for a moment.
I have attached HA log and ESPHome log at the time the error occurred.
I also attached an image of the HA log.

**LOG HA**
`2019-10-27 22:03:14 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor sensordht11 bh1750_illuminance, sending update
2019-10-27 22:03:14 INFO (MainThread) [homeassistant.components.mqtt] Got update for entity with hash: ('sensor', 'sensordht11 bh1750_illuminance') '{'unit_of_measurement': 'lx', 'icon': 'mdi:brightness-5', 'name': 'BH1750 Illuminance', 'state_topic': 'sensordht11/sensor/bh1750_illuminance/state', 'availability_topic': 'sensordht11/status', 'unique_id': 'ESPsensorbh1750_illuminance', 'device': {'identifiers': '246f28171dc0', 'name': 'sensordht11', 'sw_version': 'esphome v1.13.6 Oct 27 2019, 20:30:01', 'model': 'DOIT ESP32 DEVKIT V1', 'manufacturer': 'espressif'}, 'platform': 'mqtt', 'discovery_hash': ('sensor', 'sensordht11 bh1750_illuminance')}'
2019-10-27 22:03:14 INFO (MainThread) [homeassistant.components.mqtt] Updating component: sensor.bh1750_illuminance
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor sensordht11 bh1750_illuminance, sending update
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt] Got update for entity with hash: ('sensor', 'sensordht11 bh1750_illuminance') '{'unit_of_measurement': 'lx', 'icon': 'mdi:brightness-5', 'name': 'BH1750 Illuminance', 'state_topic': 'sensordht11/sensor/bh1750_illuminance/state', 'availability_topic': 'sensordht11/status', 'unique_id': 'ESPsensorbh1750_illuminance', 'device': {'identifiers': '246f28171dc0', 'name': 'sensordht11', 'sw_version': 'esphome v1.13.6 Oct 27 2019, 20:30:01', 'model': 'DOIT ESP32 DEVKIT V1', 'manufacturer': 'espressif'}, 'platform': 'mqtt', 'discovery_hash': ('sensor', 'sensordht11 bh1750_illuminance')}'
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt] Updating component: sensor.bh1750_illuminance
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: binary_sensor sensordht11 pir_sensor, sending update
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt] Got update for entity with hash: ('binary_sensor', 'sensordht11 pir_sensor') '{'device_class': 'motion', 'name': 'PIR Sensor', 'state_topic': 'sensordht11/binary_sensor/pir_sensor/state', 'availability_topic': 'sensordht11/status', 'unique_id': 'ESPbinary_sensorpir_sensor', 'device': {'identifiers': '246f28171dc0', 'name': 'sensordht11', 'sw_version': 'esphome v1.13.6 Oct 27 2019, 20:30:01', 'model': 'DOIT ESP32 DEVKIT V1', 'manufacturer': 'espressif'}, 'platform': 'mqtt', 'discovery_hash': ('binary_sensor', 'sensordht11 pir_sensor')}'
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt] Updating component: binary_sensor.pir_sensor
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: switch sensordht11 rel_3, sending update
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt] Got update for entity with hash: ('switch', 'sensordht11 rel_3') '{'name': 'Relé 3', 'state_topic': 'sensordht11/switch/rel_3/state', 'command_topic': 'sensordht11/switch/rel_3/command', 'availability_topic': 'sensordht11/status', 'unique_id': 'ESPswitchrel_3', 'device': {'identifiers': '246f28171dc0', 'name': 'sensordht11', 'sw_version': 'esphome v1.13.6 Oct 27 2019, 20:30:01', 'model': 'DOIT ESP32 DEVKIT V1', 'manufacturer': 'espressif'}, 'platform': 'mqtt', 'discovery_hash': ('switch', 'sensordht11 rel_3')}'
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt] Updating component: switch.rele_3
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: switch sensordht11 rel_2, sending update
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor sensordht11 temperatura_dht22, sending update
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt] Got update for entity with hash: ('switch', 'sensordht11 rel_2') '{'name': 'Relé 2', 'state_topic': 'sensordht11/switch/rel_2/state', 'command_topic': 'sensordht11/switch/rel_2/command', 'availability_topic': 'sensordht11/status', 'unique_id': 'ESPswitchrel_2', 'device': {'identifiers': '246f28171dc0', 'name': 'sensordht11', 'sw_version': 'esphome v1.13.6 Oct 27 2019, 20:30:01', 'model': 'DOIT ESP32 DEVKIT V1', 'manufacturer': 'espressif'}, 'platform': 'mqtt', 'discovery_hash': ('switch', 'sensordht11 rel_2')}'
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt] Updating component: switch.rele_2
2019-10-27 22:03:37 INFO (MainThread) [homeassistant.components.mqtt] Got update for entity with hash: ('sensor', 'sensordht11 temperatura_dht22') '{'unit_of_measurement': '°C', 'icon': 'mdi:thermometer', 'name': 'Temperatura DHT22', 'state_topic': 'sensordht11/sensor/temperatura_dht22/state', 'availability_topic': 'sensordht11/status', 'unique_id': 'ESPsensortemperatura_dht22', 'device': {'ide [... cut ...]
[22:04:31][VV][i2c:061]:     Transmission ended. Status code: 0x00
[22:04:31][VV][component:075]: set_timeout(name='illuminance', timeout=180)
[22:04:31][VV][component:128]: Running timeout 'illuminance':1 with interval=180 last_execution=5201377 (now=5201560)
[22:04:31][VV][i2c:083]: Requesting 2 bytes from 0x23:
[22:04:31][VV][i2c:126]:     Received 0b0000000001010001 (0x0051)
[22:04:32][D][bh1750.sensor:071]: 'BH1750 Illuminance': Got illuminance=67.5lx
[22:04:32][V][sensor:013]: 'BH1750 Illuminance': Received new state 67.500000`

[image]

KarlosHit, 2019-10-28:
Perhaps this video can help to know the cause if the issue....
Sometimes the system down without aparent season.

https://youtu.be/aY7rDSbqgzw

OttoWinter, 2019-10-28:
```
[22:03:13][I][mqtt:202]: MQTT Connected!
```

There we have the problem. The ESP is disconnecting from the MQTT broker (wasn't in the previous log, so I didn't think that was the issue).

But the logs you're showing here are the over-the-air logs, which are of little help when the device is disconnecting. It requires the logs from the device via USB while the issue is happening to see more clearly.

But this also looks like a classic case of https://github.com/esphome/issues/issues/455 too. Anyway, getting `"changed to unavailable"` is normal and just means that the ESP briefly disconnected from the MQTT broker. As long as it quickly reconnects I don't think this is a major issue.

KarlosHit, 2019-10-28:
You are right, however, you have to consider:
1.- That many times the disconnection does not last a few seconds, but it is several minutes.
2.- That having an event log collapsed by the message: "changed to unavaiable" is really uncomfortable.
3.- Finally, I think that, even if it was bearable, it really demonstrates the existence of a problem.
I have tried to take the whole system to the side of the router, in case it was a signal problem, but it also occurs there.
I really think that several problems may be happening, because if before there was the message "MQTT Connected!" It is because that message was not issued.
I will try to investigate a little more and expose the results.

KarlosHit, 2019-10-28:
Well, I think I understand what happens.
Once I have removed the API and keep only mqtt, it has stopped giving random messages "changed to unavaiable". Now the system is stable all the time, only that when I turn on or off a relay, and it has also been a long time since the last action, a reconnection to mqtt (the one we discussed before) occurs, resulting in a previous message of "changed to unavaiable, this same message is executed on the other binary sensors (that is, on all the binary sensors that the ESP has (motion sensor, etc.), as well as the rest of the existing relays.
Ultimately now, if I find it admissible since it is not a repetitive message, but transitional during the activation or deactivation of a relay, and only when you admire there is a reconnection to the mqtt broker. In any case, it is something that could be improved.

Take this opportunity to ask for advice on MQTT or API, which one to choose? I like MQTT because it is transversal and all the knowledge will help me for other systems, API is only for HA, this convinces me less.

I also wanted to thank the ESPHome team.
Regards.

KarlosHit, 2019-11-01:
Postdata information: In addition to the problem of API coexistence with MQTT, I have discovered that, by activating all three relays at the same time, there is a current deficit in the protoboard that sometimes causes microcontrollers to restart.
Regards.

pacomm, 2019-11-06:
The same here with a ESP32 and a bluetooth xiaomi lywsdcgq. 
Home Assistant (hass.io) / Raspberry Pi 3 b+ 

From esphome log:
```
...
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.38:6053 (192.168.1.38)
INFO Successfully connected to 192.168.1.38
...
```

From hassio log:

```
2019-11-05 10:09:14 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 10:12:45 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 10:19:51 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 10:34:06 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 14:15:16 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 14:48:27 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 15:58:55 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 17:44:08 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 18:17:08 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 18:59:13 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 21:31:04 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-05 23:12:58 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 02:24:32 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 05:01:56 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 06:29:30 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 10:15:57 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 10:29:24 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 13:56:22 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 14:06:01 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 15:35:34 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 15:39:14 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 15:49:13 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 15:49:39 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
2019-11-06 16:48:44 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
```

OttoWinter, 2019-11-07:
@pacomm That is unrelated to this issue. Please see https://github.com/esphome/issues/issues/824

closed 2019-11-07 by OttoWinter