esphome/issues #772: Transition Time not Working Properly

alexyao2015, 2019-10-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

Feit Bulb
**Affected component:**

Light, REST API

**Description of problem:**
Transition time of 0.4 does not work properly when sent over the REST API. It appears that decimal transition times are ignored and 0 is instead used as the transition. Logs look exactly the same when using a default transition time of 0.4 and when sending 0.4 as the transition time. Only difference is one transition time is actually 0.4 (default) and one just acts like its a 0 transition time (REST API).

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
[19:34:58][D][light:261]: 'color_led' Setting:
[19:34:58][D][light:283]:   Red=100%, Green=42%, Blue=0%
[19:34:58][D][light:300]:   Transition Length: 0.4s
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[19:34:58][D][light:261]: 'color_led' Setting:
[19:34:58][D][light:283]:   Red=100%, Green=42%, Blue=0%
[19:34:58][D][light:300]:   Transition Length: 0.4s
```

**Additional information and things you've tried:**

OttoWinter, 2019-10-26:
Please explain the logs you see a bit more. Only thing I see is two times the same logs (though that may be what you're meaning). Is it correct that the first one is from a request without transition length and the second one with the transition length parameter?

What happens when you send a value other than 0.4 over the network?

What version of ESPHome is this?

What light platform is this using? rgb + esp8266_pwm?

Please attach the request you're sending so that I can verify it (curl, just the URL or whatever you're using).

alexyao2015, 2019-10-27:
This is version 1.13.6.

I've added additional information into the original comment including curl examples and corresponding logs. 

It appears that ESPHome truncates transition times and does not accept a float value. For example, if I send 1.9, it shows transition time as 1s instead of 1.9s. Same 0.5 which has no transition. 

The issue here appears to be floating transition values in the url. Potentially if floating points are not allowed, use milliseconds instead of seconds when interpreting transition times?

This is using fastled rgb.

closed 2019-10-27 by OttoWinter