esphome/issues #773: Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception) after upgrading from 1.12.2 to 1.13.6 with a large amount of miflora sensors

sn3ak, 2019-10-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome 1.13.6 on FreeBSD 11.3 with python 3.7 compiling from ESPHome Dashboard

**ESP (ESP32/ESP8266, Board/Sonoff):**

MH-ET LIVE ESP32 MinKIT V1.1
**Affected component:**

https://esphome.io/components/sensor/xiaomi_miflora.html ?? Long configuration file?? substitutions?

**Description of problem:**
I upgraded my miflora sensor monitor which watches 30 miflora sensors to 1.13.6 and now it fails. I've changed my configuration a bit to match current documentation, which didn't change anything, I moved some of my substitutions to secrets. 

I've removed senors and recompiled, and finally after cutting my configuration to 15 or below it works. I've included the configuration with 16 senors, which fails.

Secondary problem: I took in a spider plant that was in need of a home, and I now have 30+. I've become the crazy plant guy.. Please send help.

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: den
  platform: ESP32
  board: mhetesp32minikit

wifi:
  ssid: !secret ssid
  password: !secret wpapsk

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret apipass

ota:
  password: !secret otapass

esp32_ble_tracker:

substitutions:
  miflora_1_name: one
  miflora_2_name: two
  miflora_3_name: three
  miflora_4_name: four
  miflora_5_name: five
  miflora_6_name: six
  miflora_7_name: seven
  miflora_8_name: eight
  miflora_9_name: nine
  miflora_10_name: ten
  miflora_11_name: eleven
  miflora_12_name: twelve
  miflora_13_name: thirteen
  miflora_14_name: fourteen
  miflora_15_name: fifteen
  miflora_16_name: sixteen
  miflora_17_name: seventeen
  miflora_18_name: eighteen
  miflora_19_name: nineteen
  miflora_20_name: twenty
  miflora_21_name: twentyone
  miflora_22_name: twentytwo
  miflora_23_name: twentythree
  miflora_24_name: twentyfour
  miflora_25_name: twentyfive
  miflora_26_name: twentysix
  miflora_27_name: twentyseven
  miflora_28_name: twentyeight
[... 244 lines cut ...]
[16:44:32]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[16:44:32]Core 1 register dump:
[16:44:32]PC      : 0x401d24e5  PS      : 0x00060136  A0      : 0x801d50c5  A1      : 0x3ffcc410  
[16:44:32]A2      : 0x3ffcc810  A3      : 0x000709ff  A4      : 0x00000100  A5      : 0x3ffcc460  
[16:44:32]A6      : 0x00000000  A7      : 0x000709ff  A8      : 0x3ffcc410  A9      : 0x3ffcc400  
[16:44:32]A10     : 0x3ffcc810  A11     : 0x3ffc04e0  A12     : 0x00000014  A13     : 0x3ffcc824  
[16:44:32]A14     : 0x00f20000  A15     : 0x00f10000  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
[16:44:32]EXCVADDR: 0x00000000  LBEG    : 0x401d24e5  LEND    : 0x401d24e9  LCOUNT  : 0x000000eb  
[16:44:32]
[16:44:32]Backtrace: 0x401d24e5:0x3ffcc410 0x401d50c2:0x3ffcc880 0x401d55ec:0x3ffcc8a0 0x401650b1:0x3ffcc960 0x401653aa:0x3ffcc990 0x400e8b47:0x3ffcc9c0 0x400e47ed:0x3ffcc9e0 0x400d3e98:0x3ffcca20 0x400d4066:0x3ffcca50 0x400d9b7a:0x3ffcca70 0x400d9935:0x3ffcca90 0x400df8b6:0x3ffccae0 0x400e51cb:0x3ffce3a0 0x400943b1:0x3ffce3c0
[16:44:32]
[16:44:32]Rebooting...
```

**Additional information and things you've tried:**
Removing all substitutions and Sensors from yaml. Result: works
Add in all 30 substitutions. Result: works
Change syntax to match miflora page and add all 30 senors. Result: fails
Remove all but 1 sensor. Result: works
Change to a total of 10 sensors. Result: works
Change to a total of 20 sensors. Result: fails
Change to a total of 15 sensors. Result: works
Change to a total of 17 sensors. Result: fails
Change to a total of 16 sensors. Result: works
Change to a total of 15 sensors. Result: works

sn3ak, 2019-10-26:
I've since built a 1.12.2 virtualenv on the same machine with the same configuration. Using the yaml that I tweaked to match the docs. All 30 sensors showed up and work correctly.

OttoWinter, 2019-10-26:
This is an out of stack memory issue.

Looks like a combination of two things:

1. Recent versions of ESPHome probably use more stack space - most variables are stored globally so that they're accessible from custom components.

2. Around the time of v1.10.0, the BLE stack was moved out from a separate FreeRTOS task (https://github.com/esphome/esphome-core/commit/db443c91c4b7317e32e8babe4495c6d8d3dd41a3#diff-20d4b1b231ae61bef48f8fce5b8442ff) to fix some race conditions.

To check what it is, please connect the device via USB to the host running the ESPHome dashboard. If you view the logs from there, all crashes in the logs will be auto-decoded. Then we can find out where the stack filled up.

sn3ak, 2019-11-03:
I upgraded to 1.14.2, Attached is the decoded trace. I did update my config for the miflora rename.
Also, I wouldn't think the BLE stack move would be a part of the issue, since 1.12.2 was working fine with this configuration.

```
Writing at 0x000e4000... (100 %)
Wrote 1508400 bytes (874960 compressed) at 0x00010000 in 22.4 seconds (effective 539.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
================================================ [SUCCESS] Took 45.51 seconds ================================================
INFO Successfully uploaded program.
INFO Starting log output from /dev/cuaU0 with baud rate 115200
[01:53:20]***ERROR*** A stack overflow in task loopTask has been detected.
[01:53:20]abort() was called at PC 0x400922dc on core 1
[01:53:20]
[01:53:20]Backtrace: 0x40092094:0x3ffcbae0 0x400922c5:0x3ffcbb00 0x400922dc:0x3ffcbb20 0x4008f3a8:0x3ffcbb40 0x4009100c:0x3ffcbb60 0x40090fc2:0x00000000
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d den -t idedata
WARNING Decoded 0x40092094: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x400922c5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x400922dc: vApplicationStackOverflowHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x4008f3a8: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
WARNING Decoded 0x4009100c: _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:406
WARNING Decoded 0x40090fc2: _frxt_int_exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:206
[01:54:00]
[01:54:00]Rebooting...
[01:54:00]ets Jun  8 2016 00:22:57
```

OttoWinter, 2019-11-04:
> Also, I wouldn't think the BLE stack move would be a part of the issue, since 1.12.2 was working fine with this configuration.

With each release, as new features are added more and more of the stack is used. Usually, that is not a problem - unless you've got a device with a lot of sensors attached.

The move BLE to new thread move may not be directly responsible, but it would help eliminate the issue.

dedors, 2020-01-03:
I have the same problem, when adding more than 15 xiaomi sensors (hhccjcy01 and lywsdcgq), I'm getting Core Panic.
I did a downgrade to 12.2, and with correct old platform name, I could add all of my 24 sensors. Home Assistant received all 24 entity names, but it only receives data of 15 sensors(!). 

I guess I have to wait for a release with the fix.

andrelung, 2020-03-21:
I got the same problem on a m5stack and and ESP-WROOM-32

Axellum, 2020-05-22:
Hi,
I have the same problem on an ESP32 with about fifty sensors for a screen. Do you think that with an 8 MB or 16MB Flash version will solve the problem?

dedors, 2020-06-02:
Any update on this issue? 12.2 supports more than 15 but is missing support for some new devices.

manonfgoo, 2020-08-16:
I had the same issue with VERBOSE logging,  I switched back to DEBUG and everything went fine

dedors, 2021-01-15:
I did some extended testing with xiaomi_lywsd03mmc, xiaomi_lywsdcgq, xiaomi_hhccjcy01 (has 5 instead of 3 sub-sensors attached) and xiaomi_cgg1 and had different results with the amount of sensors I could run. 

It seem this is a memory limitation problem. With very short names, I could add a few sensor more, and if delete some sub-sensors (e.g. removing all temp and battery entries) I could also add some more, thats why xiaomi_hhccjcy01 is more affected (5 compared to 3 sub-sensors).
With very short (unusable) names and reducing the subsensors to 1-2 each, I could get up to 35 sensors running. 

With 12.2, using same amount of sub-sensors and same names, I could add about double of the sensors (maybe more, didn't test the limit). 

Is there a chance to get some more memory back for this?

vautieri, 2021-02-19:
Sounds like the race condition "fixed" created other problems.  It would be nice to have a configuration option to use the old way and to set the stack size of the thread

stefan2205, 2022-04-15:
I ran into the same issue today, but one of the linked issues above has a workaround that works: Increase of stack size in main.cpp helps: https://github.com/esphome/issues/issues/855#issuecomment-885903536