esphome/issues #774: Flashing ESP32 (using CLI) always bootloops, works with GUI

thibmaek, 2019-10-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

macOS, Python 2.7 installed via Brew. Using SiLabs USB UART Driver

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, Wroom-32 model

**Affected component:**

Esphome CLI (`pip2.7 install esphome`)

**Description of problem:**
When trying to flash a config for this board using the CLI it will always fail and output a bootloop log. Compiling first with `esphome conf.yaml compile` and the uploading the firmware.bin with the esphome-flasher GUI tool works.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32_plant_monitor
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:

esp32_ble_tracker:

sensor:
  - platform: xiaomi_miflora
    mac_address: 'C4:7C:8D:65:CA:76'
    temperature:
      name: "Xiaomi MiFlora Temperature"
    moisture:
      name: "Xiaomi MiFlora Moisture"
    illuminance:
      name: "Xiaomi MiFlora Illuminance"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity"
    battery_level:
      name: "Xiaomi MiFlora Battery Level"
```

**Logs (if applicable):**

```log
[22:39:53]ets Jun  8 2016 00:22:57
[22:39:53]
[22:39:53]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:39:53]configsip: 0, SPIWP:0xee
[22:39:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:39:53]mode:DIO, clock div:1
[22:39:53]load:0x3fff0018,len:4
[22:39:53]load:0x3fff001c,len:808
[22:39:53]load:0x40078000,len:6084
[22:39:53]load:0x40080000,len:6696
[22:39:53]entry 0x400802e4
[22:39:53]user code done
[22:39:53]ets Jun  8 2016 00:22:57
[22:39:53]
[22:39:53]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:39:53]configsip: 0, SPIWP:0xee
[22:39:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:39:53]mode:DIO, clock div:1
[22:39:53]load:0x3fff0018,len:4
[22:39:53]load:0x3fff001c,len:808
[22:39:53]load:0x40078000,len:6084
[22:39:53]load:0x40080000,len:6696
[22:39:53]entry 0x400802e4
[22:39:53]user code done
[22:39:53]ets Jun  8 2016 00:22:57
```

**Additional information and things you've tried:**
- Updated to latest CLI: `pip2.7 install esphome --upgrade`
- Tried 4 different USB to Micro USB cables
- Referencing to [this issue](https://github.com/esphome/issues/issues/598) where I was asked to create additional bug report.

While this workaround is nice, flashing without GUI would be very nice 👍

thibmaek, 2019-10-27:
I was able to flash it once and see log ouput for the plant monitor. Trying to reflash an update config does not work:

```log
Using '/dev/cu.SLAB_USBtoUART' as serial port.
Connecting........_____....._____....._____....._____....._____....._____....._____
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: Invalid head of packet (0x1B)
```

OttoWinter, 2019-10-27:
Ah now I understand what you meant in the other issue. The issue there was referring to the latest beta.

Simple solution: Install the latest esphome beta.

That's pretty much what esphome-flasher's fix was: using the latest bootloader for the ESP32s. The latest ESPHome version has that installed.

OttoWinter, 2019-10-27:
> I was able to flash it once and see log ouput for the plant monitor. Trying to reflash an update config does not work:

That is an issue with some FTDI adapters, try holding down the BOOT button (or connect GPIO0 to GND) while flashing.

closed 2019-10-27 by OttoWinter

thibmaek, 2019-10-27:
Nice, thank you! I will work around this using using GUI for now.
As for my last comment, I mainly used ESP8266 where the flash mode was activated automatically but for ESP32 I needed to hold the BOOT button.

thibmaek, 2019-10-27:
Thank you for the quick reply and your amazing software.