esphome/issues #775: Multiple sensors on_value_range - fails to boot

kasparsv, 2019-10-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Ubuntu / VM

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 and ESP8266 NodeMCU board

**Affected component:**

sensor - on_value_range
https://esphome.io/components/sensor/index.html#on-value-range

**Description of problem:**
When adding on_value_range for multiple sensors the board fails to boot. I tried on ESP32 and NodeMCU boards with two Dallas temperature sensors (both on the same GPIO).

**Problem-relevant YAML-configuration entries:**
```yaml
dallas:
  - pin: GPIO3
    update_interval: 20s

sensor:
  - platform: dallas
    address: 0x0C011452D915AA28
    name: "Dallas 1"
    id: dallas_1
    unit_of_measurement: "°C"
    icon: "mdi:coolant-temperature"
    accuracy_decimals: 1

    on_value_range:
      - below: 50
        then:
        - if:
            condition:
              switch.is_on: cooling_relay
            then:
              - switch.turn_off: cooling_relay
      - above: 55
        then:
        - if:
            condition:
              switch.is_off: cooling_relay
            then:
              - switch.turn_on: cooling_relay

  - platform: dallas
    address: 0x11011452D5C2AA28
    name: "Dallas 2"
    id: dallas_2
    unit_of_measurement: "°C"
    icon: "mdi:temperature-celsius"
    accuracy_decimals: 1

    on_value_range:
      - below: 32
        then:
        - if:
            condition:
              switch.is_on: cooling_relay
            then:
              - switch.turn_off: cooling_relay

      - above: 36
        then:
          - if:
              condition:
                switch.is_off: cooling_relay
              then:
                - switch.turn_on: cooling_relay

switch:
  - platform: gpio
    pin: GPIO1
    id: cooling_relay
```

**Logs (if applicable):**

```
INFO Reading configuration...
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200

```

**Additional information and things you've tried:**
Tried on different ESP32 boards - same result
Tried deleting compiled files - same result
Tried on ESP8266
Every time when removing one on_value_range, the board boots successfully...

OttoWinter, 2019-10-28:
Works for me - have a `wifi_signal` sensor with two `on_value_range` triggers on it.

Your logs look incomplete - the ESP should definitely send *something*. Either it crashes and produces a stack trace or it goes into an infinite loop in which the watchdog will kick in and also produce a stack trace.

Please try getting the logs again but pressing the RESET button on the board while viewing the logs.

kasparsv, 2019-10-28:
I will try to get stack trace later this evening, as it was showing at one point. 

But just to clarify - the issue is not with two on_value_range under one sensor - the issue is with two sensors, each of which has on_value_range

OttoWinter, 2019-10-30:
Also tested that now and works fine here (uptime and wifi_signal sensors)

kasparsv, 2019-10-31:
I have tested some more and it seems now that it is not related directly to
on_value_range but something else, but it is unclear as of now whether it
is the board (or boards) themselves, wifi issue or size of the program. If
I change something in code it sometimes works, but sometimes goes into
bootloop, but I haven't localised the exact issue yet.

I will update when I know more.

closed 2019-11-11 by no-response[bot]