esphome/issues #776: LEDs flickering with ESP8266

pjmsullivan, 2019-10-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Linux 4.19.80-v7+ #1274 SMP Mon Oct 21 16:23:10 BST 2019 armv7l GNU/Linux
Raspberry pi 3
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 with 50 WS2812b LEDs on pin D1

**Affected component:**

https://esphome.io/components/light/fastled.html

**Description of problem:**
First @OttoWinter thank you so much for your work - this is fantastic.  I saw the previous issue you closed concerning ESP8266 hardware limitations results in flickering LEDs.  Previously, using an arduino sketch, effects were buttery smooth, but now they flicker with the same hardware with ESPHome.  Prior to switching to ESPHome, I resolved this issue with the inclusion of:
`#define FASTLED_ALLOW_INTERRUPTS 0`
located above `#include "FastLED.h"` (See [https://github.com/FastLED/FastLED/wiki/Interrupt-problems](url)).  
I did experiment with `max_refresh_rate` already.  Is there any way to add `FASTLED_ALLOW_INTERRUPTS 0` to eliminate the flickering?  
FYI - sorry if this is an obvious answer - I am not a pro.
Thanks again!

**Problem-relevant YAML-configuration entries:**
```yaml
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO5
    num_leds: 50
    rgb_order: RGB
    max_refresh_rate: 25ms
    name: "Kitchen LED Lights"
    effects:
      - random:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

jbkcrash, 2019-10-31:
@pjmsullivan Your mileage may vary on using this tactic. I am not an expert in ESPHome so take if for what it is worth and this is not addressing the lack of ability to do this inherent to ESPHome's device building interface.

This should work for your immediate purposes to build a glitch-free device, however it isn't something that will persist beyond your local patch and you would need to manually patch any future installations of ESPHome to get the same results.

Also the instructions would vary depending on the OS you are using, this is a Linux based fix.

The best answer is to get the ESPHome developers to add this as an option you enable in the device's .yaml file.

That all being said I had a very similar problem with a NodeMCU and this step resolved it for me:

You must locate the "fastled_light.h" that is included with the base ESPHome installation. On CentOS 7 it is here "/usr/local/lib/python3.6/site-packages/esphome/components/fastled_base/fastled_light.h".

This is obviously not the same on a Docker based installation and/or on other OS(s), but same goal, just need to find that file and add your define.

Once I found where ESPHome was including "FastLED.h", in "fastled_light.h", I used grep... I added the appropriate define above the FastLED include and it worked after rebuilding the device and new firmware.

You will also notice under the working directory where you build your device there is a copy of the fastled_light.h file buried in the src dir, but it is overwritten from the base file each time you run esphome as you build or run. So the change needs to be in the base source directory that ESPHome uses to get to compiled into your device. You can use "find / -name fastled_light.h" to find it on Linux for example.

Hopefully this works for you, So far I have a glitch-free 100 LED strip running rainbow mode for hours and no Watch Dog resets yet. My lightstrip glitched constantly prior to adding the define so for a NodeMCU it is a must to get consistent operation; now hopefully the device itself is stable.

OttoWinter, 2019-10-31:
That's certainly one way @jbkcrash.

One problem however is that disabling all interrupts for a while is not that good of an idea for ESP8266s. It will mess with all kinds of things (button presses, WiFi connectivity etc).

The neopixelbus integration has a different way of working around this: it uses the i2s or uart buses to send the data with high timing accuracy.

I would first recommend trying with `neopixelbus` - if that still persist then we can talk about potentially disabling interrupts.

jbkcrash, 2019-11-01:
@OttoWinter I understand fully. I am also not sure simply disabling interrupts is a 100% fix either, in further testing of more active lighting strategy I am still seeing an occasional glitch. Much, much better, but not totally resolved. The same LED strip using FastLED on an Arduino Nano, same power, etc... works flawlessly; so I do accept it is related to ESP device and extra work it is doing for WiFi. It took me awhile to accept this, I was originally suspicious of wiring, power noise, etc... I am ruling that out at this point.

I will look into neopixelbus_light. I have used fadecandy device previously on Raspberry Pi so I trust the library but have only used FastLED myself to date on Arduino platform. For testing I am using 100ea WS2811 LED strip, I will get you the results over the weekend.

Diaoul, 2019-11-10:
I also have this issue and reported it in #253. I'm back on this again as ESPHome is awesome and I want to manage my fleet of ESP with it. Thanks @OttoWinter for the awesome work :+1: 
I've faced this issue and solved it (or rather, used a workaround) with my library [LEDEffect](https://github.com/Diaoul/LEDEffect) and [this code](https://gist.github.com/Diaoul/63846d4bd0188f83d5c4ae68a9f3877f).
At build time I define `FASTLED_ESP8266_RAW_PIN_ORDER` `FASTLED_INTERRUPT_RETRY_COUNT` and `MQTT_MAX_PACKET_SIZE=512`.
Even without those options, I don't have as much as flickering as with ESPHome.

I seem to remember that Serial was causing issues due to a shared interrupt, is ESPHome interacting with Serial by default?

With rainbow effect, the problem manifests with one LED somewhere in the strip going crazy for a fraction of second and all the following LEDs on the strip are also affected.
It's worth noting that when there is no network activity, the flickering disappears. I could sometimes see the strip running fine a few seconds after a flash prior to a reboot.

Diaoul, 2019-11-10:
Oh one more thing also, it seems that with my code the strip "freeze" rather than "flicker" on network activity.
I need to test this more, it's been a dead project of mine for far too long...

OttoWinter, 2019-11-12:
> I seem to remember that Serial was causing issues due to a shared interrupt, is ESPHome interacting with Serial by default?

Yes, see logger->baud_rate option. You can disable serial logging.

Diaoul, 2019-11-13:
So I've tried disabling logger without success (even with baud_rate to 0 and another UART). I think the interrupt is used for .read anyways so outputing to Serial isn't affected.
Neopixelbus integration works fine, I don't have any flicker using DMA method on GPIO3.

I'm not very satisfied by this though, I don't want to give up on FastLED yet. @OttoWinter is there anyway to add the abovementioned compilation flags to ESPHome with the CLI like with PlatformIO's configuration file?
I've also found [this code](https://github.com/bkpsu/ESP-MQTT-JSON-Digital-LEDs/blob/master/ESP_MQTT_Digital_LEDs/ESP_MQTT_Digital_LEDs_w_JSON.ino#L23) that seems to have no flickering issue.
Is writing to LED also async? Or is it made in one go?

Diaoul, 2019-11-13:
Looks like those are the only options with FastLED:

* Reduce WiFi load
* Play with value of `FASTLED_INTERRUPT_RETRY_COUNT`
* `FASTLED_ALLOW_INTERRUPTS 0` will make the ESP8266 occasionnaly reset, don't use that
* Increase `WAIT_TIME` (see https://github.com/FastLED/FastLED/issues/306#issuecomment-227004501)
* Increase CPU speed to 160MHz

Or just switch to NeoPixelBus that doesn't rely on interrupts to do it's job. As far as ESPHome is concerned this looks exactly the same (can use effects on both).
Worth noting that there is a fork of FastLED with DMA support. Unless this is merged though, I'm not sure why one would want to do that instead of the neopixelbus integration.

I think ESPHome just uses more the interrupts (indirectly) than any other code, nothing really wrong about that *if the hardware wasn't faulty*.

OttoWinter, 2019-11-14:
> I'm not very satisfied by this though, I don't want to give up on FastLED yet.

Any reason for that? They're both exposed to the user through the same API interface in ESPHome, so it should not matter which ones is used.

> anyway to add the abovementioned compilation flags to ESPHome with the CLI like with 

Yes, you can edit the `common` section in the generated `platformio.ini` file. See the generated `<NODE_NAME>/platformio.ini` file.

> Is writing to LED also async? Or is it made in one go?

With these LEDs it must be done in one go - they have no clock signal so it's impossible to suspend execution in the meantime. The only option is to put the writing part to some peripheral, like UART or SPI via some hacks.

> I think ESPHome just uses more the interrupts (indirectly) than any other code

No, ESPHome does not use any interrupts directly or indirectly through platform code. There's no ticker or pin interrupts if the user hasn't explicitly put that in the config.

The only interrupts that happen are WiFi-related, and are the same as with any ESP project.

closed 2019-11-14 by no-response[bot]