esphome/issues #781: Hass.io Dashboard edit using iOS Safari

hagenuck1, 2019-10-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

General

**Affected component:**

ESPhome Dashboard

**Description of problem:**
Configuration changes can't be saved by using iOS devices and Safari. You can change the config, but it gets lost, when pressing save or upload.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-10-30:
Please post your Hass.io addon logs here. (The ones you see when you click the addon page for ESPHome and scroll down to logs)

I don't have an iOS device so can not test.

hagenuck1, 2019-10-30:
Sure, this is what the logs are looking after starting the add on, open the dashboard in the side menu, click edit on a config, add a new line anywhere, click save and then close. 
```
s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Hass.io Add-on: ESPHome
 ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.
-----------------------------------------------------------
 Add-on version: 1.13.6
 You are running the latest version of this add-on.
 System: HassOS 3.3  (armv7 / raspberrypi4)
 Home Assistant version: 0.100.2
 Supervisor version: 192
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[17:14:31] INFO: Waiting for dashboard to come up...
[17:14:31] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[17:14:33] INFO: Starting NGINX...
INFO 304 GET / (0.0.0.0) 80.84ms
INFO 304 GET /static/materialize-stepper.min.css?hash=a7ad7ee2 (0.0.0.0) 20.04ms
INFO 304 GET /static/esphome.css?hash=ad6176a4 (0.0.0.0) 2.59ms
INFO 304 GET /static/jquery.min.js?hash=2f6b11a7 (0.0.0.0) 3.27ms
INFO 304 GET /static/materialize.min.css?hash=ec1df3ba (0.0.0.0) 2.82ms
INFO 304 GET /static/materialize.min.js?hash=4be20daf (0.0.0.0) 2.63ms
INFO 304 GET /static/ace.js?hash=2280cf05 (0.0.0.0) 3.57ms
INFO 304 GET /static/jquery-ui.min.js?hash=a0b1425d (0.0.0.0) 3.19ms
INFO 304 GET /static/materialize-stepper.min.js?hash=9f34c4ff (0.0.0.0) 2.64ms
INFO 304 GET /static/jquery.validate.min.js?hash=b2d2cf16 (0.0.0.0) 2.58ms
INFO 304 GET /static/materialize.min.js?hash=4be20daf (0.0.0.0) 1.92ms
INFO 304 GET /static/esphome.js?hash=1805848d (0.0.0.0) 2.15ms
INFO 304 GET /static/materialize.min.js?hash=4be20daf (0.0.0.0) 1.60ms
INFO 304 GET /static/ace.js?hash=2280cf05 (0.0.0.0) 1.66ms
INFO 304 GET /static/materialize.min.js?hash=4be20daf (0.0.0.0) 1.69ms
INFO 304 GET /static/mode-yaml.js (0.0.0.0) 7.68ms
INFO 304 GET /static/theme-dreamweaver.js (0.0.0.0) 3.42ms
INFO 101 GET /ace (0.0.0.0) 1.72ms
INFO Running command 'esphome --dashboard -q /config/esphome vscode --ace'
INFO 304 GET /edit?configuration=stromzaehler.yaml (0.0.0.0) 8.98ms
```

OttoWinter, 2019-10-31:
Ok, looks like the YAML file is loaded correctly (as seen by the GET command).

Though the saving doesn't work (no POST command seen). Probably an issue in the JS somewhere that might need a polyfill for Safari for iOS. Though as I said I can't fix this because I have no such device.

bensuffolk, 2019-11-06:
I find on safari iOS you need to his save 2 or 3 times and it does eventually save it, just not first time.

closed 2020-04-29 by glmnet