esphome/issues #786: MQTT Fingerprint Compilation Error

moritzj29, 2019-10-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip on macOS

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, wemos D1 mini

**Affected component:**

https://esphome.io/components/mqtt.html#mqtt-ssl-fingerprints

**Description of problem:**
First of all, thank you very much for the esphome project! It is a great help!

I've been struggeling to set up a secure MQTT connection using the mqtt_fingerprint configuration. esphome does not compile successfully when setting a mqtt_fingerprint due to error of ESPAsyncTCP library. When omitting the mqtt_fingerprint option the same YAML compiles just fine.

I'm just wondering if this is just a problem of my system or if this is a general issue. Maybe it is just an issue of ESPAsyncTCP and not really an issue of esphome itself.

**Problem-relevant YAML-configuration entries:**
```yaml
mqtt:
  broker: IP_removed
  port: 8883
  username: user
  password: !secret mqtt-password-esphotwater
  ssl_fingerprints:
    - fingerprint_removed
  discovery: false #disable HA autodiscovery through MQTT

```

**Logs (if applicable):**

```
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0

[...]

Compiling .pioenvs/esphotwater/lib8a2/AsyncMqttClient_ID346/AsyncMqttClient.cpp.o
Compiling .pioenvs/esphotwater/lib8a2/AsyncMqttClient_ID346/AsyncMqttClient/Packets/ConnAckPacket.cpp.o
.piolibdeps/esphotwater/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'err_t AsyncServer::_poll(tcp_pcb*)':
.piolibdeps/esphotwater/ESPAsyncTCP/src/ESPAsyncTCP.cpp:1324:31: error: no matching function for call to 'AsyncClient::_recv(tcp_pcb*&, pbuf*&, int)'
         c->_recv(pcb, p->pb, 0);
                               ^
.piolibdeps/esphotwater/ESPAsyncTCP/src/ESPAsyncTCP.cpp:1324:31: note: candidate is:
.piolibdeps/esphotwater/ESPAsyncTCP/src/ESPAsyncTCP.cpp:565:6: note: void AsyncClient::_recv(std::shared_ptr<ACErrorTracker>&, tcp_pcb*, pbuf*, err_t)
 void AsyncClient::_recv(std::shared_ptr<ACErrorTracker>& errorTracker, tcp_pcb* pcb, pbuf* pb, err_t err) {
      ^
.piolibdeps/esphotwater/ESPAsyncTCP/src/ESPAsyncTCP.cpp:565:6: note:   candidate expects 4 arguments, 3 provided
Compiling .pioenvs/esphotwater/lib8a2/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PingRespPacket.cpp.o
Compiling .pioenvs/esphotwater/lib8a2/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubAckPacket.cpp.o
*** [.pioenvs/esphotwater/lib377/ESPAsyncTCP/ESPAsyncTCP.cpp.o] Error 1

```

**Additional information and things you've tried:**
using current git version of ESPAsyncTCP library, changing board to esp01_1m

OttoWinter, 2019-10-31:
That indeed looks like an issue in the ESPAsyncTCP library.

> using current git version of ESPAsyncTCP library

What do you mean by that? How did you manually install the dev version of ESPAsyncTCP? Did you also try with the normal version?

Please also try with latest beta.

moritzj29, 2019-11-02:
I tried using the development version of the ESPAsyncTCP library by including this dependecy in `platformio.ini`

```[common]
lib_deps =
    https://github.com/me-no-dev/ESPAsyncTCP.git
```

```Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient-esphome> 0.8.3
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.2 #1547686
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPAsyncTCP> 1.2.2 #1547686
|   |-- <ESP8266WiFi> 1.0
```

I tried the latest esphome 1.14.0 release as well as the current dev branch.

In the meantime I found the following issue and PR at the ESPAsyncTCP repo:
https://github.com/me-no-dev/ESPAsyncTCP/pull/129#issue-326553487
https://github.com/me-no-dev/ESPAsyncTCP/issues/131

This solves the compilation issues when replacing the files in the ESPAsyncTCP-esphome. 

But this leads to an issue with the MQTT connection, solved by this PR for the AsyncMqttClient lib:
https://github.com/marvinroger/async-mqtt-client/pull/165

Now encrypted MQTT is finally working! :)

Are the "-esphome" versions of these libraries managed by you? Shall I file additional issues in the corresponding repos to merge the PRs?

Thanks for your help!

moritzj29, 2019-11-02:
Unfortunately, the logs are not displayed. Apparently esphome cannot connect to the Broker: 
```
INFO Reading configuration ./ESPHotWater.yaml...
INFO Starting log output from esphotwater/debug
ERROR Cannot connect to MQTT broker: Int or String expected
```
Connecting with an MQTT Explorer to my broker indicates that the flashed device is working fine and MQTT messages are emitted. Other devices without SSL are working fine, the logs are displayed in esphome dashboard/CLI. 

Does the logging not work with SSL enabled?

OttoWinter, 2019-11-04:
> Are the "-esphome" versions of these libraries managed by you? Shall I file additional issues in the corresponding repos to merge the PRs?

Yes, that would be good. You can find the forks on my GitHub account (https://github.com/OttoWinter/ESPAsyncTCP). I think the async-mqtt-client library has gone stale, at least there's not been an update in years I believe. Plus we had some issues with some of the libraries and had to add some small patches - that's why there are those forks.

moritzj29, 2019-11-05:
I filed 2 PRs in the corresponding repos. So I guess this issue can be closed here, as it is not caused by esphome directly.

closed 2019-11-05 by moritzj29