esphome/issues #788: Pulse Counter stopped to work on beta 1.14.0b4

diraimondo, 2019-10-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hass.io: bug present in actual beta 1.14.0b4 but also on dev channel

**ESP (ESP32/ESP8266, Board/Sonoff):** Olimex ESP32-POE

I used since long time a photo-sensor on GPIO04 to count the pulse from my central energy meter (see cookbook: https://esphome.io/cookbook/power_meter.html )

Yesterday I upgraded to the long-time waited coming release (for my Yashima climate componente but also for the Xiaomi ClearGlass support) but my main ethernet wired ESP32 stopped to report the energy counts. Rolling back to stable channel (recompiling the firmware) it works again. I doesn't work even with the dev channel.

**Problem-relevant YAML-configuration entries:**
```yaml
# Olimex ESP32-POE

esphome:
  name: esp_entrance
  platform: ESP32
  board: esp-wrover-kit
  
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12

logger:

ota:
  password: 'XXX'

api:
  password: 'XXX'
  
esp32_ble_tracker:

time:
  - platform: homeassistant
    id: homeassistant_time
    
sensor:
  - platform: pulse_counter
    pin: GPIO4
    unit_of_measurement: 'kW'
    name: "Consumo Elettrico Istantaneo"
    id: power_usage_sensor
    filters:
      - multiply: 0.06
  - platform: total_daily_energy
    power_id: power_usage_sensor
    name: "Consumo Elettrico Giornaliero"
    id: daily_power_usage_sensor
```

**Logs (if applicable):**

```
[16:41:27][C][logger:175]: Logger:
[16:41:27][C][logger:176]:   Level: DEBUG
[16:41:27][C][logger:177]:   Log Baud Rate: 115200
[16:41:27][C][logger:178]:   Hardware UART: UART0
[16:41:27][C][esp32_ble_tracker:443]: BLE Tracker:
[16:41:27][C][esp32_ble_tracker:444]:   Scan Duration: 300 s
[16:41:27][C][esp32_ble_tracker:445]:   Scan Interval: 512 ms
[16:41:27][C][esp32_ble_tracker:446]:   Scan Window: 320 ms
[16:41:27][C][esp32_ble_tracker:447]:   Scan Type: ACTIVE
[16:41:27][C][pulse_counter:147]: Pulse Counter 'Consumo Elettrico Istantaneo'
[16:41:27][C][pulse_counter:147]:   Unit of Measurement: 'kW'
[16:41:27][C][pulse_counter:147]:   Accuracy Decimals: 2
[16:41:27][C][pulse_counter:147]:   Icon: 'mdi:pulse'
[16:41:27][C][pulse_counter:148]:   Pin: GPIO4 (Mode: INPUT)
[16:41:27][C][pulse_counter:149]:   Rising Edge: INCREMENT
[16:41:27][C][pulse_counter:150]:   Falling Edge: DISABLE
[16:41:27][C][pulse_counter:151]:   Filtering pulses shorter than 13 µs
[16:41:27][C][total_daily_energy:022]: Total Daily Energy 'Consumo Elettrico Giornaliero'
[16:41:27][C][total_daily_energy:022]:   Unit of Measurement: 'kWh'
[16:41:27][C][total_daily_energy:022]:   Accuracy Decimals: 4
[16:41:27][C][total_daily_energy:022]:   Icon: 'mdi:pulse'
[16:41:27][C][homeassistant.time:010]: Home Assistant Time:
[16:41:27][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.5.0/2,M10.4.0/3'
[16:41:27][C][ethernet:066]: Ethernet:
[16:41:27][C][ethernet:220]:   IP Address: 192.168.200.39
[16:41:27][C][ethernet:221]:   Hostname: 'esp_entrance'
[16:41:27][C][ethernet:222]:   Subnet: 255.255.255.0
[16:41:27][C][ethernet:223]:   Gateway: 192.168.200.1
[16:41:27][C][ethernet:226]:   DNS1: 192.168.200.1
[16:41:27][C][ethernet:228]:   DNS2: 0.0.0.0
[16:41:27][C][ethernet:231]:   MAC Address: 3C:71:BF:0C:8E:43
[16:41:27][C][ethernet:232]:   Is Full Duplex: YES
[16:41:27][C][ethernet:233]:   Link Up: YES
[16:41:27][C][ethernet:234]:   Link Speed: 100
[16:41:27][C][ethernet:068]:   Power Pin: GPIO12 (Mode: OUTPUT)
[16:41:27][C][ethernet:069]:   MDC Pin: 23
[16:41:27][C][ethernet:070]:   MDIO Pin: 18
[16:41:27][C][ethernet:071]:   Type: LAN8720
[16:41:27][C][ota:029]: Over-The-Air Updates:
[16:41:27][C][ota:030]:   Address: esp_entrance.local:3232
[16:41:27][C][ota:032]:   Using Password.
[16:41:27][C][api:095]: API Server:
[16:41:27][C][api:096]:   Address: esp_entrance.local:6053
[16:41:39][D][api.connection:578]: Client 'Home Assistant 0.101.0 (192.168.200.45)' connected successfully!
[16:41:39][D][time:029]: Synchronized time: Thu Oct 31 16:41:39 2019
[16:42:57][I][ota:046]: Boot seems successful, resetting boot loop counter.
[16:45:58][D][esp32_ble_tracker:148]: Starting scan...
[16:45:58][D][esp32_ble_tracker:457]: Found device 46:A6:F6:57:BF:A4 RSSI=-77
[16:45:58][D][esp32_ble_tracker:478]:   Address Type: RANDOM
[16:45:58][D][esp32_ble_tracker:457]: Found device 47:97:FF:AF:CF:6B RSSI=-63
[16:45:58][D][esp32_ble_tracker:478]:   Address Type: RANDOM
[16:45:58][D][esp32_ble_tracker:457]: Found device 53:92:72:13:4D:9F RSSI=-89
[16:45:58][D][esp32_ble_tracker:478]:   Address Type: RANDOM
[16:50:12][D][esp32_ble_tracker:457]: Found device 69:89:D9:62:F2:09 RSSI=-72
[16:50:12][D][esp32_ble_tracker:478]:   Address Type: RANDOM
```

**Additional information and things you've tried:**

diraimondo, 2019-10-31:
This is the output of the same yaml above using the stable release:
```
[17:06:49][C][logger:137]: Logger:
[17:06:49][C][logger:138]:   Level: DEBUG
[17:06:49][C][logger:139]:   Log Baud Rate: 115200
[17:06:49][C][logger:140]:   Hardware UART: UART0
[17:06:49][C][esp32_ble_tracker:459]: BLE Tracker:
[17:06:49][C][esp32_ble_tracker:460]:   Scan Interval: 300 s
[17:06:49][C][pulse_counter:147]: Pulse Counter 'Consumo Elettrico Istantaneo'
[17:06:49][C][pulse_counter:147]:   Unit of Measurement: 'kW'
[17:06:49][C][pulse_counter:147]:   Accuracy Decimals: 2
[17:06:49][C][pulse_counter:147]:   Icon: 'mdi:pulse'
[17:06:49][C][pulse_counter:148]:   Pin: GPIO4 (Mode: INPUT)
[17:06:49][C][pulse_counter:149]:   Rising Edge: INCREMENT
[17:06:49][C][pulse_counter:150]:   Falling Edge: DISABLE
[17:06:49][C][pulse_counter:151]:   Filtering pulses shorter than 13 µs
[17:06:49][C][total_daily_energy:022]: Total Daily Energy 'Consumo Elettrico Giornaliero'
[17:06:49][C][total_daily_energy:022]:   Unit of Measurement: 'kWh'
[17:06:49][C][total_daily_energy:022]:   Accuracy Decimals: 4
[17:06:49][C][total_daily_energy:022]:   Icon: 'mdi:pulse'
[17:06:49][C][homeassistant.time:010]: Home Assistant Time:
[17:06:49][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.5.0/2,M10.4.0/3'
[17:06:49][C][ethernet:066]: Ethernet:
[17:06:49][C][ethernet:220]:   IP Address: 192.168.200.39
[17:06:49][C][ethernet:221]:   Hostname: 'esp_entrance'
[17:06:49][C][ethernet:222]:   Subnet: 255.255.255.0
[17:06:49][C][ethernet:223]:   Gateway: 192.168.200.1
[17:06:49][C][ethernet:226]:   DNS1: 192.168.200.1
[17:06:49][C][ethernet:228]:   DNS2: 0.0.0.0
[17:06:49][C][ethernet:231]:   MAC Address: 3C:71:BF:0C:8E:43
[17:06:49][C][ethernet:232]:   Is Full Duplex: YES
[17:06:49][C][ethernet:233]:   Link Up: YES
[17:06:49][C][ethernet:234]:   Link Speed: 100
[17:06:50][C][ethernet:068]:   Power Pin: GPIO12 (Mode: OUTPUT)
[17:06:50][C][ethernet:069]:   MDC Pin: 23
[17:06:50][C][ethernet:070]:   MDIO Pin: 18
[17:06:50][C][ethernet:071]:   Type: LAN8720
[17:06:50][C][ota:029]: Over-The-Air Updates:
[17:06:50][C][ota:030]:   Address: esp_entrance.local:3232
[17:06:50][C][ota:032]:   Using Password.
[17:06:50][C][api:103]: API Server:
[17:06:50][C][api:104]:   Address: esp_entrance.local:6053
[17:06:53][D][api:573]: Client 'Home Assistant 0.101.0 (192.168.200.45)' connected successfully!
[17:06:53][D][time:030]: Synchronized time: Thu Oct 31 17:06:53 2019
[17:07:07][D][pulse_counter:158]: 'Consumo Elettrico Istantaneo': Retrieved counter: 2.00 pulses/min
[17:07:07][D][sensor:092]: 'Consumo Elettrico Istantaneo': Sending state 0.12000 kW with 2 decimals of accuracy
[17:07:07][D][sensor:092]: 'Consumo Elettrico Giornaliero': Sending state 0.00139 kWh with 4 decimals of accuracy
[17:08:07][D][pulse_counter:158]: 'Consumo Elettrico Istantaneo': Retrieved counter: 4.00 pulses/min
[17:08:07][D][sensor:092]: 'Consumo Elettrico Istantaneo': Sending state 0.24000 kW with 2 decimals of accuracy
[17:08:07][D][sensor:092]: 'Consumo Elettrico Giornaliero': Sending state 0.00539 kWh with 4 decimals of accuracy
[17:08:24][I][ota:046]: Boot seems successful, resetting boot loop counter.
```

OttoWinter, 2019-10-31:
I can't reproduce this with a pulse_counter sensor connected to a push button.

I think you may not be running 1.14.0b4 - in betas before there was a small issue with the scheduler that *may* have resulted in this behavior.

But your logs have the top line about the esphome version cut off - please post that line.

diraimondo, 2019-10-31:
Sorry, I cut away former lines misled by different color on the console: here the full log of the buggy behavior:
```
INFO Connecting to esp_entrance.local:6053 (192.168.200.39)
INFO Successfully connected to esp_entrance.local
[18:04:36][I][app:100]: esphome version 1.14.0b4 compiled on Oct 31 2019, 18:00:41
[18:04:36][C][logger:175]: Logger:
[18:04:36][C][logger:176]:   Level: DEBUG
[18:04:36][C][logger:177]:   Log Baud Rate: 115200
[18:04:36][C][logger:178]:   Hardware UART: UART0
[18:04:36][C][esp32_ble_tracker:442]: BLE Tracker:
[18:04:36][C][esp32_ble_tracker:443]:   Scan Duration: 300 s
[18:04:36][C][esp32_ble_tracker:444]:   Scan Interval: 512 ms
[18:04:36][C][esp32_ble_tracker:445]:   Scan Window: 320 ms
[18:04:36][C][esp32_ble_tracker:446]:   Scan Type: ACTIVE
[18:04:36][C][pulse_counter:147]: Pulse Counter 'Consumo Elettrico Istantaneo'
[18:04:36][C][pulse_counter:147]:   Unit of Measurement: 'kW'
[18:04:36][C][pulse_counter:147]:   Accuracy Decimals: 2
[18:04:36][C][pulse_counter:147]:   Icon: 'mdi:pulse'
[18:04:36][C][pulse_counter:148]:   Pin: GPIO4 (Mode: INPUT)
[18:04:36][C][pulse_counter:149]:   Rising Edge: INCREMENT
[18:04:36][C][pulse_counter:150]:   Falling Edge: DISABLE
[18:04:36][C][pulse_counter:151]:   Filtering pulses shorter than 13 µs
[18:04:36][C][total_daily_energy:022]: Total Daily Energy 'Consumo Elettrico Giornaliero'
[18:04:36][C][total_daily_energy:022]:   Unit of Measurement: 'kWh'
[18:04:37][C][total_daily_energy:022]:   Accuracy Decimals: 4
[18:04:37][C][total_daily_energy:022]:   Icon: 'mdi:pulse'
[18:04:37][C][homeassistant.time:010]: Home Assistant Time:
[18:04:37][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.5.0/2,M10.4.0/3'
[18:04:37][C][ethernet:066]: Ethernet:
[18:04:37][C][ethernet:220]:   IP Address: 192.168.200.39
[18:04:37][C][ethernet:221]:   Hostname: 'esp_entrance'
[18:04:37][C][ethernet:222]:   Subnet: 255.255.255.0
[18:04:37][C][ethernet:223]:   Gateway: 192.168.200.1
[18:04:37][C][ethernet:226]:   DNS1: 192.168.200.1
[18:04:37][C][ethernet:228]:   DNS2: 0.0.0.0
[18:04:37][C][ethernet:231]:   MAC Address: 3C:71:BF:0C:8E:43
[18:04:37][C][ethernet:232]:   Is Full Duplex: YES
[18:04:37][C][ethernet:233]:   Link Up: YES
[18:04:37][C][ethernet:234]:   Link Speed: 100
[18:04:37][C][ethernet:068]:   Power Pin: GPIO12 (Mode: OUTPUT)
[18:04:37][C][ethernet:069]:   MDC Pin: 23
[18:04:37][C][ethernet:070]:   MDIO Pin: 18
[18:04:37][C][ethernet:071]:   Type: LAN8720
[18:04:37][C][ota:029]: Over-The-Air Updates:
[18:04:37][C][ota:030]:   Address: esp_entrance.local:3232
[18:04:37][C][ota:032]:   Using Password.
[18:04:37][C][api:095]: API Server:
[18:04:37][C][api:096]:   Address: esp_entrance.local:6053
[18:04:38][D][api.connection:578]: Client 'Home Assistant 0.101.0 (192.168.200.45)' connected successfully!
[18:04:38][D][time:029]: Synchronized time: Thu Oct 31 18:04:38 2019
[18:05:12][D][esp32_ble_tracker:456]: Found device 79:CB:92:01:EA:40 RSSI=-77
[18:05:12][D][esp32_ble_tracker:477]:   Address Type: RANDOM
[18:06:04][I][ota:046]: Boot seems successful, resetting boot loop counter.
```

closed 2019-10-31 by OttoWinter

diraimondo, 2019-11-01:
I can confirm that last beta fixed my issue.