esphome/issues #790: ESPHome 1.14 Beta 4 - not working?

rdehuyss, 2019-11-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, esp32-poe

**Affected component:**

ble_presence
xiaomi_miflora

**Description of problem:**
Does not validate anymore

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: kitchen
  platform: ESP32
  board: esp32-poe

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
  manual_ip:
    static_ip: 192.168.1.215
    gateway: 192.168.1.1
    subnet: 255.255.255.0

esp32_ble_tracker:
  scan_interval: 180s

i2c:
  sda: 13
  scl: 16
  scan: True

binary_sensor:
  - platform: gpio
    pin: 0
    name: "Kitchen Movement"
    device_class: motion
  - platform: ble_presence
    mac_address: FF:FF:C5:14:DC:51
    name: "Ronald BLE Sleutelhanger"
  - platform: ble_presence
    service_uuid: '2b9e6a0a-1492-4174-ac06-267bbb4bb050'
    name: "Ronald BLE GSM"
  - platform: ble_presence
    mac_address: FF:FF:C5:14:DC:58
    name: "Evelien BLE Sleutelhanger"

sensor:
  - platform: bme680
    temperature:
      name: "Kitchen Temperature"
      filters:
        - offset: -2.5
    pressure:
      name: "Kitchen Pressure"
    humidity:
      name: "Kitchen Humidity"
    gas_resistance:
      name: "Kitchen Gas Resistance"
    address: 0x77
    iir_filter: 3x
    update_interval: 60s
  - platform: adc
    pin: 39
    name: "Kitchen Brightness"
    attenuation: 11db
    update_interval: 60s
  - platform: xiaomi_miflora
    mac_address: 'C4:7C:8D:6A:D7:BF'
    temperature:
      name: "Basil Temperature"
    moisture:
      name: "Basil Moisture"
    illuminance:
      name: "Basil Illuminance"
    conductivity:
      name: "Basil Soil Conductivity"
  - platform: xiaomi_miflora
    mac_address: 'C4:7C:8D:6A:D8:EF'
    temperature:
      name: "Coriander Temperature"
    moisture:
      name: "Coriander Moisture"
    illuminance:
      name: "Coriander Illuminance"
    conductivity:
      name: "Coriander Soil Conductivity"
    # battery_level:
    #   name: "Coriander Battery Level"

# Enable Home Assistant API
#api:
mqtt:
  broker: 192.168.1.250
  username: hasshome
  password: S8HYDcQaGG

# Enable logging
logger:

ota:

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Reading configuration /config/kitchen.yaml...
Failed config

esp32_ble_tracker: [source /config/kitchen.yaml:27]
  
  This option has been removed in 1.14 (Reason: it never had an effect).
  scan_interval: 180s [source /config/kitchen.yaml:27]
binary_sensor.ble_presence: [source /config/kitchen.yaml:42]
  
  'mac_address' is a required option for [binary_sensor.ble_presence].
  platform: ble_presence
  
  [service_uuid] is an invalid option for [binary_sensor.ble_presence]. Please check the indentation.
  service_uuid: 2b9e6a0a-1492-4174-ac06-267bbb4bb050 [source /config/kitchen.yaml:43]
  name: Ronald BLE GSM
sensor.xiaomi_miflora: [source /config/kitchen.yaml:69]
  
  This sensor has been renamed to xiaomi_hhccjcy01.
  platform: xiaomi_miflora
  mac_address: C4:7C:8D:6A:D7:BF
  temperature: 
    name: Basil Temperature
  moisture: 
    name: Basil Moisture
  illuminance: 
    name: Basil Illuminance
  conductivity: 
    name: Basil Soil Conductivity
sensor.xiaomi_miflora: [source /config/kitchen.yaml:79]
  
  This sensor has been renamed to xiaomi_hhccjcy01.
  platform: xiaomi_miflora
  mac_address: C4:7C:8D:6A:D8:EF
  temperature: 
    name: Coriander Temperature
  moisture: 
    name: Coriander Moisture
  illuminance: 
    name: Coriander Illuminance
  conductivity: 
    name: Coriander Soil Conductivit
```

**Additional information and things you've tried:**

OttoWinter, 2019-11-01:
> Does not validate anymore

I don't think it could have validated before, `service_uuid` has not been an option before.

Also, the changelog has not been fully written/published yet, so the breaking changes have not been brought documented together in the changelog. The messages however document them clearly:

`This option has been removed in 1.14 (Reason: it never had an effect).`
`This sensor has been renamed to xiaomi_hhccjcy01.`

Additionally, `service_uuid` is *not* part of 1.14 - it hasn't been tagged into the 1.14 milestone and has not been cherry-picked into the beta.

closed 2019-11-01 by OttoWinter

rdehuyss, 2019-11-01:
Hi Otto, thanks for your Swift response... Which version has the change with the ble tracking based on UUID? Since the merge got accepted, there was this release so I assumed it would be in there. 

Thanks again for this great tool!

OttoWinter, 2019-11-01:
If it's not tagged into 1.14 it will be on 1.15 dev