esphome/issues #792: ESP8266 DHT22 connection failing (Tried all listed solutions)

WMTaylor3, 2019-11-01:
---
name: Bug report
about: Create a report to help ESPHome improve

---

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10 running ESPHome 1.13.6
Macbook Air running ESPHome 1.13.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (ESP01_1M)
ESP8266 (Sparkfun Thing Dev)

**Affected component:**

https://esphome.io/components/sensor/dht.html

**Description of problem:**
### Problem:
Regardless of what I try, I continuously am only able to get the:
```
[22:35:01][W][dht:120]: Requesting data from DHT failed!
[22:35:01][W][dht:058]: Invalid readings! Please check your wiring (pull-up resistor, pin number) and consider manually specifying the DHT model using the model option.
```
error. I have tried every fix I can find online for this to no avail.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: totallyrandom
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "[Redacted]"
  password: "[Redacted]"

logger:

api:

ota:

sensor:
  - platform: dht
    pin: 2
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    update_interval: 10s
```

**Logs (if applicable):**

```
INFO Connecting to totallyrandom.local:6053 (xxx.xxx.xxx.229)
INFO Successfully connected to totallyrandom.local
[22:29:50][I][app:096]: esphome version 1.13.6 compiled on Nov  1 2019, 22:27:01
[22:29:50][C][wifi:372]: WiFi:
[22:29:50][C][wifi:254]:   SSID: '[Redacted]'
[22:29:50][C][wifi:255]:   IP Address: xx.xx.xx.229
[22:29:50][C][wifi:257]:   BSSID: xx:xx:xx:xx:xx:xx
[22:29:50][C][wifi:258]:   Hostname: 'totallyrandom'
[22:29:50][C][wifi:262]:   Signal strength: -71 dB ▂▄▆█
[22:29:50][C][wifi:263]:   Channel: 11
[22:29:50][C][wifi:264]:   Subnet: 255.255.255.0
[... 110 lines cut ...]
  }

  // Compute heat index in Fahrenheit (the default)
  float hif = dht.computeHeatIndex(f, h);
  // Compute heat index in Celsius (isFahreheit = false)
  float hic = dht.computeHeatIndex(t, h, false);

  Serial.print(F("Humidity: "));
  Serial.print(h);
  Serial.print(F("%  Temperature: "));
  Serial.print(t);
  Serial.print(F("°C "));
  Serial.print(f);
  Serial.print(F("°F  Heat index: "));
  Serial.print(hic);
  Serial.print(F("°C "));
  Serial.print(hif);
  Serial.println(F("°F"));
}
```

Apologies for the information overload, I figure if there's any chance of solving this, it's with all the info I can provide, I'm happy to test anything out that you guys suggest, post any code, anything. If I sound desperate, it's cause I am. I've been at this for what feels like forever.

Thank you in advance,
William

nbergont, 2019-11-08:
I have notice that DHT22 sensor need power reboot (remove power supply) after flashing or OTA for working properly.

grischard, 2019-11-17:
I'm seeing the same behaviour even after a power reboot.

grischard, 2019-11-17:
Might be related to #239

WMTaylor3, 2019-11-17:
> I have notice that DHT22 sensor need power reboot (remove power supply) after flashing or OTA for working properly.

Hi, thanks for the suggestion unfortunately in this case that doesn't seem to be helping much :/ have power cycled on several occasions while switching between different power supplies to no avail.

WMTaylor3, 2019-11-17:
> I'm seeing the same behaviour even after a power reboot.

Interesting to see someone else having the same issue, to verify we are in the same boat, have you tried uploading a simple Arduino sketch to the device which reads from the DHT and prints the value to serial? That's what I did to verify that it wasn't an issue with the board or the wiring. Can send you the sketch if you like.

grischard, 2019-11-18:
I haven't - the Sonoff is near the ceiling and I don't have the hardware to flash it over serial here.

CheongKoo, 2019-12-01:
Hello,
My DHT sensor works. I found that I had to specify the model of the sensor in order for it to work. See below. Hopefully this will help.
```
sensor:
  platform: dht
  pin: D4
  model: AM2302
  temperature:
     name: "Study temp"
  humidity:
     name: "Study humidity"
  update_interval: 20s
```

WMTaylor3, 2019-12-01:
> Hello,
> My DHT sensor works. I found that I had to specify the model of the sensor in order for it to work. See below. Hopefully this will help.
>
> ```
> sensor:
>   platform: dht
>   pin: D4
>   model: [...]

CheongKoo,
Thanks for your reply and I'm glad your having success, unfortunately I've already tried the solution that worked for you (per the attempted resolutions section of my initial post) with no luck.

Nonetheless, I've since moved to my own custom Arduino code with reporting of temperatures using MQTT and have been running with complete success with that for the past few months so I'm happy with this matter being closed. If anyone else has the same issue and wants to give my solution a go, I'm happy to share the code.

Thanks all,
William

closed 2019-12-01 by WMTaylor3