esphome/issues #793: sun: on_sunrise and on_sunset at wrong time

dzikus, 2019-11-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

sun:

**Description of problem:**
on_sunrise is 1h before real sunrise and on_sunset is 1h after real sunset in my location

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: schody
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret domain

logger:

api:

ota:

sun:
  latitude: !secret home_latitude
  longitude: !secret home_longitude
  on_sunrise:
  - then:
    - light.turn_off:
        id: light1
  on_sunset:
  - then:
    - light.turn_on:
        id: light1
time:
  - platform: homeassistant

binary_sensor:
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic Schody button"
    on_press:
      - light.toggle: light1
  - platform: status
    name: "Sonoff Basic Schody Status"

output:
  - platform: gpio
    pin: GPIO12
    id: relay
  - platform: esp8266_pwm
    id: blue_led
    pin: GPIO13
    inverted: True

light:
  - platform: binary
    id: light1
    name: "Sonoff Basic Schody light"
    output: relay
  - platform: monochromatic
    name: "Sonoff Basic Schody Blue LED"
    output: blue_led
    internal: true
    id: blue_led_light

sensor:
  - platform: wifi_signal
    name: "Sonoff Basic Schody WiFi Signal"
    update_interval: 60s
  - platform: uptime
    name: "Sonoff Basic Schody Uptime"
    update_interval: 60s
  - platform: sun
    name: Sun Elevation
    type: elevation
    update_interval: 60s
  - platform: sun
    name: Sun Azimuth
    type: azimuth
    update_interval: 60s

text_sensor:
  - platform: sun
    name: Sun Next Sunrise
    type: sunrise
  - platform: sun
    name: Sun Next Sunset
    type: sunset
```

**Logs (if applicable):**

```
[14:00:15][D][text_sensor:015]: 'Sun Next Sunrise': Sending state '05:3X:XX'
[14:00:19][D][text_sensor:015]: 'Sun Next Sunset': Sending state '17:4X:XX'
```

**Additional information and things you've tried:**

Real sunrise on my location today is around 6:30, real sunset on my location today is around 16:30. I changed timezone to test it without success, I changed timesource to:
```yaml
  - platform: sntp
    id: sntp_time
    timezone: !secret timezone
    servers:
      - !secret home_ntp
```
also without success.

OttoWinter, 2019-11-01:
I can't replicate the issue with my setup.

Unfortunately, to debug this you will need to provide some information like timezone and location. If you don't want to use your own location, you can also choose a similar one that exhibits the same behavior.

Especially timezone would be useful, because ±1 hour sounds a lot like a timezone issue.

Please also check the time is synchronized properly - after startup you will see a message like "Synchronized time: ..." that displays the synchronized time.

dzikus, 2019-11-02:
```yaml
esphome:
  name: schody
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret domain

logger:

api:

ota:

sun:
  latitude: 50.2648
  longitude: 19.0237
  on_sunrise:
  - then:
    - light.turn_off:
        id: light1
  on_sunset:
  - then:
    - light.turn_on:
        id: light1

time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Warsaw
    servers:
      - !secret home_ntp
      - time.google.com
  - platform: homeassistant

binary_sensor:
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic Schody button"
    on_press:
      - light.toggle: light1
  - platform: status
    name: "Sonoff Basic Schody Status"

output:
  - platform: gpio
    pin: GPIO12
    id: relay
  - platform: esp8266_pwm
    id: blue_led
    pin: GPIO13
    inverted: True

light:
  - platform: binary
    id: light1
    name: "Sonoff Basic Schody light"
    output: relay
  - platform: monochromatic
    name: "Sonoff Basic Schody Blue LED"
    output: blue_led
    internal: true
    id: blue_led_light

sensor:
  - platform: wifi_signal
    name: "Sonoff Basic Schody WiFi Signal"
    update_interval: 60s
  - platform: uptime
    name: "Sonoff Basic Schody Uptime"
    update_interval: 60s
  - platform: sun
    name: Sun Elevation
    type: elevation
    update_interval: 60s
  - platform: sun
    name: Sun Azimuth
    type: azimuth
    update_interval: 60s

text_sensor:
  - platform: sun
    name: Sun Next Sunrise
    type: sunrise
  - platform: sun
    name: Sun Next Sunset
    type: sunset
```

```
[10:56:30][C][sntp:043]: SNTP Time:
[10:56:30][C][sntp:044]:   Server 1: '172.x.x.254'
[10:56:30][C][sntp:045]:   Server 2: 'time.google.com'
[10:56:30][C][sntp:046]:   Server 3: ''
[10:56:30][C][sntp:047]:   Timezone: 'CET-1CEST-2,M3.5.0/2,M10.4.0/3'

[10:56:32][D][time:030]: Synchronized time: Sat Nov  2 10:56:32 2019
[10:56:54][D][text_sensor:015]: 'Sun Next Sunrise': Sending state '05:38:22'
[10:56:59][D][text_sensor:015]: 'Sun Next Sunset': Sending state '17:49:25'
```

https://www.google.com/search?q=katowice+poland+sunrise+sunset+time

dzikus, 2019-11-02:
Esphome also display:
```
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 03/29/20 02:00:00 to 10/25/20 03:00:00
```
before compilation.

jumblies, 2019-11-04:
I confirm.  Discrepant sunrise time with same lat/long between HomeAssistant and ESPhome.

OttoWinter, 2019-11-06:
Ok, I figured it out. This happens when the sun declination is negative. There was an erroneous check that constrained sun declination in the range [0,1], but it should have been [-1,1]

dzikus, 2019-11-07:
I tested your changes and still there is something wrong. It is much closer to the real sunrise and sunset times but still other then in Google or HomeAssistant:
On ESPHome (with patch from 003326f2eb0f52e974325347bcc2e1d76d3c2283):
```
[08:40:24][D][text_sensor:015]: 'Sun Next Sunset': Sending state '16:28:13'
[08:40:26][D][text_sensor:015]: 'Sun Next Sunrise': Sending state '06:58:56'
```
On HA:
Zachód Za 7 godzin(y) 16:12
Wschód Za 22 godzin(y) 6:44
On Google https://www.google.com/search?q=katowice+sunset+sunrise+time+7.11.2019 :
06:44
16:11

I tested it on this location:
```yaml
latitude: 50.2648
longitude: 19.0237
```

OttoWinter, 2019-11-07:
Some discrepancy is to be expected - calculating the real position of the sun makes the calculations just way to complex.

However, that error was at most around ~2 minutes during my testing. So there may be another issue here.

dzikus, 2019-11-08:
I also tested it on my light switch, and
```yaml
  on_sunrise:
  - then:
    - light.turn_off:
        id: light1
    elevation: 0
  on_sunset:
  - then:
    - light.turn_on:
        id: light1
    elevation: 0
```
turned off the light at other times than text_sensor gives, today light.turn_off was at 7:06, sunrise according to HA was at 6:46 and ESPHome text_sensor gives:
```
[09:35:19][D][text_sensor:015]: 'Sun Next Sunrise': Sending state '07:00:21'
```

I checked also light.turn_on yesterday and it was at 16:22, sunset according to HA was at 16:16 and text_sensor claims that next sunset is at 16:27.

All times are not much different but different, discrepancy is more than 2 minutes in every case that i tested.

jumblies, 2019-11-08:
Same here.  Actually, My sunsets are still an hour off.  

As you say, some discrepancy is expected.  
**ESPHOME**
```

ESPHome 1.14.2 

[05:10:11][D][text_sensor:015]: 'Sun Next Sunset': Sending state '18:23:32'
[05:10:49][D][text_sensor:015]: 'Sun Next Sunrise': Sending state '06:14:47'

sun:
  latitude: 36.0726
  longitude: -79.7920
  # 36.0726° N, 79.7920° W
```
**HOMEASSISTANT**
```
Rising In 1 hour  6:46 AM
Setting In 12 hours  5:18 PM
```

 

@OttoWinter This must be incredibly frustrating for you because your project really works well.  Is there any more debug info we can provide to help?

OttoWinter, 2019-11-08:
@dzikus There's an issue with your config, you're setting elevation to 0°, but sunrise/set is measured at around -0.883° (elevation refers to center of sun disk, but sunrise is first moment the top of disk is visible).

@jumblies I don't think you have that patch - the patch is not in any release yet (will be in 1.14.3 once release though).

I tested with the values you guys gave me and there is indeed a larger discrepancy - with checking a lot of the intermediary values against wolframalpha searches, there's not really a bug here, it's indeed just a calculation error.

The current sun code is based on approximating the sun's position - the more complex formulas google etc are using are way too complex to include in this codebase.

Some smaller accuracy errors are introduced in a few places, adding up to that result:
 - Sun Declination (earth's orbit around the sun plus earth's axis tilt) is not the best approximation. There are betters ones that also include difference in radius over the years etc.
 - Altitude - a lot of online tools are also factoring in the altitude of the location.
 - Sun Dick - as mentioned, sunset/rise is defined by when the last/first part of the sun disk is visible. The current value used is `-0.883°`, but in reality there's some slight variation on a bunch of factors.

I think this error, while not great, is acceptable. The error you get just from the environmental conditions (mostly hot/cold air) can change sunset/rise by easily 5 minutes anyway. If someone wants to implement more complex approximations for the values described above, please do.

jumblies, 2019-11-08:
@OttoWinter thanks.  You are amazingly patient.  Will wait until changes pushed.

> Sun Dick - as mentioned, sunset/rise is defined by when the last/first part of the sun disk is visible. The current value used is -0.883°, but in reality there's some slight variation on a bunch of factors.

Typo just made my day

dzikus, 2019-11-11:
I corrected my config, now:
ESPHome (with patch and after elevation correction):
```
[21:32:28][D][text_sensor:015]: 'Sun Next Sunrise': Sending state '07:05:05'
[21:32:41][D][text_sensor:015]: 'Sun Next Sunset': Sending state '16:22:36'
```
ESPHome (without patch and after elevation correction):
```
[21:12:08][D][text_sensor:015]: 'Sun Next Sunrise': Sending state '05:38:38'
[21:12:19][D][text_sensor:015]: 'Sun Next Sunset': Sending state '17:49:39'
```
HA:
```
 6:51
16:05
```
ESPEasy:
```
%sunset% | 16:06 | 16%3a06
%sunrise% | 6:49 | 6%3a49
```

brandond, 2019-11-11:
ESPEasy does seem to be closer. The code for that is here and doesn't seem to be significantly more complicated than what ESPHome is currently doing:
https://github.com/letscontrolit/ESPEasy/blob/mega/src/TimeESPeasy.ino

OttoWinter, 2019-11-12:
@brandond Yeah I know - they are using a specialized equation that finds the sunrise/sunset time. But that's using a lot of "magic" constants that are hard to verify. Plus with that system it get's really hard to check "when does the sun cross this elevation in degrees" - the only thing it can answer is when the sun crosses 0°.

ESPHome uses a different option - it binary searches based on the calculated elevation value. That then depends on how accurate the elevation calculation is.

I have another theory though: I think most web services use that magic equation too. For example, I looked for the sunrise times for middle of winter/summer using wolframalpha. When I then plugged that time into a query "sun elevation at {time}", I got results varying by a many arcminutes - but that should be constant.

brandond, 2019-11-12:
So do you think esphome is closer, or the magic equation? Folks probably aren't too concerned with actual accuracy, just consistency with other systems. If I'm setting up automations using a mix of esphome, home assistant, and 3rd party systems (google/tuya/etc) I probably don't care if the center of the sun disk is exactly aligned with the horizon of the geoid when my lights come on - I just want them all to come on around the same time.

dzikus, 2020-04-05:
This isn’t fully patched. There is still drift as I showed in https://github.com/esphome/issues/issues/793#issuecomment-552604784

Sarabveer, 2020-10-16:
I have the same issue:

HomeAssistant:
```
next_setting: 2020-10-16T22:16:03+00:00
```
(Timezone is America/New_York, so 22:16 UT would be 18:16 local time)

ESPHome:
```
[13:25:19][D][text_sensor:015]: 'Maharaaj Ji Room Light Sunset': Sending state '18:29:57'
```

Sarabveer, 2020-10-16:
Tasmota's Sunset calculation is only ~1 minute off.

https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_09_timers.ino#L60

dzikus, 2021-02-15:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

This isn’t fully patched yet. There is still drift as I showed in[ #793 (comment)](https://github.com/esphome/issues/issues/793#issuecomment-552604784)

closed 2021-04-07 by OttoWinter

dzikus, 2021-05-20:
I tested it and it works much better than before. Right now it has few seconds drift to Google and HomeAssistant. It's working stable since few weeks.