esphome/issues #795: SenseAir S8 - Reading from UART timed out | Reading data from SenseAir failed

reinisb, 2019-11-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome 1.15.0-dev, Hass.io, Virtualbox

**ESP (ESP32/ESP8266, Board/Sonoff):**

[SparkFun ESP32 Thing](https://cdn.sparkfun.com/datasheets/Wireless/WiFi/ESP32ThingV1.pdf) (3.3v) + [SparkFun Logic Level Converter - Single Supply](https://www.sparkfun.com/products/14765) (3.3v -> 5v)

**Affected component:**

https://github.com/esphome/esphome/pull/705

**Description of problem:**
Today I tried setting up the SenseAir S8 sensor for the very first time, but I have not succeeded as the reading from the sensor always fails. I suppose the issue lies somewhere with UART. I would appreciate if you could please help. Thanks!

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sparkfun_node
  platform: ESP32
  board: esp32thing

wifi:
  ssid: "erased"
  password: "erased"

switch:
  - platform: gpio
    name: "The Blue LED"
    pin: 5

uart:
  rx_pin: 16
  tx_pin: 17
  baud_rate: 9600
  id: "mans_uart"
     
sensor:
  - platform: senseair
    co2:
      name: "SenseAir CO2 Value"
    update_interval: 10s
    uart_id: "mans_uart"

# Enable logging
logger:
  #level: VERBOSE
  # Disable UART Logging
  # baud_rate: 0

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
[19:28:10][I][app:100]: ESPHome version 1.15.0-dev compiled on Nov  1 2019, 19:27:28
[19:28:10][C][wifi:409]: WiFi:
[19:28:10][C][wifi:277]:   SSID: [redacted]
[19:28:10][C][wifi:278]:   IP Address: 10.99.99.106
[19:28:10][C][wifi:280]:   BSSID: [redacted]
[19:28:10][C][wifi:281]:   Hostname: 'sparkfun_node'
[19:28:10][C][wifi:285]:   Signal strength: -50 dB ▂▄▆█
[19:28:10][C][wifi:289]:   Channel: 9
[19:28:10][C][wifi:290]:   Subnet: 255.255.255.0
[19:28:10][C][wifi:291]:   Gateway: 10.99.99.100
[19:28:10][C][wifi:292]:   DNS1: 10.99.99.100
[19:28:10][C][wifi:293]:   DNS2: 0.0.0.0
[19:28:10][C][uart:040]: UART Bus:
[19:28:10][C][uart:042]:   TX Pin: GPIO17
[19:28:10][C][uart:045]:   RX Pin: GPIO16
[19:28:10][C][uart:047]:   Baud Rate: 9600 baud
[19:28:10][C][uart:048]:   Stop bits: 1
[19:28:10][C][switch.gpio:042]: GPIO Switch 'The Blue LED'
[19:28:10][C][switch.gpio:043]:   Pin: GPIO5 (Mode: OUTPUT)
[19:28:10][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[19:28:10][C][logger:175]: Logger:
[19:28:10][C][logger:176]:   Level: DEBUG
[19:28:10][C][logger:177]:   Log Baud Rate: 115200
[19:28:10][C][logger:178]:   Hardware UART: UART0
[19:28:10][C][senseair:074]: SenseAir:
[19:28:10][C][senseair:075]:   CO2 'SenseAir CO2 Value'
[19:28:10][C][senseair:075]:     Unit of Measurement: 'ppm'
[19:28:10][C][senseair:075]:     Accuracy Decimals: 0
[19:28:10][C][senseair:075]:     Icon: 'mdi:periodic-table-co2'
[19:28:10][C][ota:029]: Over-The-Air Updates:
[19:28:10][C][ota:030]:   Address: sparkfun_node.local:3232
[19:28:10][C][api:095]: API Server:
[19:28:10][C][api:096]:   Address: sparkfun_node.local:6053
[19:28:13][E][uart:101]: Reading from UART timed out at byte 8!
[19:28:13][W][senseair:015]: Reading data from SenseAir failed!
[19:28:23][E][uart:101]: Reading from UART timed out at byte 8!
[19:28:24][W][senseair:015]: Reading data from SenseAir failed!
[19:28:33][E][uart:101]: Reading from UART timed out at byte 8!
[19:28:33][W][senseair:015]: Reading data from SenseAir failed!
[19:28:43][E][uart:101]: Reading from UART timed out at byte 8!
[19:28:43][W][senseair:015]: Reading data from SenseAir failed!
[19:28:53][E][uart:101]: Reading from UART timed out at byte 8!
[19:28:53][W][senseair:015]: Reading data from SenseAir failed!
[19:29:03][E][uart:101]: Reading from UART timed out at byte 8!
[19:29:03][W][senseair:015]: Reading data from SenseAir failed!
[19:29:13][E][uart:101]: Reading from UART timed out at byte 8!
[19:29:13][W][senseair:015]: Reading data from SenseAir failed!
[19:29:23][E][uart:101]: Reading from UART timed out at byte 8!
[19:29:23][W][senseair:015]: Reading data from SenseAir failed!
[19:29:33][E][uart:101]: Reading from UART timed out at byte 8!
[19:29:33][W][senseair:015]: Reading data from SenseAir failed!
[19:29:43][E][uart:101]: Reading from UART timed out at byte 8!

```

**Additional information and things you've tried:**
The sensor light blinks every 10 seconds or so, hence the sensor is powered. I tried swapping rx_pin/rx_pin values, nothing changes, the same error.

PS GPIO Switch 'The Blue LED' works, hence I at least know that communication with the board works, but I suppose something is wrong with UART.

[image]

reinisb, 2019-11-06:
It is clear that the Senseair S8 LP sensor requires 5V power supply. Now, what about the logic pins (UART_RxD and UART_TxD) - do they require 5V as well? Or are they 3.3V?

OttoWinter, 2019-11-07:
Those will be 5V level too. Technically you should shift that to 3.3V for the ESP, but I've successfully used 5V on those pins already. You just have to be careful you don't mix up TX/RX.

reinisb, 2019-11-07:
@OttoWinter so you are saying that Senseair S8 LP has 5V both on power (G+) and logic pins (UART_RxD and UART_TxD). In that case, something is wrong with my logic converter. I have an ESP32 board with 5V power-out pin and 3.3v TX/RX logic pins. When I connect the 5V power pin along ground to the sensor, then the sensor powers on properly. If I then connect the 3.3V TX/RX pins to sensor's 5V(?) UART_RxD/UART_TxD pins then I can read sensor readings. So everything works.

Now, when I connect the 3.3V TX/TX pins via a 3.3V-to-5V logic level shifter to the UART_RxD/UART_TxD then I cannot read sensor readings. Can you please confirm that connecting ESP32 3.3V logic level TX/RX to the sensor via a 3.3V-to-5V logic level shifter work for you and you can read sensor readings?

OttoWinter, 2019-11-07:
It should work if the logic level conversion is bi-directional (or uni-directional but different for each pin).

I don't have that device here, so I can't test it. I can only tell you I successfully used an MH-Z19 with the same 5V setup.

reinisb, 2019-11-17:
I have the [SparkFun Logic Level Converter - Single Supply](https://www.sparkfun.com/products/14765), which seems to be bi-directional. I am pretty sure that the _Senseair S8 LP_  has TX/RX pins that operate at 3.3v (not 5v) - I would appreciate it if anyone could please confirm. Thanks!

OttoWinter, 2019-11-17:
@reinisb You can check that with a multimeter.

But as you noted, it works fine when connecting it directly, which means it's an issue with how you connected the device, not a software bug. I will therefore close this report - if a solution is found please update the docs.

closed 2019-11-17 by OttoWinter