esphome/issues #799: Xiaomi CGG1 - same model, different firmware?

Alex9779, 2019-11-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io and pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32dev and some derivates

**Affected component:**

https://esphome.io/components/sensor/xiaomi_cgg1.html

**Description of problem:**
I received four new CGG! devices from China some days ago, my first CGG1 was ordere on Amazon Germany.
Apparently those new ones are different, they had a slightly different packaging, unfortunately I am not able to read Chinese so no idea what the signs in color tags mean or if there is any different in labelling indicating that they are different.
The problem is they seem to send slightly different advertising packets.
My original sends this for example:
```
50.30.47.03.9B.8C.7A.10.34.2D.58.0D.10.04.DC.00.F7.01
```
while the new ones send this:
```
50.30.47.03.5C.DB.52.10.34.2D.58.06.10.02.EC.01.0A.10.01.5B
```
The header is identical `50.30.47.03`, then according to the Mijia protocol a counter, followed by the mac addresses (reversed) `8C.7A.10.34.2D.58` and `DB.52.10.34.2D.58`.
Then the type, `0D` is for temp and humidity, `06` just for humidity.
After that a fixed `10` and the the lenght of the values following, so `04` for the new and `02` for the old in this cases.
But the new ones have some more information added as you can see `01.0A.10.01.5B`, on the old one the message is over after the values.
The problem is that in the xiaomi_ble component we are checking for an expected length of the message calculated by some fixed values and the given length of the passed values and this check fails for the new devices.
For the moment I have no idea what that data could be.
It seems the first four added bytes are fixed, I checked two devices so far, but the last byte is different. I have no idead when it changes but I watched one device for five minutes where it has been the same, after removing the battery it changed one but consecutive changes kept it stady again...
So I just commented the check in the code and got results from the new devices parsed properly.
The question is how we are going to handle this. Just remove the check? Try to figure out what the values are for?
Maybe someone has an idea?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Alex9779, 2019-11-01:
Monitoring for some minutes now and it seems the new devices don't send decodable battery information with the current implementation too...

Alex9779, 2019-11-01:
Ok I played around a bit and it seem the last byte of that new message is the battery level... I tried a new battery and got `64` which is 100 (%) while a battery reporting 87% in the old device has then `57` in the message of the new one...

OttoWinter, 2019-11-01:
@Alex9779 Great research!

Yeah I think the way to fix it would be to make `parse_xiaomi_data_byte` be called in a loop until the data is fully read.

Something like this:

```cpp
  uint8_t raw_offset = is_lywsdcgq || is_cgg1 ? 11 : 12;

  // Byte 0: Type
  // Byte 1: Fixed 0x10
  // Byte 2: Length
  // Byte 3..3+len-1: data
  const uint8_t *raw_process = &raw[raw_offset];
  const uint8_t remaining_length = device.get_service_data().size() - raw_offset - 3;

  bool any_success = false;
  XiaomiParseResult result;
  while (true) {
    if (remaining_length < 3)
      // at least 3 bytes required
      break;
    const uint8_t datapoint_type = raw_process[0];
    const uint8_t datapoint_length = raw_process[2];
    if (remaining_length < 3 + datapoint_length)
      break;
    const uint8_t *datapoint_data = &raw_process[3];
    if (parse_xiaomi_data_byte(datapoint_type, datapoint_data, datapoint_length, result))
      any_success = true;
    raw_process += 3 + datapoint_length;
    remaining_length -= 3 + datapoint_length;
  }
  if (!any_success)
    return {};
```

Alex9779, 2019-11-02:
Hmmmm just had another look on the data and it seems more clear now.
0x0A defined the battery level, it seems they just append that now.
So your suggestion is good as there is the same scheme:
you get data type, then the fixed 0x10, then the length of the actual data/value in one byte.
This applies to that what the new devices submit it is just concatenated...
I will write something up and create an PR, then we can discuss is further...

Alex9779, 2019-11-02:
Just to add, I compared the marking on the devices which are embossed in the lid in front of the batter, the are 100% indentical, so we just face some different firmware here...

closed 2019-11-02 by OttoWinter