esphome/issues #800: ESP32 doesn't connect to Wi-Fi after update to 1.14.0

definitio, 2019-11-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Virtual environments with Python 2.7.17 and Python 3.7.4.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**
Wi-Fi doesn't connect and causes reboots.

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  password: !secret wifi_password
  channel: !secret wifi_channel
```

**Logs (if applicable):**

```
[13:04:23][W][wifi_esp32:329]: Event: Disconnected ssid='<my_ssid>' bssid=<my_bsid> reason=Auth Expired
[13:04:23][W][wifi:445]: Error while connecting to network.
[13:04:23][W][wifi:482]: Restarting WiFi adapter...
[13:04:25][W][wifi_esp32:329]: Event: Disconnected ssid='<my_ssid>' bssid=00:00:00:00:00:00 reason=AP Not Found
[13:04:25][W][wifi:458]: WiFi network can not be found anymore.
[13:04:25][W][wifi:482]: Restarting WiFi adapter...
[13:04:32][W][wifi_esp32:329]: Event: Disconnected ssid='<my_ssid>' bssid=<my_bsid> reason=Auth Expired
[13:04:32][W][wifi:445]: Error while connecting to network.
[13:04:32][W][wifi:482]: Restarting WiFi adapter...
[13:04:34][W][wifi_esp32:329]: Event: Disconnected ssid='<my_ssid>' bssid=<my_bsid> reason=Auth Expired
[13:04:34][W][wifi:445]: Error while connecting to network.
[13:04:34][W][wifi:482]: Restarting WiFi adapter...
[13:04:41][W][wifi_esp32:329]: Event: Disconnected ssid='<my_ssid>' bssid=00:00:00:00:00:00 reason=AP Not Found
[13:04:41][W][wifi:445]: Error while connecting to network.
[13:04:41][W][wifi:482]: Restarting WiFi adapter...
[13:04:43][W][wifi_esp32:329]: Event: Disconnected ssid='<my_ssid>' bssid=00:00:00:00:00:00 reason=AP Not Found
[13:04:43][W][wifi:458]: WiFi network can not be found anymore.
[13:04:43][W][wifi:482]: Restarting WiFi adapter...
[13:04:51][W][wifi_esp32:329]: Event: Disconnected ssid='<my_ssid>' bssid=00:00:00:00:00:00 reason=AP Not Found
[13:04:51][W][wifi:458]: WiFi network can not be found anymore.
[13:04:51][W][wifi:482]: Restarting WiFi adapter...
[13:04:52][W][wifi_esp32:329]: Event: Disconnected ssid='<my_ssid>' bssid=<my_bsid> reason=Auth Expired
[13:04:52][W][wifi:445]: Error while connecting to network.
[13:04:52][W][wifi:482]: Restarting WiFi adapter...
[13:05:00][W][wifi_esp32:329]: Event: Disconnected ssid='<my_ssid>' bssid=<my_bsid> reason=Auth Expired
[13:05:00][W][wifi:445]: Error while connecting to network.
[13:05:00][W][wifi:482]: Restarting WiFi adapter...
[13:05:00][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 2 restarts
[13:05:00]abort() was called at PC 0x401d4ebb on core 1
[13:05:00]
[13:05:00]Backtrace: 0x40092d8c:0x3ffcdf60 0x40092fbd:0x3ffcdf80 0x401d4ebb:0x3ffcdfa0 0x401d4f02:0x3ffcdfc0 0x401d3ea7:0x3ffcdfe0 0x401d3fc2:0x3ffce000 0x400d71ef:0x3ffce020 0x400d72eb:0x3ffce050 0x400d7352:0x3ffce070 0x400d7dfe:0x3ffce0d0 0x400d8f21:0x3ffce110 0x400d7aee:0x3ffce160 0x401f18f1:0x3ffce180 0x400e016d:0x3ffce1a0 0x400e926b:0x3ffce1f0 0x400f7a87:0x3ffcef80 0x4008f4a1:0x3ffcefa0
WARNING Found stack trace! Trying to decode it
[13:05:00]
[13:05:00]Rebooting...
[13:05:00]ets Jun  8 2016 00:22:57
[13:05:00]
[13:05:00]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:05:00]configsip: 0, SPIWP:0xee
[13:05:00]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:05:00]mode:DIO, clock div:2
[13:05:00]load:0x3fff0018,len:4
[13:05:00]load:0x3fff001c,len:1044
[13:05:00]load:0x40078000,len:8896
[13:05:00]load:0x40080400,len:5828
[13:05:00]entry 0x400806ac
```

**Additional information and things you've tried:**

OttoWinter, 2019-11-02:
Thanks for including the device logs in the PR template! Can't tell you how many times I need to ask for those 🙄 

Some thoughts:

 - This is ESP32, so it's not a case of https://github.com/esphome/issues/issues/455
 - I've had similar issues (as in `Auth Expired` on ESP32s) when connecting an ESP32 to a Android Hotspot. There I always needed to restart the hotspot to make the ESP32 reconnect again. That was due to some issue in esp-idf, *but* it existed before the 1.14 release too.
 - Might have to do with a change I pulled in from upstream Arduino: https://github.com/esphome/esphome/commit/21bbafb63df75fc64f8b3c5a7303634c253fd4c1
 - The `abort()` plus stacktrace are not a good sign - something is crashing somewhere. Odd though that the stack trace can't get decoded, could mean that all those addresses are from the precompiled esp-idf where no demangled names are available.

timon71, 2019-11-03:
Same issue here. ESP32 (Sonoff 01). Logs are similar. I reverted back to 1.13.

Haringstad, 2019-11-04:
Same issue. I have this issue with 2 of my Sonoff basic, and one ESP32 is acting up too. This one is jumping Online, Offline, with the same "reason=Auth Expired". It got more stable after I put sntp in the config, but not as stable as 1.13.

aqualx, 2019-11-04:
Similar issue. ESP32. Here is my log. Starting from flash down to crash.

```
========================= [SUCCESS] Took 60.28 seconds =========================
INFO Successfully compiled program.
INFO Running:  platformio run -d /config/esphome/esp32 -t upload --upload-port /dev/ttyUSB0
Processing esp32 (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.5
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient-esphome> 0.8.3
|   |-- <AsyncTCP> 1.1.1
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Update> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size /data/esp32/.pioenvs/esp32/firmware.elf
Checking size /data/esp32/.pioenvs/esp32/firmware.elf
DATA:    [==        ]  18.7% (used 61364 bytes from 327680 bytes)
PROGRAM: [========= ]  94.2% (used 1543894 bytes from 1638400 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, ftdi, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading /data/esp32/.pioenvs/esp32/firmware.bin
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: ***
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15872 bytes to 10319...
***
Writing at 0x000ec000... (100 %)
Wrote 1544016 bytes (902335 compressed) at 0x00010000 in 22.5 seconds (effective 550.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================= [SUCCESS] Took 32.26 seconds =========================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[12:34:36][I][logger:156]: Log initialized
[12:34:36][C][ota:364]: There have been 7 suspected unsuccessful boot attempts.
[12:34:36][I][app:028]: Running through setup()...
[12:34:36][C][wifi:033]: Setting up WiFi...
[12:34:36][D][wifi:298]: Starting scan...
[12:34:38][D][wifi:313]: Found networks:
[12:34:38][I][wifi:359]: - 'zorro' [redacted]▂▄▆█
[12:34:38][D][wifi:360]:     Channel: 7
[12:34:38][D][wifi:361]:     RSSI: -61 dB
[12:34:38][I][wifi:187]: WiFi Connecting to 'zorro'...
[12:34:40][W][wifi_esp32:329]: Event: Disconnected ssid='zorro' bssid=[redacted] reason=AP Not Found
[12:34:40][W][wifi:458]: WiFi network can not be found anymore.
[12:34:40][W][wifi:482]: Restarting WiFi adapter...
[... 75 lines cut ...]
WARNING Decoded 0x400d477d: esphome::mqtt::MQTTClientComponent::publish_json(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (ArduinoJson::JsonObject&)> const&, unsigned char, bool) at /config/esphome/esp32/src/esphome/components/mqtt/mqtt_client.cpp:465
WARNING Decoded 0x400d532f: esphome::mqtt::MQTTComponent::send_discovery_() at /config/esphome/esp32/src/esphome/components/mqtt/mqtt_component.cpp:150
WARNING Decoded 0x400d540b: esphome::mqtt::MQTTComponent::call_setup() at /config/esphome/esp32/src/esphome/components/mqtt/mqtt_component.cpp:150
WARNING Decoded 0x401ea479: esphome::Component::call() at /config/esphome/esp32/src/esphome/core/component.cpp:82
WARNING Decoded 0x400dde0d: esphome::Application::setup() at /config/esphome/esp32/src/esphome/core/application.cpp:37
WARNING Decoded 0x400e2b8a: setup() at /config/esphome/esp32/src/main.cpp:1089
WARNING Decoded 0x400f042b: loopTask(void*) at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:14
WARNING Decoded 0x4008e7dd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[12:35:06]
[12:35:06]Rebooting...
[12:35:06]ets Jun  8 2016 00:22:57
[12:35:06]
[12:35:06]rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[12:35:06]configsip: 0, SPIWP:0xee
[12:35:06]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:35:06]mode:DIO, clock div:2
[12:35:06]load:0x3fff0018,len:4
[12:35:06]load:0x3fff001c,len:1044
[12:35:06]load:0x40078000,len:8896
[12:35:06]load:0x40080400,len:5828
[12:35:06]entry 0x400806ac
[12:35:06][I][logger:156]: Log initialized
[12:35:06][C][ota:364]: There have been 8 suspected unsuccessful boot attempts.
[12:35:06][I][app:028]: Running through setup()...
```

OttoWinter, 2019-11-04:
@timon71 @Haringstad Those are ESP8266 devices. This issue is specifically about ESP32. See previous comment for ESP8266 thread.

----

@aqualx The logs show the ESP32 is running out of memory. Could you share what components you have enabled and what board that is? Also try `arduino_version: 1.0.2` in your core-config to use the esp-idf version that 1.13.6 used.

aqualx, 2019-11-05:
> @aqualx The logs show the ESP32 is running out of memory. Could you share what components you have enabled and what board that is? Also try `arduino_version: 1.0.2` in your core-config to use the esp-idf version that 1.13.6 used.

With this works fine:
```
esphome:
  name: esp32
  platform: ESP32
  board: esp-wrover-kit
  arduino_version: 1.0.2
```

I have: TTGO ESP32-WROVER 8MB PSRAM

```
Hardware Specifications
Chipset: ESPRESSIF-ESP32 240MHz
FLASH: 8MB PSRAM
SRAM: 520 kB SRAM
```
In configuration enabled:
* web_server
* logger
* mqtt
* esp32_ble_tracker
* platform: xiaomi_lywsdcgq, xiaomi_hhccjcy01, uptime, sntp

OttoWinter, 2019-11-05:
`esp-wrover-kit` I think that's the problem. The default WROVER only has 320kb of RAM, which can be too little with newer esp-idf versions. The compiler thus thinks it only has 320kb of ram. Try `m5stack-grey` (even if that's technically the correct board) - it has 512kb defined.

Also try the other thing I mentioned, arduino_version, please.

aqualx, 2019-11-06:
> `esp-wrover-kit` I think that's the problem. The default WROVER only has 320kb of RAM, which can be too little with newer esp-idf versions. The compiler thus thinks it only has 320kb of ram. Try `m5stack-grey` (even if that's technically the correct board) - it has 512kb defined.

With this configuration
```
esphome:
  name: esp32
  platform: ESP32
  board: m5stack-grey
```

Still get same crashes:
```
 [03:52:40][W][wifi_esp32:329]: Event: Disconnected ssid='zorro' bssid=[redacted] reason=Beacon Timeout
[03:52:40][W][wifi:094]: WiFi Connection lost... Reconnecting...
[03:52:40][W][wifi:482]: Restarting WiFi adapter...
[03:52:40][W][mqtt:247]: MQTT Disconnected: WiFi disconnected.
[03:52:45][D][wifi:298]: Starting scan...
[03:52:47][D][wifi:313]: Found networks:
[03:52:47][I][wifi:359]: - 'zorro' [redacted]▂▄▆█
[03:52:47][D][wifi:360]:     Channel: 7
[03:52:47][D][wifi:361]:     RSSI: -68 dB
[03:52:47][I][wifi:187]: WiFi Connecting to 'zorro'...
[03:52:48][I][wifi:417]: WiFi Connected!
[03:52:48][C][wifi:277]:   SSID: [redacted]
[03:52:48][C][wifi:278]:   IP Address: **
[03:52:48][C][wifi:280]:   BSSID: [redacted]
[03:52:48][C][wifi:281]:   Hostname: 'esp32'
[03:52:48][C][wifi:285]:   Signal strength: -67 dB ▂▄▆█
[03:52:48][C][wifi:289]:   Channel: 7
[03:52:48][C][wifi:290]:   Subnet: ***
[03:52:48][C][wifi:291]:   Gateway:***
[03:52:48][C][wifi:292]:   DNS1: ***
[03:52:48][C][wifi:293]:   DNS2: ***
[03:52:48][I][mqtt:162]: Connecting to MQTT...
[03:52:48][I][mqtt:202]: MQTT Connected!
[03:52:48]abort() was called at PC 0x401ced0f on core 1
[03:52:48]
[03:52:48]Backtrace: 0x40092094:0x3ffce580 0x400922c5:0x3ffce5a0 0x401ced0f:0x3ffce5c0 0x401ced56:0x3ffce5e0 0x401ce0fd:0x3ffce600 0x401cde08:0x3ffce620 0x401ce345:0x3ffce640 0x401ce462:0x3ffce660 0x401ce735:0x3ffce6a0 0x401ce7ae:0x3ffce6c0 0x400d553d:0x3ffce6e0 0x400d5662:0x3ffce7b0 0x400d591a:0x3ffce7f0 0x400d2f0f:0x3ffce8f0 0x400d37ba:0x3ffce910 0x400d47e9:0x3ffce940 0x400d539b:0x3ffce970 0x400d5410:0x3ffce9c0 0x401e9fb1:0x3ffce9e0 0x400dd08a:0x3ffcea00 0x400df3c6:0x3ffcea30 0x400f1489:0x3ffcea50 0x4008e7dd:0x3ffcea70
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /config/esphome/esp32 -t idedata
WARNING Decoded 0x40092094: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x400922c5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x401ced0f: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401ced56: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401ce0fd: __cxa_allocate_exception at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc:313
WARNING Decoded 0x401cde08: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
WARNING Decoded 0x401ce345: __gnu_cxx::new_allocator<char>::allocate(unsigned int, void const*) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h:281
 (inlined by) std::allocator_traits<std::allocator<char> >::allocate(std::allocator<char>&, unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/alloc_traits.h:360
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned int&, unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.tcc:157
WARNING Decoded 0x401ce462: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned int, unsigned int, char const*, unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h:281
WARNING Decoded 0x401ce735: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_append(char const*, unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h:281
WARNING Decoded 0x401ce7ae: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::append(char const*) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h:281
WARNING Decoded 0x400d553d: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > std::operator+<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, char const*) at /config/esphome/esp32/src/esphome/components/mqtt/mqtt_component.cpp:150
 (inlined by) esphome::mqtt::MQTTComponent::get_default_topic_for_(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const at /config/esphome/esp32/src/esphome/components/mqtt/mqtt_component.cpp:21
[... cut ...]
[03:52:52]load:0x3fff0018,len:4
[03:52:52]load:0x3fff001c,len:1044
[03:52:52]load:0x40078000,len:8896
[03:52:52]load:0x40080400,len:5828
[03:52:52]entry 0x400806ac
[03:52:52][I][logger:156]: Log initialized
[03:52:52][C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[03:52:52][I][app:028]: Running through setup()...
```
> Also try the other thing I mentioned, arduino_version, please.

Works only with:
```
esphome:
  name: esp32
  platform: ESP32
  board: esp-wrover-kit
  arduino_version: 1.0.2
```

OttoWinter, 2019-11-06:
Now that I looked at the original post again, your problem is not the one from @definitio - you're getting lack of memory errors, whereas it's "Auth Expired" for @definitio. So for your issue, I guess the solution is just to use the old esp-idf version with `arduino_version`. There's not much that can be done if the SDK uses more memory for BLE now. I will also hide the comments to keep this issue focused on the original issue. If you want, you can open a new issue to discuss this issue further.

aqualx, 2019-11-06:
@OttoWinter: In my first log you can see that I also had "Auth Expired" while connecting to wifi network.
All BLE/sensor messages skipped from my post intentionally. According to log BLE works fine. Crash is right after "WiFi Connected!". Until wifi connected everything works as expected according to log.

stripeyhorse, 2019-11-09:
im having the same issue. This node was working perfectly before the new 1.14.0 version.

Using a few sensors on the nodes as you can see..

LOG:
[0;36m[D][wifi:298]: Starting scan...[0m
[0;36m[D][esp-idf:000]: E (9734) wifi: esp_wifi_scan_start 1245 wifi not start[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -81.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -77.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -78.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -77.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -78.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -93.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][binary_sensor:036]: 'Downstairs PIR Sensor': Sending state OFF[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -82.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -82.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -80.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -82.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][tsl2561:126]: Got illuminance=29.6lx[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor Light': Sending state 29.60874 lx with 1 decimals of accuracy[0m
[0;36m[D][ccs811:077]: Got co2=65021 ppm, tvoc=65021 ppb, baseline=0xF4FF[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor eCO2 Value': Sending state 65021.00000 ppm with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor TVOC': Sending state 65021.00000 ppb with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -77.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -77.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -75.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -78.00000 dB with 0 decimals of accuracy[0m
[1;31m[E][wifi:306]: Scan timeout![0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -90.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -87.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Kens Wallet RSSI': Sending state -92.00000 dB with 0 decimals of accuracy[0m
[0;36m[D][wifi:298]: Starting scan...[0m
[0;36m[D][esp-idf:000]: E (44746) wifi: esp_wifi_scan_start 1245 wifi not start[0m
[0;36m[D][sht3xd:065]: Got temperature=17.01°C humidity=53.79%[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor Temperature': Sending state 17.01305 °C with 1 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor Humidity': Sending state 53.79110 % with 1 decimals of accuracy[0m

OttoWinter, 2019-11-09:
@stripeyhorse Looks like a separate issue too - your device fails to even start the scan. The log string `wifi not start` suggests that the wifi interface was not started - which would be a couple of lines higher in the log than you posted. I suspect there's an error message somewhere about the WiFi interface not being able to start because of some error in esp-idf. Could you post those lines too?

stripeyhorse, 2019-11-11:
SORRY... this is after i pressed the reset button.. 

I have a StripeyIOT network and nothing has changed between old version and new version with my network (UniFi Nano HDs x 2 with UniFi controller and USG)

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[0;32m[I][logger:156]: Log initialized[0m
[0;35m[C][status_led:014]: Setting up Status LED...[0m
[0;35m[C][ota:364]: There have been 4 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:028]: Running through setup()...[0m
[0;36m[D][binary_sensor:034]: 'Downstairs PIR Sensor': Sending initial state ON[0m
[0;36m[D][esp32_ble_tracker:148]: Starting scan...[0m
[0;36m[D][binary_sensor:034]: 'Downstairs Sensor Status': Sending initial state OFF[0m
[0;35m[C][tsl2561:018]: Setting up TSL2561...[0m
[0;35m[C][sht3xd:019]: Setting up SHT3xD...[0m
[0;35m[C][wifi:033]: Setting up WiFi...[0m
[0;36m[D][wifi:298]: Starting scan...[0m
[0;36m[D][ccs811:077]: Got co2=65021 ppm, tvoc=65021 ppb, baseline=0xF4FF[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor eCO2 Value': Sending state 65021.00000 ppm with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor TVOC': Sending state 65021.00000 ppb with 0 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Uptime Sensor': Sending state 0.82400 s with 0 decimals of accuracy[0m
[0;36m[D][binary_sensor:036]: 'Downstairs PIR Sensor': Sending state OFF[0m
[0;36m[D][esp32_ble_tracker:456]: Found device 32:38:5D:7D:F8:39 RSSI=-86[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM[0m
[0;36m[D][esp32_ble_tracker:456]: Found device 4C:B2:5F:D8:E1:9C RSSI=-60[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM[0m
[0;36m[D][esp32_ble_tracker:481]:   TX Power: 2[0m
[0;36m[D][sht3xd:065]: Got temperature=20.87°C humidity=46.81%[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor Temperature': Sending state 20.87167 °C with 1 decimals of accuracy[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor Humidity': Sending state 46.80857 % with 1 decimals of accuracy[0m
[0;36m[D][esp32_ble_tracker:456]: Found device C8:69:CD:43:2A:D6 RSSI=-82[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: PUBLIC[0m
[0;36m[D][esp32_ble_tracker:481]:   TX Power: 2[0m
[0;36m[D][esp32_ble_tracker:456]: Found device 4E:B7:6B:55:DF:A6 RSSI=-99[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM[0m
[0;36m[D][tsl2561:126]: Got illuminance=16.2lx[0m
[0;36m[D][sensor:092]: 'Downstairs Sensor Light': Sending state 16.21755 lx with 1 decimals of accuracy[0m
[0;36m[D][esp32_ble_tracker:456]: Found device 62:DA:06:A0:80:61 RSSI=-66[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM[0m
[0;36m[D][esp32_ble_tracker:456]: Found device 55:D2:F9:CC:16:F9 RSSI=-65[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM[0m
[0;36m[D][esp32_ble_tracker:481]:   TX Power: 2[0m
[0;36m[D][esp32_ble_tracker:456]: Found device 53:35:66:7E:06:27 RSSI=-78[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM[0m
[0;36m[D][esp32_ble_tracker:456]: Found device 26:A6:85:90:37:77 RSSI=-86[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM[0m
[0;36m[D][esp32_ble_tracker:456]: Found device 49:24:30:EF:F8:B0 RSSI=-73[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM[0m
[0;36m[D][esp32_ble_tracker:481]:   TX Power: 2[0m
[0;36m[D][esp32_ble_tracker:456]: Found device 25:28:24:A9:49:D2 RSSI=-78[0m
[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM[0m
[0;36m[D][wifi:313]: Found networks:[0m
[0;32m[I][wifi:359]: - 'StripeyIOT' [5m(FE:EC:DA:95:13:5B) [6m[0;33m▂▄▆[0;37m█[0m[0m
[0;36m[D][wifi:360]:     Channel: 11[0m
[0;36m[D][wifi:361]:     RSSI: -54 dB[0m
[0;32m[I][wifi:359]: - 'StripeyIOT' [5m(FE:EC:DA:90:05:A8) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:360]:     Channel: 6[0m
[0;36m[D][wifi:361]:     RSSI: -88 dB[0m
[0;36m[D][wifi:363]: - [5m'StripeyWIFI'[6m [5m(FC:EC:DA:D5:13:5B) [6m[0;33m▂▄▆[0;37m█[0m[0m
[0;36m[D][wifi:363]: - [5m'HP-Print-08-ENVY 4500 series'[6m [5m(28:80:23:EE:F3:08) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:363]: - [5m'StripeyWIFI'[6m [5m(FC:EC:DA:D0:05:A8) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;36m[D][wifi:363]: - [5m'TelstraC27F9B'[6m [5m(F8:AB:05:C2:7F:A1) [6m[0;31m▂[0;37m▄▆█[0m[0m
[0;32m[I][wifi:187]: WiFi Connecting to 'StripeyIOT'...[0m
[0;33m[W][wifi_esp32:329]: Event: Disconnected ssid='StripeyIOT' bssid=[5m00:00:00:00:00:00[6m reason=AP Not Found[0m
[0;33m[W][wifi:458]: WiFi network can not be found anymore.[0m
[0;33m[W][wifi:482]: Restarting WiFi adapter...[0m
[0;32m[I][wifi:187]: WiFi Connecting to 'StripeyIOT'...[0m
[0;36m[D][esp-idf:000]: E (4737) wifi: esp_wifi_disconnect 1153 wifi not start[0m
[1;31m[E][wifi:215]: wifi_sta_connect_ failed![0m

perrycharlton, 2019-11-12:
Same problem. Using QuinDeca's x 10 all were working fine. I had 3 on all the time, but since the upgrade, they keep disconnecting. There seems to be a trigger when I refresh the Web Page.
I have the same issue with all the units I reflashed

␛[0;36m[D][xiaomi_ble:161]:   Temperature: 20.6°C␛[0m
␛[0;36m[D][xiaomi_ble:164]:   Humidity: 38.5%␛[0m
abort() was called at PC 0x401d075f on core 1

Backtrace: 0x40092158:0x3fffa330 0x40092389:0x3fffa350 0x401d075f:0x3fffa370 0x401d07a6:0x3fffa390 0x401cfb39:0x3fffa3b0 0x401cf844:0x3fffa3d0 0x400eab65:0x3fffa3f0 0x400dde21:0x3fffa420 0x400ea4a5:0x3fffa470 0x400ea601:0x3fffa4a0 0x400ea681:0x3fffa4e0 0x401ec802:0x3fffa500 0x400ead75:0x3fffa520 0x400e7533:0x3fffa540 0x400e76b4:0x3fffa570 0x400e7d05:0x3fffa590 0x4008e8a1:0x3fffa5c0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
␛[0;32m[I][logger:156]: Log initialized␛[0m
␛[0;35m[C][ota:364]: There have been 1 suspected unsuccessful boot attempts.␛[0m
␛[0;32m[I][app:028]: Running through setup()...␛[0m
␛[0;35m[C][switch.gpio:011]: Setting up GPIO Switch 'Onboard LED'...␛[0m
␛[0;36m[D][switch:025]: 'Onboard LED' Turning OFF.␛[0m
␛[0;36m[D][switch:045]: 'Onboard LED': Sending state ON␛[0m
␛[0;36m[D][switch:025]: 'Onboard LED' Turning OFF.␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-1'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-1' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-2'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-2' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-3'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-3' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-4'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-4' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-5'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-5' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-6'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-6' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-7'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-7' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-8'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-8' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-9'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-9' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;35m[C][light:097]: Setting up light 'garage_desk_channel-10'...␛[0m
␛[0;36m[D][light:264]: 'garage_desk_channel-10' Setting:␛[0m
␛[0;36m[D][light:277]:   Brightness: 100%␛[0m
␛[0;36m[D][esp32_ble_tracker:148]: Starting scan...␛[0m
␛[0;35m[C][dallas.sensor:032]: Setting up DallasComponent...␛[0m
␛[0;35m[C][wifi:033]: Setting up WiFi...␛[0m
␛[0;32m[I][wifi:187]: WiFi Connecting to 'Ron2.4G'...␛[0m
␛[1;31m[E][dallas.sensor:123]: Requesting conversion failed␛[0m
␛[0;36m[D][esp32_ble_tracker:456]: Found device 5D:8A:21:D9:1B:C8 RSSI=-67␛[0m
␛[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM␛[0m
␛[0;36m[D][esp32_ble_tracker:456]: Found device 57:97:5A:D1:EA:1B RSSI=-76␛[0m
␛[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM␛[0m
␛[0;36m[D][esp32_ble_tracker:456]: Found device 5B:0F:05:A2:B1:99 RSSI=-89␛[0m
␛[0;36m[D][esp32_ble_tracker:477]:   Address Type: RANDOM␛[0m
␛[0;36m[D][esp32_ble_tracker:456]: Found device 6E:F2:1C:AC:AE:CF RSSI=-80␛[0m
[... 213 lines cut ...]
␛[0;35m[C][dallas.sensor:085]:   Device 'Q1_tempsensor'␛[0m
␛[0;35m[C][dallas.sensor:085]:     Unit of Measurement: '°C'␛[0m
␛[0;35m[C][dallas.sensor:085]:     Accuracy Decimals: 1␛[0m
␛[0;35m[C][dallas.sensor:085]:     Icon: 'mdi:thermometer'␛[0m
␛[0;35m[C][dallas.sensor:087]:     Index 0␛[0m
␛[1;31m[E][dallas.sensor:089]: Couldn't find sensor by index - not connected. Proceeding without it.␛[0m
␛[0;35m[C][web_server:120]: Web Server:␛[0m
␛[0;35m[C][web_server:121]:   Address: 10.0.4.52:80␛[0m
␛[0;35m[C][ota:029]: Over-The-Air Updates:␛[0m
␛[0;35m[C][ota:030]:   Address: 10.0.4.52:3232␛[0m
␛[0;35m[C][api:095]: API Server:␛[0m
␛[0;35m[C][api:096]:   Address: 10.0.4.52:6053␛[0m
␛[0;33m[W][wifi_esp32:329]: Event: Disconnected ssid='Ron2.4G' bssid=␛[5m72:FF:7B:2A:6C:F0␛[6m reason=Mic Failure␛[0m
␛[0;33m[W][wifi:094]: WiFi Connection lost... Reconnecting...␛[0m
␛[0;36m[D][xiaomi_ble:158]: Got Xiaomi LYWSDCGQ (58:2D:34:32:2D:68):␛[0m
␛[0;36m[D][xiaomi_ble:164]:   Humidity: 38.4%␛[0m
␛[0;32m[I][wifi:187]: WiFi Connecting to 'Ron2.4G'...␛[0m
␛[0;33m[W][wifi_esp32:329]: Event: Disconnected ssid='Ron2.4G' bssid=␛[5mA2:DA:C4:66:97:A8␛[6m reason=Auth Expired␛[0m
␛[0;33m[W][wifi:445]: Error while connecting to network.␛[0m
␛[0;33m[W][wifi:482]: Restarting WiFi adapter...␛[0m
␛[0;32m[I][wifi:187]: WiFi Connecting to 'Ron2.4G'...␛[0m
␛[0;36m[D][xiaomi_ble:158]: Got Xiaomi LYWSDCGQ (58:2D:34:32:2D:68):␛[0m
␛[0;36m[D][xiaomi_ble:161]:   Temperature: 20.6°C␛[0m
␛[0;36m[D][xiaomi_ble:164]:   Humidity: 38.6%␛[0m
␛[0;36m[D][xiaomi_ble:158]: Got Xiaomi LYWSDCGQ (58:2D:34:32:2D:68):␛[0m

OttoWinter, 2019-11-12:
@perrycharlton @stripeyhorse Unrelated, see https://github.com/esphome/issues/issues/824#issuecomment-552120649

dd1982, 2019-11-16:
https://www.espressif.com/sites/default/files/documentation/ESP32_FAQs__EN.pdf

See 5.3.2 - is it related? Not sure what esphome sets Software controls WiFi/Bluetooth coexistence to but it seems the defaults don't match the recommended values either way?

"Please note that if Option “Software controls WiFi/Bluetooth coexistence” is enabled, the BLE scan interval shall not exceed 0x100 slots (about 160 ms)."

"However, if this option is not enabled, please note that the “BLE scan interval - BLE scan window” should be larger than 150 ms, and the BLE scan interval should be less than 500 ms."

OttoWinter, 2019-11-17:
@dd1982 Please use the correct thread, #824.

definitio, 2020-03-16:
I just tried to compile and it works.

closed 2020-03-16 by definitio