esphome/issues #802: ESP32 Bootloop with MH-SR505

balk77, 2019-11-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome 1.14 on Ubuntu, ESPHome-flasher 1.2.0 on Mac OS High Sierra

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, from Ali:
https://www.aliexpress.com/item/32992390199.html

MH-SR505 Pir sensor
https://www.aliexpress.com/item/32711395571.html

**Affected component:**

**Description of problem:**
Similar to the issue described here: https://github.com/esphome/issues/issues/598 my ESP32 is in a bootloop (see error below). The difference is that it only happens on this device when I attache a MH-SR505 PIR sensor to it. Without it, it boots fine. I did follow the steps to [upload](https://github.com/esphome/issues/issues/598#issuecomment-545903272) a new bootloader

**Problem-relevant YAML-configuration entries:**
```yaml
Does not make a difference; also occurs with an bare minimum YAML config.

```

**Logs (if applicable):**

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

```

**Additional information and things you've tried:**
I have loaded a new bootloader with this command, did not help
```
/Applications/ESPHome-Flasher-1.2.0-macOS.appontents/MacOS/ESPHome-Flasher ir_receiver_co.bin --bootloader bootloader_qio_80m.bin 
```
Same error with a similar board from the same shipment.

OttoWinter, 2019-11-02:
Please post a picture of how you wired up the PIR sensor.

I've had an ESP32 with wrong silkscreen labels - what was labeled as a GND pin was actually the CLK line to the flash chip. Try connecting the PIR sensor to different pins.

balk77, 2019-11-02:
Forgot to mention; it did work in a wemos d1 (presumably with identical
wiring :) )

I will try different hookup though

balk77, 2019-11-08:
I hope this is sufficiently clear. It does workvwith the same wiring on another board. It fails also when only the + of the sensor is connected.

[image]

OttoWinter, 2019-11-08:
See the GND pin in the lower left corner.

On my board, it was exactly the same: That pin is actually `CLK` and mislabeled.

Please connect the wire you have connected there to another GND pin on the board.

closed 2019-11-22 by no-response[bot]