esphome/issues #803: Can't compile after update to 1.14 ('APIBuffer' has not been declared)

rradar, 2019-11-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

local pip3 install

**ESP (ESP32/ESP8266, Board/Sonoff):**

- esp8285 (Blitzwolf shp2)
- esp8266 (d1 mini)

**Affected component:**

`esphome/components/api/api_message.cpp`

**Description of problem:**

Can't compile anymore after upgrade to esphome 1.14. Lateste 1.13.6 worked without warnings/errors

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: shp2
  upper_devicename: SHP2
  device_hardware: Blitzwolf SHP2 (16A)

  # Higher value gives lower watt readout
  current_res: "0.00221"
  # Lower value gives lower voltage readout
  voltage_div: "871"

# include wifi, api, ota
<<: !include "include/common.yaml" 

esphome:
  platform: ESP8266
  board: esp8285
  name: $devicename

logger:
  level: VERBOSE

web_server:
  port: 80

time:
  - platform: homeassistant
    id: hass_time

status_led:
  pin:
    number: GPIO02 # Blue Led
    inverted: True

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO3
      inverted: True
    name: "${upper_devicename} Button"
    on_press:
      - switch.toggle: relay

  - platform: status
    name: "${upper_devicename} Status"

output:
  - platform: gpio
    id: red_led
    pin: GPIO13
    inverted: True

sensor:
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: true
    cf_pin: GPIO4
    cf1_pin: GPIO5
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
    current:
      name: '${upper_devicename} Current'
      unit_of_measurement: 'A'
      accuracy_decimals: 3
      icon: mdi:current-ac
    voltage:
      name: '${upper_devicename} Voltage'
      unit_of_measurement: 'V'
      icon: mdi:flash
    power:
      name: '${upper_devicename} Power'
      id: power
      unit_of_measurement: 'W'
      icon: mdi:power
    change_mode_every: 4
    update_interval: 10s

  - platform: total_daily_energy
    name: "${upper_devicename} Total Daily Energy"
    power_id: power
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
    unit_of_measurement: kWh
    icon: mdi:counter

  - platform: wifi_signal
    name: "${upper_devicename} Signal"

  - platform: uptime
    name: "${upper_devicename} Uptime"
    id: uptime_sec

  - platform: template
    name: "${upper_devicename} Uptime Days"
    lambda: |-
      return (id(uptime_sec).state/60)/60/24;
    icon: mdi:clock-start
    unit_of_measurement: days

switch:
  - platform: gpio
    name: "${upper_devicename} Relay"
    pin: GPIO14
    restore_mode: RESTORE_DEFAULT_OFF
    id: relay
    on_turn_on:
      - output.turn_on:
          id: red_led
    on_turn_off:
      - output.turn_off:
          id: red_led

  - platform: restart
    name: "${upper_devicename} Restart"

text_sensor:
  - platform: version
    name: "${upper_devicename} Version"

  - platform: template
    name: "${upper_devicename} Hardware"
    lambda: |-
      return {"${device_hardware}"};
    icon: 'mdi:tag'
    update_interval: 24h
```

**Logs (if applicable):**

```yaml
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.5
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
Compiling .pioenvs/shp6/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/shp6/src/esphome/components/api/api_message.cpp.o
Compiling .pioenvs/shp6/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/shp6/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/api/api_message.cpp:1:0:
src/esphome/components/api/api_message.h:75:23: error: 'APIBuffer' has not been declared
   virtual void encode(APIBuffer &buffer);
                       ^
src/esphome/components/api/api_message.cpp:12:25: error: variable or field 'encode' declared void
 void APIMessage::encode(APIBuffer &buffer) {}
                         ^
src/esphome/components/api/api_message.cpp:12:25: error: 'APIBuffer' was not declared in this scope
src/esphome/components/api/api_message.cpp:12:36: error: 'buffer' was not declared in this scope
 void APIMessage::encode(APIBuffer &buffer) {}
                                    ^
src/esphome/components/api/api_message.cpp: In member function 'void esphome::api::APIMessage::decode(const uint8_t*, size_t)':
src/esphome/components/api/api_message.cpp:18:72: error: 'proto_decode_varuint32' was not declared in this scope
     auto res = proto_decode_varuint32(&buffer[i], length - i, &consumed);
                                                                        ^
*** [.pioenvs/shp6/src/esphome/components/api/api_message.cpp.o] Error 1
================================================================== [FAILED] Took 8.87 seconds ==================================================================
```

**Additional information and things you've tried:**

rradar, 2019-11-02:
Problem solved.

**Was releated to old/broken pip install(s)**

I removed all esphome installs I had:

- `pip2 uninstall esphome`
- `pip3 uninstall esphome`

Went to `/.local/lib/python3.6/site-packages/` and `/.local/lib/python2.7/site-packages/` and removed all esphome related folders.

Did a new install with `pip3 install esphome` and compiling works again :+1:

OttoWinter, 2019-11-02:
Looks like https://github.com/esphome/issues/issues/752

Yeah there's not much that can be done unfortunately - apparently setuptools doesn't remove old files when installing under specific circumstances.

closed 2019-11-02 by OttoWinter