esphome/issues #804: Devices crashes on start (Exception (3): WARNING Exception type: unknown)

rradar, 2019-11-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

local pip3 install

**ESP (ESP32/ESP8266, Board/Sonoff):**

wemos d1 mini
sonoff pow r2 (esp01_1m)
blitzwolf shp6 (esp8285)

**Affected component:**

setup?

**Description of problem:**

I started upgrading my devices from 1.13 to 1.14 and did 10 with success. Then I did some streamlining (just some comments/naming/icons) in the yaml's and did upload them again with ota. Now I experienced problems with some of them.

First thing went down was a d1 mini (no more connecting to wifi and not opening an ap like configured). Then my blitzwolf shp6 (esp8285) didn't upgraded via command line. It always started and faild mostly under 10% with a broken pipe error. I could fix this once using the web server to upload my firmware.bin over the browser. But after the last ota comand line update it doesn't connect to the wifi anymore (also no ap) and blinks the status led fast (the broken d1 mini does this as well). Last device I lost was a sonoff pow r2 which went down and doesn't even light the status led anymore.

The only device I could easily access logging is the wemos d1 mini. Here comes the log:

**Logs (if applicable):**

```yaml
[19:44:21]Exception (3):
WARNING Exception type: unknown
[19:44:21]epc1=0x40222ccd epc2=0x00000000 epc3=0x00000000 excvaddr=0x400710de depc=0x00000000
[19:44:21]
[19:44:21]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[19:44:21]
[19:44:21]ctx: cont
[19:44:21]sp: 3ffffab0 end: 3fffffc0 offset: 01a0
[19:44:21]3ffffc50:  0000049c 402615e0 000003fd 402230ac  
[19:44:21]3ffffc60:  40100461 3fff10e4 402615e0 401005dc  
[19:44:21]3ffffc70:  40252492 4025245e 000003fd 402524af  
[19:44:21]3ffffc80:  3fff10e4 40252517 3fff2f14 0000049c  
[19:44:21]3ffffc90:  000003fd 3ffffd60 3fff2f14 402524fa  
[19:44:21]3ffffca0:  ffffff00 55aa55aa 00000009 00000020  
[19:44:21]3ffffcb0:  00000020 00000010 00000010 aa55aa55  
[19:44:21]3ffffcc0:  000003ff 402529e4 3fff2f14 3fff2f14  
[19:44:21]3ffffcd0:  000000ff 3fff2f14 0000049c 00000000  
[19:44:21]3ffffce0:  40100461 00000001 3fff2f24 40252bfe  
[19:44:21]3ffffcf0:  00000008 3fff2f14 000000ff 3ffffd60  
[19:44:21]3ffffd00:  3ffffd80 3fff2f4b 00000009 00000020  
[19:44:21]3ffffd10:  3fff2fd4 3ffffdc1 00000001 40252cae  
[19:44:21]3ffffd20:  3ffffd60 40261240 3fff1724 0000002c  
[19:44:21]3ffffd30:  3fff3314 3ffffd80 3fff924c 40252c7d  
[19:44:21]3ffffd40:  3fff2f14 40252ce4 00000001 0000000b  
[19:44:21]3ffffd50:  4020e4ea 0000000b 3fff1724 4020e476  
[19:44:21]3ffffd60:  00000000 00000000 00000000 00000000  
[19:44:21]3ffffd70:  00000000 00000000 00000000 00000000  
[19:44:21]3ffffd80:  00000000 00000000 00000000 00000000  
[19:44:21]3ffffd90:  00000000 00000000 00000000 00000000  
[19:44:21]3ffffda0:  00000000 00000000 00000000 00000000  
[19:44:21]3ffffdb0:  00000000 00000000 00000000 00000000  
[19:44:21]3ffffdc0:  00000000 00000000 00000000 00000000  
[19:44:21]3ffffdd0:  00010101 ff000000 3fff1f7c 3fff20b8  
[19:44:21]3ffffde0:  3fff1724 0000000b 3fff1724 4020d0e2  
[19:44:21]3ffffdf0:  00000000 0000000b 3fff07e0 4022418c  
[19:44:21]3ffffe00:  3fff2484 00000008 3fff07e0 402241f9  
[19:44:21]3ffffe10:  3fff16ec 0000000a 3fff07e0 402100cd  
[19:44:21]3ffffe20:  3fff2484 00000010 00000001 4022efb2  
[19:44:21]3ffffe30:  00000000 00000000 00000000 402119f6  
[19:44:21]3ffffe40:  00000000 3fffff38 3fff0878 3fff1c74  
[19:44:21]3ffffe50:  00000000 3fffff38 3fff0878 4021291e  
[19:44:21]3ffffe60:  3fff1880 00000000 00000000 3fff18a8  
[19:44:21]3ffffe70:  00000000 00000000 00000000 40225e50  
[19:44:21]3ffffe80:  00000000 40225e50 00000000 40225e50  
[19:44:21]3ffffe90:  00000000 40225e50 00000000 40225e50  
[19:44:21]3ffffea0:  00000000 00000000 3fff18d0 00000000  
[19:44:21]3ffffeb0:  00000000 3fff18e8 00000000 00000000  
[19:44:21]3ffffec0:  00000000 40225e50 00000000 40225e50  
[19:44:21]3ffffed0:  00000000 40225e50 00000000 40225e50  
[19:44:21]3ffffee0:  00000000 40225e50 00000000 00000000  
[19:44:21]3ffffef0:  3fff1840 00000000 00000000 3fff1858  
[19:44:21]3fffff00:  00000000 00000000 00000000 40225e50  
[19:44:21]3fffff10:  00000000 40225e50 00000000 40225e50  
[19:44:21]3fffff20:  00000000 40225e50 00000000 40225e50  
[19:44:21]3fffff30:  00000000 00000000 3fff2384 00000000  
[19:44:21]3fffff40:  402119c8 40211a10 3fff23b4 3fff1ed4  
[19:44:21]3fffff50:  3fff1e34 3fff1dec 3fff1d9c 3fff1d34  
[19:44:21]3fffff60:  3fff1c74 3fff1c04 3fff1b9c 3fff1e74  
[19:44:21]3fffff70:  feefeffe feefeffe feefeffe feefeffe  
[19:44:21]3fffff80:  3fff1dec 3fff0890 3fff23b4 3ffffea8  
[19:44:21]3fffff90:  feefeffe feefeffe feefeffe 3fff0b28  
[19:44:21]3fffffa0:  3fffdad0 00000000 3fff0af8 402212b8  
[19:44:21]3fffffb0:  feefeffe feefeffe 3ffe8580 40100869  
[19:44:21]<<<stack<<<
[19:44:21]
[19:44:21] ets Jan  8 2013,rst cause:2, boot mode:(3,7)
[19:44:21]
[19:44:21]load 0x4010f000, len 1384, room 16 
[19:44:21]tail 8
[19:44:21]chksum 0x2d
[19:44:21]csum 0x2d
[19:44:21]v8b899c12
[19:44:21]~ld
```

**Additional information and things you've tried:**

rradar, 2019-11-02:
I'm crying already about my blitzwolf shp6 which was freed ota with tuya convert and will be horror to gain serial access

[image]

OttoWinter, 2019-11-02:
> Then I did some streamlining (just some comments/naming/icons) in the yaml's and did upload them again with ota. Now I experienced problems with some of them.

Sounds a bit like that darn https://github.com/esphome/issues/issues/455 bug :(

> Last device I lost was a sonoff pow r2 which went down and doesn't even light the status led anymore. Toggling the relay with the button also isn't possible anymore with the blitzwolf and the sonoff.

Uh oh, that means something more than just WiFi is messed up - the automations (if they're defined in the device YAML) should run independently of WiFi. The stacktrace might show some insight.

> The only device I could easily access logging is the wemos d1 mini. Here comes the log:

There was an issue in 1.14.0 where logs were not decoded. If possible, try getting the logs again with .1 (currently being released).

rradar, 2019-11-02:
I tried two times to flash again with 1.14.0 via local usb and the error remained the same.
Now I upgraded to 1.14.1 flashed via local usb and the error is a new one:

```yaml
[21:13:08]Exception (3):
WARNING Exception type: unknown
[21:13:08]epc1=0x402232e5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4007119e depc=0x00000000
WARNING Decoded 0x402232e5: _umm_free at umm_malloc.cpp
[21:13:08]
[21:13:08]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[21:13:08]
[21:13:08]ctx: cont
[21:13:08]sp: 3ffffaa0 end: 3fffffc0 offset: 01a0
[21:13:08]3ffffc40:  0000049c 40261d50 000003fd 402236c4  
WARNING Decoded 0x40261d50: sleep_reset_analog_rtcreg_8266
WARNING Decoded 0x402236c4: free
[21:13:08]3ffffc50:  40100461 3fff11a4 40261d50 401005dc  
WARNING Decoded 0x40100461: __wrap_spi_flash_read
WARNING Decoded 0x40261d50: sleep_reset_analog_rtcreg_8266
WARNING Decoded 0x401005dc: vPortFree
[21:13:08]3ffffc60:  40252b9a 40252b66 000003fd 40252bb7  
WARNING Decoded 0x40252b9a: wifi_param_save_protect_with_check
WARNING Decoded 0x40252b66: wifi_param_save_protect_with_check
WARNING Decoded 0x40252bb7: wifi_param_save_protect_with_check
[21:13:08]3ffffc70:  3fff11a4 40252c1f 3fff323c 0000049c  
WARNING Decoded 0x40252c1f: system_param_save_with_protect
[21:13:08]3ffffc80:  000003fd 3ffffd50 3fff323c 40252c02  
WARNING Decoded 0x40252c02: system_param_save_with_protect
[21:13:08]3ffffc90:  ffffff00 55aa55aa 00000009 00000020  
[21:13:08]3ffffca0:  00000020 00000010 00000010 aa55aa55  
[21:13:08]3ffffcb0:  000003ff 402530ec 3fff323c 3fff323c  
WARNING Decoded 0x402530ec: wifi_station_ap_number_set
[21:13:08]3ffffcc0:  000000ff 3fff323c 0000049c 00000000  
[21:13:08]3ffffcd0:  40100461 00000001 3fff324c 40253306  
WARNING Decoded 0x40100461: __wrap_spi_flash_read
WARNING Decoded 0x40253306: wifi_station_ap_number_set
[21:13:08]3ffffce0:  00000008 3fff323c 000000ff 3ffffd50  
[21:13:08]3ffffcf0:  3ffffd70 3fff3273 00000009 00000020  
[21:13:08]3ffffd00:  3fff32fc 3ffffdb1 00000001 402533b6  
WARNING Decoded 0x402533b6: wifi_station_ap_number_set
[21:13:08]3ffffd10:  3ffffd50 402619b0 3fff17e4 0000002c  
WARNING Decoded 0x402619b0: sleep_reset_analog_rtcreg_8266
[21:13:08]3ffffd20:  3fff363c 3ffffd70 3fff9574 40253385  
WARNING Decoded 0x40253385: wifi_station_ap_number_set
[21:13:08]3ffffd30:  3fff323c 402533ec 00000001 0000000b  
WARNING Decoded 0x402533ec: wifi_station_set_config
[21:13:08]3ffffd40:  4020e4ea 0000000b 3fff17e4 4020e476  
WARNING Decoded 0x4020e4ea: esphome::wifi::WiFiComponent::wifi_sta_pre_setup_()
WARNING Decoded 0x4020e476: esphome::wifi::WiFiComponent::wifi_sta_pre_setup_()
[21:13:08]3ffffd50:  00000000 00000000 00000000 00000000  
[21:13:08]3ffffd60:  00000000 00000000 00000000 00000000  
[21:13:08]3ffffd70:  00000000 00000000 00000000 00000000  
[21:13:08]3ffffd80:  00000000 00000000 00000000 00000000  
[21:13:08]3ffffd90:  00000000 00000000 00000000 00000000  
[21:13:08]3ffffda0:  00000000 00000000 00000000 00000000  
[21:13:08]3ffffdb0:  00000000 00000000 00000000 00000000  
[21:13:08]3ffffdc0:  00010101 ff000000 3fff2084 3fff2148  
[21:13:08]3ffffdd0:  3fff17e4 0000000b 3fff17e4 4020d0e2  
WARNING Decoded 0x4020d0e2: esphome::wifi::WiFiComponent::setup()
[21:13:08]3ffffde0:  00000000 0000000b 3fff0890 402247a4  
WARNING Decoded 0x402247a4: esphome::Component::call_setup()
[21:13:08]3ffffdf0:  3fff281c 00000008 3fff0890 40224811  
WARNING Decoded 0x40224811: esphome::Component::call()
[21:13:08]3ffffe00:  3fff17ac 0000000a 3fff0890 402104f9  
WARNING Decoded 0x402104f9: esphome::Application::setup()
[21:13:08]3ffffe10:  3fff281c 00000013 3fff2790 3ffea457  
[21:13:08]3ffffe20:  3fff26e0 3fffff68 3fff26e0 4022f90a  
WARNING Decoded 0x4022f90a: std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string() at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:541
[21:13:08]3ffffe30:  3fffff01 3fff0ce4 3fff26e0 3ffea457  
[... 30 lines cut ...]
[21:13:09]3fffff20:  00000000 00000000 3fff26e0 00000000  
[21:13:09]3fffff30:  40211df4 40211e3c 3fff2668 3fff255c  
WARNING Decoded 0x40211df4: std::_Function_base::_Base_manager<setup::{lambda()#2}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<setup::{lambda()#2}> const&, std::_Manager_operation) at main.cpp
WARNING Decoded 0x40211e3c: std::_Function_handler<esphome::optional<float> (), setup::{lambda()#2}>::_M_invoke(std::_Any_data const&) at main.cpp
[21:13:09]3fffff40:  3fff2018 3fff1f94 3fff1ef4 3fff1eac  
[21:13:09]3fffff50:  3fff1e5c 3fff1df4 3fff1d34 3fff1cc4  
[21:13:09]3fffff60:  3fff1c5c 3fff1f34 3fff2790 feefeffe  
[21:13:09]3fffff70:  3ffffe98 3fffff38 3fff2524 000000e8  
[21:13:09]3fffff80:  3fff1eac 3ffffe98 feefeffe feefeffe  
[21:13:09]3fffff90:  feefeffe feefeffe feefeffe 3fff0be8  
[21:13:09]3fffffa0:  3fffdad0 00000000 3fff0bb8 402218d0  
WARNING Decoded 0x402218d0: loop_wrapper() at core_esp8266_main.cpp
[21:13:09]3fffffb0:  feefeffe feefeffe 3ffe8580 40100869  
WARNING Decoded 0x40100869: cont_wrapper
[21:13:09]<<<stack<<<
[21:13:09]
[21:13:09] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[21:13:09]
[21:13:09]load 0x4010f000, len 1384, room 16 
[21:13:09]tail 8
[21:13:09]chksum 0x2d
[21:13:09]csum 0x2d
[21:13:09]v8b899c12
[21:13:09]~ld
```

OttoWinter, 2019-11-02:
Thanks very much @rradar for the logs. And I'm very sorry for the lost device :(

I was now able to pinpoint the changes that caused the issue. Or at least the lines in the stacktrace point to the one change made to the ESP8266 WiFi class.
But those changes were just changes I pulled in from the upstream Arduino library.

It looks like something is wrong with saving wifi data to the flash section. I don't know what exactly because (again) the SDK there is closed source. 

In https://github.com/esphome/esphome/pull/831, I'm reverting those changes.

If possible, please also post as much information as possible about the devices. What exact boards are they, a link to the site you bought them from and so on - this stacktrace is from before the ESP tries to connect to wifi even. I've been running the beta on my devices (mostly NodeMCU boards, but also some Sonoff&Tuya devices) for about a month without this issue, so I assume it's device-specific.

rradar, 2019-11-02:
I flashed a couple of more times via local usb but the (new) error/crash persists.
Removing `wifi, api, ota` from my yaml gives a stable build but adding network functionalities again makes it immediately crashing on boot.

The wemos d1 mini (v3.0.0, no original one) is running esphome since a long time. In this particular setup since june this year without problems. I can't really tell when and where I bought it but the chance is very high that I have between 2 and 4 of this batch of wemos d1 mini's and they all running esphome (probably all but maybe one still on 1.13.6).

The blitzwolf shp6 was just flashed recently (still with esphome 1.13.6) after it was freed with tuya-convert and initially flashed with espurna. It did everything (switch, power measurement) correctly (actually I had some weird extremes in voltage measurements but in all no stability problems...).

For the sonoff pow r2 I can't really tell yet if it's exactly the same error because of the led not flashing at all. But I have in total 4 of this devices (2 different batches) and the other 3 ones I flashed already to 1.14 (couple of times with different yaml) didn't make any problems. I need to hook up a serial adapter here to see more.

rradar, 2019-11-02:
And what I just found out few seconds ago: When erasing the flash of the esp via

`esptool.py --port /dev/ttyUSB0 erase_flash`

and then flash (the formerly non working yaml including `wifi, ota, api`) with esphome again it just works again :roll_eyes:

OttoWinter, 2019-11-02:
Hmm, that's an interesting observation!

That probably means that the error is due to some faulty data being in the flash area for wifi settings. That would also make sense if those devices had a different fw installdd before that used espressif's wifi_station_set config persistent methods.

The change there (that I'm now reverting) was actually meant to fix invalid data in that section from causing issues. It basically looks at that section on each boot, and if there's data there it gets cleared. No idea why the *clearing* would cause issues though 🙄

Related: https://github.com/esp8266/Arduino/issues/1997 https://www.esp8266.com/viewtopic.php?f=159&t=14051

rradar, 2019-11-02:
Wouldn't do this https://github.com/esp8266/Arduino/issues/1997#issuecomment-436673828 do the trick in 2019 too by accident?

OttoWinter, 2019-11-02:
ESPHome has been doing that for a long time already.

The problem was that even if you set it to non-persistent mode, there's still (potentially invalid) data in the flash from before. That patch was supposed to remove invalid data from flash.

OttoWinter, 2019-11-02:
Do you happen to know what was previously installed on those ESPs? I'd like to reproduce the issue and I *think* I might be able to do it by flashing a different fw.

rradar, 2019-11-02:
What's really nasty about this is that it can't recover. Even I have the fallback AP defined my Blitzwolf plug is no literally bricked for now.

I can tell you the exact flash history of this Blitzwolf SHP2 power meter plug (esp8285) - he head a short live :cry: 

It first came with the stock tuya firmware but was freed directly with tuya convert and flashed with the included ESPurna 1.13.5 binary (espurna-base.bin from https://github.com/ct-Open-Source/tuya-convert/blob/master/files/espurna-base.bin). Then the esphome show begins:
[image]
The end is important, close up:
[image]
In total I flashed (after the inital 1.13.6 on the 30th of october thru the espurna web server) another 3 times successful flashes with version 1.14.0 (two of them via ota, one like described earlier via the esphome web server). The last flash (flash of death) isn't in the chart because it came never back under the living.

One more thought: The sonoff pow r2 which also went away (no flashing led, no logs yet) was also just recently flashed over from espurna to esphome via the web server.

OttoWinter, 2019-11-02:
I just checked with Tasmota - couldn't reproduce; tried ESPurna too, but that wouldn't let me plug in my throwaway WiFi password because it "isn't secure enough". But from the interface I doubt they're using that API.

Anyway, I'm reverting that patch and publishing 1.14.2 so that hopefully not many more people are affected.

DavidFW1960, 2019-11-03:
Since 14.2 and our discord convo yesterday (see also my convo today) My nodemcu is showing up in my router as connected and I get a green light in the ESPhome interface (hass.io). The log still shows it can't connect and if I add to HA integration it can't connect either. I also tried to telnet to the nodemcu but no connection. I can ping it though.
Like I said, 13.6 worked for weeks... 14.0 killed it. 455 bug it seems but maybe you're getting closer to killing it.

rradar, 2019-11-03:
I took apart my sonoff pow r2 and attached uart. I couldn't take any logging because baud rate is set to 0 for energy measurement via uart. Anyhow I just flashed the same sketch I had on it (compiled with 1.14.0/ota) over again via uart - this time with the newest version 1.14.2 and my device back again :tada:

Now I will try to get my last device back alive. I just get a log from this one because it doesn't use uart for power measurement... stay tuned!

OttoWinter, 2019-11-03:
@rradar That's great! Thank you again for reporting this bug early and quickly giving updated logs! I hope the quick .2 patch release update will prevent other people from running into this issue.

Also sorry the beta didn't catch this - this time the beta even was for a longer period but this issue wasn't caught unfortunately :( I'm going to have to rethink how to roll out wifi changes in the future - it appears like any small change to the WiFi code can have pretty bad effects unfortunately. Maybe some opt-in way to try some fixes but then there's again going to be the problem of not enough people trying it (like in beta phase).

----

@DavidFW1960 
> The log still shows it can't connect and if I add to HA integration it can't connect either.

Hmm, that looks like a different issue. Have you made sure you re-enabled `api:` in the config file? I think I told you to try a minimal config, maybe that's what's on the device?

> I also tried to telnet to the nodemcu but no connection. 

There's no telnet service for logs with ESPHome. Logs run through the communication method of choice: native API or MQTT.

rradar, 2019-11-03:
And the last device is also under the living again. :champagne: :fireworks: 

Here is the log (two stacks, slightly different I think) from the blitzwolf shp6 while looping with exception (3):

```
Exception (3):
WARNING Exception type: unknown
[12:31:23]epc1=0x40223e35 epc2=0x00000000 epc3=0x00000000 excvaddr=0x400717b6 depc=0x00000000
WARNING Decoded 0x40223e35: umm_make_new_block$constprop$1 at umm_malloc.cpp
[12:31:23]
[12:31:23]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[12:31:23]
[12:31:23]ctx: cont
[12:31:23]sp: 3ffffa70 end: 3fffffc0 offset: 01a0
[12:31:23]3ffffc10:  0000049c 402655c0 000000fd 40224214  
WARNING Decoded 0x402655c0: sleep_reset_analog_rtcreg_8266
WARNING Decoded 0x40224214: std::_List_base<ScheduledFunctions::ScheduledElement, std::allocator<ScheduledFunctions::ScheduledElement> >::_M_clear()
[12:31:23]3ffffc20:  401004d9 3fff17bc 402655c0 40100874  
WARNING Decoded 0x401004d9: __wrap_spi_flash_read
WARNING Decoded 0x402655c0: sleep_reset_analog_rtcreg_8266
WARNING Decoded 0x40100874: vPortFree
[12:31:23]3ffffc30:  4025631e 402562ea 000000fd 4025633b  
WARNING Decoded 0x4025631e: system_param_save_with_protect
WARNING Decoded 0x402562ea: system_param_save_with_protect
WARNING Decoded 0x4025633b: system_param_save_with_protect
[12:31:23]3ffffc40:  3fff17bc 402563a3 3fff3e24 0000049c  
WARNING Decoded 0x402563a3: system_save_sys_param
[12:31:23]3ffffc50:  000000fd 3ffffd20 3fff3e24 40256386  
WARNING Decoded 0x40256386: system_param_save_with_protect
[12:31:23]3ffffc60:  ffffff00 55aa55aa 0000000b 00000020  
[12:31:23]3ffffc70:  00000020 00000010 00000010 aa55aa55  
[12:31:23]3ffffc80:  000000ff 40256870 3fff3e24 3fff3e24  
WARNING Decoded 0x40256870: wifi_station_ap_number_set
[12:31:23]3ffffc90:  000000ff 3fff3e24 0000049c 00000000  
[12:31:23]3ffffca0:  401004d9 00000001 3fff3e34 40256a8a  
WARNING Decoded 0x401004d9: __wrap_spi_flash_read
WARNING Decoded 0x40256a8a: wifi_station_ap_number_set
[12:31:23]3ffffcb0:  00000008 3fff3e24 000000ff 3ffffd20  
[12:31:23]3ffffcc0:  3ffffd40 3fff3e5b 0000000b 00000020  
[12:31:23]3ffffcd0:  3fff3ee4 3ffffd81 00000001 40256b3a  
WARNING Decoded 0x40256b3a: wifi_station_get_current_ap_id
[12:31:23]3ffffce0:  3ffffd20 40265220 3fff1e44 0000003c  
WARNING Decoded 0x40265220: sleep_reset_analog_rtcreg_8266
[12:31:23]3ffffcf0:  3fff4224 3ffffd40 3fffa15c 40256b09  
WARNING Decoded 0x40256b09: wifi_station_set_config_current
[12:31:23]3ffffd00:  3fff3e24 40256b70 00000001 0000000f  
WARNING Decoded 0x40256b70: wifi_station_ap_check
[12:31:23]3ffffd10:  4020e6ca 0000000f 3fff1e44 4020e656  
WARNING Decoded 0x4020e6ca: esphome::wifi::WiFiComponent::wifi_pre_setup_()
WARNING Decoded 0x4020e656: esphome::wifi::WiFiComponent::wifi_sta_pre_setup_()
[12:31:23]3ffffd20:  00000000 00000000 00000000 00000000  
[12:31:23]3ffffd30:  00000000 00000000 00000000 00000000  
[12:31:23]3ffffd40:  00000000 00000000 00000000 00000000  
[12:31:23]3ffffd50:  00000000 00000000 00000000 00000000  
[12:31:23]3ffffd60:  00000000 00000000 00000000 00000000  
[12:31:23]3ffffd70:  00000000 00000000 00000000 00000000  
[12:31:23]3ffffd80:  00000000 00000000 00000000 00000000  
[12:31:23]3ffffd90:  00010101 ff000000 3fff0da4 3fff2cd0  
[12:31:23]3ffffda0:  3fff1e44 0000000f 3fff1e44 4020d2c2  
WARNING Decoded 0x4020d2c2: esphome::wifi::WiFiComponent::setup()
[12:31:23]3ffffdb0:  00000000 0000000f 3fff0da4 402253c4  
WARNING Decoded 0x402253c4: esphome::Component::call()
[12:31:23]3ffffdc0:  3fff3274 00000009 3fff0da4 40225431  
WARNING Decoded 0x40225431: std::_Function_handler<void (bool), esphome::Controller::setup_controller()::{lambda(bool)#1}>::_M_invoke(std::_Any_data const&, bool) at controller.cpp
[12:31:23]3ffffdd0:  3fff1e04 0000000e 3fff0da4 402106d9  
WARNING Decoded 0x402106d9: esphome::Application::setup()
[12:31:23]3ffffde0:  3fff3274 00000017 3fffff08 3fff261c  
[12:31:23]3ffffdf0:  00000001 40232f0a 00000010 40224030  
WARNING Decoded 0x40232f0a: std::string::_M_move(char*, char const*, unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:365
[... 167 lines cut ...]
[12:31:24]3fffff20:  feefeffe feefeffe feefeffe feefeffe  
[12:31:24]3fffff30:  feefeffe feefeffe feefeffe feefeffe  
[12:31:24]3fffff40:  feefeffe feefeffe feefeffe feefeffe  
[12:31:24]3fffff50:  feefeffe feefeffe feefeffe feefeffe  
[12:31:24]3fffff60:  feefeffe feefeffe feefeffe feefeffe  
[12:31:24]3fffff70:  3fffff08 00000050 feefeffe 000000d8  
[12:31:24]3fffff80:  000000e8 feefeffe feefeffe feefeffe  
[12:31:24]3fffff90:  feefeffe feefeffe feefeffe 3fff1138  
[12:31:24]3fffffa0:  3fffdad0 00000000 3fff1108 40222338  
WARNING Decoded 0x40222338: optimistic_yield
[12:31:24]3fffffb0:  feefeffe feefeffe 3ffe85d0 40100b45  
WARNING Decoded 0x40100b45: cont_wrapper
[12:31:24]<<<stack<<<
[12:31:24]
[12:31:24] ets Jan  8 2013,rst cause:1, boot mode:(3,7)
[12:31:24]
[12:31:24]load 0x4010f000, len 1384, room 16 
[12:31:24]tail 8
[12:31:24]chksum 0x2d
[12:31:24]csum 0x2d
[12:31:24]v8b899c12
[12:31:24]~ld
```

 looks the same for me like the wemos d1 mini. For this one I also needed to erase the flash with esptool before. Otherwise the new flashed yaml (esphome 1.14.2, uart) also boot looped right away.

DavidFW1960, 2019-11-03:
I was using the full config file with API enabled. I can also flash OTA as well.. it's just the log can't connect and HA can't connect.

OttoWinter, 2019-11-12:
The last change to wifi has now been reverted in https://github.com/esphome/esphome/pull/835 - the wifi codebase is now pretty much back to 1.13.x

Closing

closed 2019-11-12 by OttoWinter