esphome/issues #805: Compilation and upload problem

kprzeb, 2019-11-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker aarch64 odroidc2
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 sonoff mini
**Affected component:**

**Description of problem:**
Compilation and upload problem

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: korytarz_gora
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "ssid"
  password: "pass"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO04
      mode: INPUT_PULLUP
      inverted: True
    name: "Światło korytarz góra przełącznik"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Światło korytarz góra"
    pin: GPIO12
    id: relay

```

**Logs (if applicable):**

INFO Reading configuration /config/esphome/korytarz_gora.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/korytarz_gora
Processing korytarz_gora (board: esp01_1m; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/korytarz_gora/.pioenvs/korytarz_gora/src/esphome/components/api/api_connection.cpp.o
Illegal instruction (core dumped)
Compiling /data/korytarz_gora/.pioenvs/korytarz_gora/src/esphome/components/api/api_pb2.cpp.o
Illegal instruction (core dumped)
Compiling /data/korytarz_gora/.pioenvs/korytarz_gora/src/esphome/components/api/api_pb2_service.cpp.o
Illegal instruction (core dumped)
*** [/data/korytarz_gora/.pioenvs/korytarz_gora/src/esphome/components/api/api_connection.cpp.o] Error 132
Compiling /data/korytarz_gora/.pioenvs/korytarz_gora/src/esphome/components/api/api_server.cpp.o
*** [/data/korytarz_gora/.pioenvs/korytarz_gora/src/esphome/components/api/api_pb2.cpp.o] Error 132
*** [/data/korytarz_gora/.pioenvs/korytarz_gora/src/esphome/components/api/api_pb2_service.cpp.o] Error 132
Illegal instruction (core dumped)
*** [/data/korytarz_gora/.pioenvs/korytarz_gora/src/esphome/components/api/api_server.cpp.o] Error 132
========================== [FAILED] Took 7.66 seconds ==========================

**Additional information and things you've tried:**

chrisp250, 2019-11-03:
Same issue here. The same configuration works fine when building for ESP32.
This was working prior to 1.14 update. 
Thank you

brandond, 2019-11-04:
Note that the xtensa compiler itself is crashing. ESP8266 and ESP32 have different toolchains so that would explain why one is working and not the other.

OttoWinter, 2019-11-04:
Yeah so this is almost certainly due to the Arduino framework update.

For the 2.4.2 version, I manually compiled a toolchain for aarch64. For the latest one I was under the impression the one supplied by the framework worked for aarch64 - at least it did on my RPi 3b+. Maybe it's because it's armv8 with both aarch64 and armhf support, whereas your board does not.

You can use the old version by putting `arduino_version: 2.4.2` in your core-esphome config.

chrisp250, 2019-11-05:
I tried using:
`arduino_version: 2.4.2`
but I still get the same result.

zvldz, 2019-11-13:
Why can't you use a ready-made toolchain for aarch64 ?
I tried it in a docker, it works well for me.

`curl -L https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-4/aarch64-linux-gnu.xtensa-lx106-elf-b40a506.1563313032.tar.gz | tar --strip-components=1 -xzf - -C /root/.platformio/packages/toolchain-xtensa`

OttoWinter, 2019-11-14:
> Why can't you use a ready-made toolchain for aarch64 ?

That's not the toolchain arduino SDK 2.5.2 is using.

To clear this up:

 - Arduino SDK 2.6.0 (not released on platformio yet) is using the earlephilhower 2.5.0-4 toolchain, which works well for aarch64
 - Arduino SDK 2.5.0 (current default) is using a different toolchain, can't remember which one again
 - Arduino SDK 2.4.2 (default in esphome v1.13.x) used yet another toolchain, for which I compiled a separate toolchain for aarch64 - it appears though SDK 2.4.2 is using the 2.5.0 toolchain, and the patch is not active.

I don't think SDK 2.5.0 can be used with toolchain 2.5.0-4, most toolchain updates are not forwards-compatible, so likely also not backwards-compatible.

zvldz, 2019-11-14:
> I don't think SDK 2.5.0 can be used with toolchain 2.5.0-4, most toolchain updates are not forwards-compatible, so likely also not backwards-compatible.

With 2.5.0-4 I compiled 2.5.2, 2.5.1, 2.5.0 and 2.4.2.
Why can't using different toolchain depending on sdk ?

deece, 2019-11-18:
I have the same problem using the Hass.io ESPHome plugin, 1.14.3, on an Orange Pi Prime.

zvldz, 2019-11-18:
> I have the same problem using the Hass.io ESPHome plugin, 1.14.3, on an Orange Pi Prime (Allwinner H5, Armbian).

Try it - https://github.com/esphome/issues/issues/805#issuecomment-553428445

deece, 2019-11-18:
> > I have the same problem using the Hass.io ESPHome plugin, 1.14.3, on an Orange Pi Prime (Allwinner H5, Armbian).
> 
> Try it - [#805 (comment)](https://github.com/esphome/issues/issues/805#issuecomment-553428445)
> It's in a docker with esphome.

Yup, that worked, thanks @zvldz 

For others who are stuck:
Find the Container ID of the ESPHome container:
`docker ps`
get a shell in that container:
`docker exec -it <Container ID> /bin/bash`
Fetch the toolchain:
`curl -L https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-4/aarch64-linux-gnu.xtensa-lx106-elf-b40a506.1563313032.tar.gz | tar --strip-components=1 -xzf - -C /root/.platformio/packages/toolchain-xtensa/`

novikovav69, 2019-12-12:
> Yup, that worked, thanks @zvldz
>
> For others who are stuck:
> Find the Container ID of the ESPHome container:
> `docker ps`
> get a shell in that container:
> `docker exec -it <Container ID> /bin/bash`
> Fetch [...]

I have new problem after this:
`|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/light_bedroom_upperlight/.pioenvs/light_bedroom_upperlight/src/esphome/components/api/api_connection.cpp.o
Compiling /data/light_bedroom_upperlight/.pioenvs/light_bedroom_upperlight/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: Exec format errorsh: 1: 
xtensa-lx106-elf-g++: Exec format error
*** [/data/light_bedroom_upperlight/.pioenvs/light_bedroom_upperlight/src/esphome/components/api/api_pb2.cpp.o] Error 2
*** [/data/light_bedroom_upperlight/.pioenvs/light_bedroom_upperlight/src/esphome/components/api/api_connection.cpp.o] Error 2
========================== [FAILED] Took 7.15 seconds ==========================`

zvldz, 2019-12-12:
> I have new problem after this:
> `|-- <ESP8266mDNS> 1.2 | |-- <ESP8266WiFi> 1.0 Compiling [...]

You probably don't have arm64

novikovav69, 2019-12-12:
> curl -L https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-4/aarch64-linux-gnu.xtensa-lx106-elf-b40a506.1563313032.tar.gz | tar --strip-components=1 -xzf - -C /root/.platformio/packages/toolchain-xtensa/

Yes. It's my trouble. I set up i686-linux-gnu.xtensa-lx106-elf-b40a506.1563313032.tar.gz and it's working! Thanx

brandond, 2019-12-12:
Are you running a 32-bit OS? I think all the non-ARM Docker containers are set up expecting x86_64.

OttoWinter, 2019-12-31:
True, there are no x86/i686 versions of the docker images available - iirc that's because the base image for Hass.io (ubuntu-hassio-base) does not have a 32-bit x86 variant.

geoffreyheller, 2020-01-03:
I am seeing the exact same error message when I try to compile my esphome yamls while running hassio on an Odroid C2.

metrafonic, 2020-01-13:
Having the same issue with vanilla hassio on an ODroid C2.

vks007, 2020-01-18:
I had the same error and I was also able to solve my error by following the below instructions with a little difference:

> Find the Container ID of the ESPHome container:
docker ps
get a shell in that container:
docker exec -it <Container ID> /bin/bash
Fetch the toolchain:
curl -L https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-4/aarch64-linux-gnu.xtensa-lx106-elf-b40a506.1563313032.tar.gz | tar --strip-components=1 -xzf - -C /root/.platformio/packages/toolchain-xtensa/

I had to delete the contents of the toolchain-xtensa directory inside docker first else it did not work. 
Also after deleting and getting its contents afresh , on my first compile ESPHome installed the toolchain-xtensa files as below:
`INFO Reading configuration /config/esphome/tank_controller.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/tank_controller
Processing tank_controller (board: d1_mini; framework: arduino; platform: espressif8266@2.2.3)
PackageManager: Installing toolchain-xtensa @ <3
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
toolchain-xtensa @ 2.40802.191122 has been successfully installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash`

Things worked after this, I was able to compile well. I am using RPi3, Hassio/Buster/addon ESPHome 1.14.3