esphome/issues #806: Text_sensor values are missing in 1.14.0

jarcikw, 2019-11-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip , python 3.7.5

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff T1 (2Gang) x3 , same behaviour
**Affected component:**

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ${name}_Adress_IP
  - platform: version

**Description of problem:**
Values are not displayed in web server. In Home Assistant data are displayed correctly

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  name: sypialnia_t1_switch

esphome:
  name: ${name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

web_server:
  port: 80

# Enable logging
logger:
# Enable Home Assistant API
api:
  password: !secret api_pass
  
ota:
  password: !secret api_pass

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: ${name}_1
    on_press:
      - switch.toggle: ${name}_1

  - platform: gpio
    pin:
      number: GPIO9
      mode: INPUT_PULLUP
      inverted: True
    name: ${name}_2
    on_press:
      - switch.toggle: ${name}_2  

switch:
  - platform: gpio
    name: ${name}_1
    pin: GPIO12
    id: ${name}_1
    icon: mdi:lightbulb
  - platform: gpio
    name: ${name}_2
    pin: GPIO5
    id: ${name}_2
    icon: mdi:lightbulb
  
  - platform: restart
    name: Restart_${name}

status_led:
  pin: GPIO13
  
sensor:  
  - platform: uptime
    name: ${name}_Uptime
    icon: mdi:clock-outline
    update_interval: 300s
    unit_of_measurement: days
    filters:
      - multiply: 0.000011574
    
  - platform: wifi_signal
    name: "${name}_wifi_Signal"
    update_interval: 300s    

    
text_sensor:
  - platform: wifi_info
    ip_address:
  - platform: version
    name: "${name}_ESPHome_Version"

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Screenshots
https://ibb.co/4KsGVJT (version 1.13.6)
https://ibb.co/qMLMf0c (version 1.14.0)

rradar, 2019-11-03:
Can confirm. For me even the binary sensor states are missing :thinking: 

[image]

[image]

jarcikw, 2019-11-05:
Output logs , Version 1.14.2 , Sonoff T1 (2Gang) 
```}
[08:09:05][VV][api.service:150]: send_text_sensor_state_response: TextSensorStateResponse {
  key: 2314437119
  state: '1.14.2 Nov  5 2019, 08:08:09'
}
[08:09:05][VV][api.service:150]: send_text_sensor_state_response: TextSensorStateResponse {
  key: 920115296
  state: '192.168.2.33'
}
[08:09:05][VV][api.service:150]: send_text_sensor_state_response: TextSensorStateResponse {
  key: 2065534755
  state: 'Jarcik'
}
[08:09:05][VV][api.service:150]: send_text_sensor_state_response: TextSensorStateResponse {
  key: 3110980525
  state: 'AC:84:C6:8E:05:17'
}```

But on Sonoff Basic Switch data displayed correctly, please see below

![image](https://user-images.githubusercontent.com/35450305/68186973-b9004480-ffa5-11e9-923e-746756b79653.png)

OttoWinter, 2019-11-05:
@jarcikw I'm not sure I understand. So on another device it does work correctly, is that right? Or does it work for all devices with 1.14.2?

jarcikw, 2019-11-05:
No, it not working on all devices on 1.14.2. on Sonoff Basic worka, but on Sonoff T1 switch it does not working. In  version 1.13.6 was working fine both devices

OttoWinter, 2019-11-06:
I just tried it and it works fine for me with NodeMCU ESP32 and ESP8266.

There have not been any changes to web server since 1.13, so I'm not sure where the issue lies.

I think there are two possibilities:
 - ESPAsyncWebServer library update. The HTTP library used was upgraded to the latest version, maybe they had a regression.
 - CORS issues. Can you open the debug console in your browser while you're viewing the page? I see @rradar you already printed the HTML; if there's something in the console tab that would help.

Also @rradar, in your inspect view all HTML attributes start with `sensor-name_` - the `name_` here is the confusing part for me, as it doesn't match the value shown in the friendly name. Could you share the YAML you used for this device or a snippet that reproduces the behavior?

jarcikw, 2019-11-06:
Nothing suspicious in dev console
[image]

jarcikw, 2019-11-06:
@OttoWinter i have another type of device (my custom , based on ESP12S). And here is partially working, some text sensors values are displayed and some not (BSSID).
[image]
and YAML file

```
substitutions:
  name: 'salon_jarcik_v6_rolety'
  pass: !secret api_pass
  ssid: !secret wifi_ssid
  ssid_pass: !secret wifi_pass
  cover_name: 'Salon'
  ds18b20: '0xCF0315A279419128'
esphome:
  name: ${name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ${ssid}
  password: ${ssid_pass}

web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ${pass}
  
ota:
  password: ${pass}

binary_sensor:
- platform: gpio
  pin:
    number: GPIO16
    inverted: true
  id: up
  on_state:
    then:
      - lambda: |
          if (id(my_cover).current_operation == COVER_OPERATION_IDLE) {
            // Cover is idle, check current state and either open or close cover.
           id(my_cover).open();
          } else {
            id(my_cover).stop();
          }
- platform: gpio
  pin:
    number: GPIO14
    inverted: true    
  id: down
  on_state:
    then:
      - lambda: |
          if (id(my_cover).current_operation == COVER_OPERATION_IDLE) {

           id(my_cover).close();
          } else {
            // Cover is opening/closing. Stop it.
          id(my_cover).stop();
          }
          
switch:
- platform: gpio
  pin: GPIO12
  interlock: &interlock [open_cover, close_cover]
  id: open_cover
- platform: gpio
  pin: GPIO13
  interlock: *interlock
  id: close_cover

- platform: restart
  name: Restart_${name}

cover:
- platform: time_based
  device_class: window
  name: ${cover_name}
  id: my_cover
  open_action:
    - switch.turn_on: open_cover
  open_duration: 22s
  close_action:
    - switch.turn_on: close_cover
  close_duration: 21s
  stop_action:
    - switch.turn_off: open_cover
    - switch.turn_off: close_cover

sensor:  
  - platform: uptime
    name: ${name}_Uptime
    icon: mdi:clock-outline
    update_interval: 300s
    unit_of_measurement: days
    filters:
      - multiply: 0.000011574
    
  - platform: wifi_signal
    name: "${name}_wifi_Signal"
    update_interval: 300s    

  - platform: dallas
    address: ${ds18b20}
    name: ${name}_Temp
    
dallas:
  - pin: GPIO2
    update_interval: 60s
    
text_sensor:
  - platform: wifi_info
    ip_address:
      name: ${name}_Adress_IP
    ssid:
      name: ${name}_SSID    
    bssid:
      name: ${name}_BSSID

  - platform: version
    name: "${name}_ESPHome_Version"  
```

jarcikw, 2019-11-06:
@OttoWinter i have just make some test and when i comment out sensor ds18b20, values are shown. Please have a look 
[image]
and YAML file
```
substitutions:
  name: 'salon_jarcik_v6_rolety'
  pass: !secret api_pass
  ssid: !secret wifi_ssid
  ssid_pass: !secret wifi_pass
  cover_name: 'Salon'
#  ds18b20: '0xCF0315A279419128'
esphome:
  name: ${name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ${ssid}
  password: ${ssid_pass}

web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ${pass}
  
ota:
  password: ${pass}

binary_sensor:
- platform: gpio
  pin:
    number: GPIO16
    inverted: true
  id: up
  on_state:
    then:
      - lambda: |
          if (id(my_cover).current_operation == COVER_OPERATION_IDLE) {
            // Cover is idle, check current state and either open or close cover.
           id(my_cover).open();
          } else {
            id(my_cover).stop();
          }
- platform: gpio
  pin:
    number: GPIO14
    inverted: true    
  id: down
  on_state:
    then:
      - lambda: |
          if (id(my_cover).current_operation == COVER_OPERATION_IDLE) {

           id(my_cover).close();
          } else {
            // Cover is opening/closing. Stop it.
          id(my_cover).stop();
          }
          
switch:
- platform: gpio
  pin: GPIO12
  interlock: &interlock [open_cover, close_cover]
  id: open_cover
- platform: gpio
  pin: GPIO13
  interlock: *interlock
  id: close_cover

- platform: restart
  name: Restart_${name}

cover:
- platform: time_based
  device_class: window
  name: ${cover_name}
  id: my_cover
  open_action:
    - switch.turn_on: open_cover
  open_duration: 22s
  close_action:
    - switch.turn_on: close_cover
  close_duration: 21s
  stop_action:
    - switch.turn_off: open_cover
    - switch.turn_off: close_cover

sensor:  
  - platform: uptime
    name: ${name}_Uptime
    icon: mdi:clock-outline
    update_interval: 300s
    unit_of_measurement: days
    filters:
      - multiply: 0.000011574
    
  - platform: wifi_signal
    name: "${name}_wifi_Signal"
    update_interval: 300s    

#  - platform: dallas
#    address: ${ds18b20}
#    name: ${name}_Temp
    
#dallas:
#  - pin: GPIO2
#    update_interval: 60s
    
text_sensor:
  - platform: wifi_info
    ip_address:
      name: ${name}_Adress_IP
    ssid:
      name: ${name}_SSID    
    bssid:
      name: ${name}_BSSID

  - platform: version
    name: "${name}_ESPHome_Version"  

```
But it does not not explain behaviour on Sonoff T1 switch and Sonoff Basic

rradar, 2019-11-07:
I also just checked. The console only complains about the missing favicon @jarcikw posted already

glmnet, 2020-02-20:
Kind of replicated this issue, seems a problem with the ESPAsyncWebServer it just stops sending the events to the server, code seems to continue execution though.

Then browser reconnects to events but it stops working at the same point.

Does not seem to be related to a sensor type or value, of course log messages stops working too,

Anybody experienced this on ESP32?? Seems like a ESP8266 issue only (memory may be)

glmnet, 2020-04-14:
Kind of fixed it, however I see the events starts failing after the third toggle button pressed.. will investigate more

Please give feedback

glmnet, 2020-04-19:
Also found the issue that the feedback stops comming back after a while, this is related to: https://github.com/me-no-dev/ESPAsyncWebServer/pull/621

I can get it working by commenting this line https://github.com/OttoWinter/ESPAsyncWebServer/blob/ae1258fee1f1e9753bc5cc46aa1d2e67184f6a65/src/AsyncEventSource.cpp#L146

I cannot send a patch for this as the repo used by esphome is controlled by Otto, I believe I could point to an old version os ESPAsyncWebServer insted of Otto's one but I don't know how well that would work.

Also tried to load page and see if all states are retrieved properly... but not, the same behavior as before applying https://github.com/esphome/esphome/pull/1018 so here seems to be two issues.

nkitanov, 2020-06-24:
I do not have it on ESP32 - all sensors are listed with the same config I have for 8266. However I updated esphome to latest recently and didn't tried on ESP8266 to see if it's fixed . Normally on esp8266 I do not enable http server as it gives too much issues.

rradar, 2020-07-28:
@OttoWinter just did a quick check with 1.15.0b2 and the problem is still persistent:

`[I][app:105]: ESPHome version 1.15.0b2 compiled on Jul 28 2020, 14:37:31`

[image]

glmnet, 2020-07-28:
Ohh I tried to improve string handling in https://github.com/esphome/esphome/pull/1018 but the esp crashed when loading the sstream code.

In any case that is just a "performance/memory" issue, not a big deal in my opinion as the json built string shouldn't be that big either, and it's built only for the first request.

This issue has a solution and requires 3 pr merges:

1. https://github.com/esphome/esphome-docs/pull/542
2. https://github.com/esphome/esphome/pull/1018
3. https://github.com/OttoWinter/ESPAsyncTCP/pull/2

closed 2020-07-29 by OttoWinter