esphome/issues #810: mDNS fails after a few seconds an a Sonof T1 UK

bensuffolk, 2019-11-03:
**Hass.io:**

Hass.io running in docker on a raspberry pi 3

ESPHome version 1.14.1

**Sonoff T1 UK:**

**Wifi:**

https://esphome.io/components/wifi.html

**a few seconds after reboot the T1 stops responding to mDNS requests:**

I am slowly working through all my sonof devices migrating them from tasmota to ESPHome and I have just flashed a T1, I noticed after a few seconds it shows as offline on the dashboard.

Using the IP address works if I add it as an integration, and it stay connected and works fine as it should, just no mDNS

I have to specify the use_address configuration to be able to view the logs or upload OTA as the mDNS resolution does not work.

I have tried pinging the T1 via the .local host name and it works just after the device reboots, but a few seconds later it can no longer resolve the name.

Other sonof basics are working fine, so I don’t think it’s a configuration issue on my access point etc.

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: office_t1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True
  manual_ip:
    static_ip: 10.0.0.103
    gateway: 10.0.0.1
    subnet: 255.255.255.0
  use_address: 10.0.0.103
    
logger:

api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    id: button
    on_press:
      then:
        - light.toggle: light_1

  - platform: status
    name: "Office Light Switch Status"

output:
  - platform: gpio
    pin: GPIO12
    id: relay

light:
  - platform: binary
    name: "Office"
    id: light_1
    output: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-11-04:
- Did you add those `fast_connect`, `manual_ip` etc entries after mDNS did not work?

- The observation that it works with most devices but this one is interesting. Could you do an mDNS scan and verify it's not showing any mDNS data (for example with an mDNS chrome extension) - I'd like to eliminate the OS or DNS system as a cause.

bensuffolk, 2019-11-05:
Yes I added the manual_ip after it was failing on mDNS, but I have also tried with issuing a static IP via DHCP server based on the mac address.

The fast_connect is because I'm using a hidden SSID, but I have tried with a visible SSID, and removing the fast_connect.

I have also tried changing the hostname to just office in case it did not like the underscore or the number (although all the other hosts have underscores in them).

I have tried scanning from the PI, and from my mac. They both detect it on reboot, and then a short while later it goes offline and a new scan reveals nothing.

Here is one example scan from the mac looking for esphome devices

```
[suffo_b@Bens-MacBook-Pro-5255 ~]$ dns-sd -B _esphomelib local
Browsing for _esphomelib._tcp.local
DATE: ---Tue 05 Nov 2019---
 8:33:58.091  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
 8:33:58.091  Add        3   6 local.               _esphomelib._tcp.    under_floor_heating
 8:33:58.091  Add        3   6 local.               _esphomelib._tcp.    porch_heater
 8:33:58.091  Add        3   6 local.               _esphomelib._tcp.    kitchen_heater
 8:33:58.091  Add        3   6 local.               _esphomelib._tcp.    conservatory_ls
 8:33:58.091  Add        2   6 local.               _esphomelib._tcp.    office_t1
^C
[suffo_b@Bens-MacBook-Pro-5255 ~]$ dns-sd -B _esphomelib local
Browsing for _esphomelib._tcp.local
DATE: ---Tue 05 Nov 2019---
 8:36:13.238  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
 8:36:13.239  Add        3   6 local.               _esphomelib._tcp.    under_floor_heating
 8:36:13.239  Add        3   6 local.               _esphomelib._tcp.    conservatory_ls
 8:36:13.239  Add        3   6 local.               _esphomelib._tcp.    porch_heater
 8:36:13.239  Add        2   6 local.               _esphomelib._tcp.    kitchen_heater
^C
```
 From the Pi after it can't be detected (removed unrelated entries).
```
root@hass:~# mdns-scan 
+ Home._home-assistant._tcp.local
+ porch_heater._esphomelib._tcp.local
+ under_floor_heating._esphomelib._tcp.local
+ conservatory_ls._esphomelib._tcp.local
+ kitchen_heater._esphomelib._tcp.local
+ Home Assistant Bridge._hap._tcp.local
Browsing ... \           
```
If I leave the scanner running and reboot, it gets added to the list, but then if you stop the scanner and start it again, you get the original list without it in anymore. So I know its not broadcasting any additional services.

Once it has stopped responding to mDNS queries, it continues to respond to pings, and works fine within HASS by its IP address,

OttoWinter, 2019-11-06:
Thanks for testing @bensuffolk.

Unfortunately, it's not evident to me what the error is still. 

> I have also tried changing the hostname to just office in case it did not like the underscore or the number (although all the other hosts have underscores in them).

Just FYI, that's not a problem. Yes in theory it's not a valid hostname, but the HA integration and esphome roll their own mDNS client implementation that accepts underscores.

------

I looked over the codebase and identified one change that could maybe have influenced the mDNS behavior: an upgrade of the default ESP8266 SDK. The newer SDK is using a different mDNS implementation (lea-mdns). Could you check manually using the old SDK again? Thanks

```yaml
esphome:
  # ...
  arduino_version: 1.0.2
```

bensuffolk, 2019-11-06:
```
INFO Reading configuration /config/esphome/office_t1.yaml...
Failed config

esphome: [source /config/esphome/office_t1.yaml:2]
  
  Unfortunately the arduino framework version '{}' is unsupported at this time. You can override this by manually using espressif8266@<platformio version>.
  arduino_version: 1.0.2 [source /config/esphome/office_t1.yaml:2]
  name: office_t1
  platform: ESP8266
  board: esp01_1m
```

OttoWinter, 2019-11-06:
Ah sorry, this is an ESP8266 not ESP32, it should be 

```yaml
esphome:
  # ...
  arduino_version: 2.4.2
```

bensuffolk, 2019-11-06:
I’ll have to try the scanner when I’m at my computer,  but I have done this via the web interface and it seems to be worse, it never shows as online at all.

bensuffolk, 2019-11-07:
I had a spare T2 in a box that I was not using (Same age as the T1 - About 18 months). I flashed that and got the same results so its not a one off faulty board. Its clearly related to the specific processor / hardware combination in the T series.

I also did some more scanning, and when using 2.4.2 framework there is no mDNS response at all, so the newer lea-mdns does make an improvement, just only for a 30 seconds or so.

cybershoe, 2022-01-28:
Resurrecting this issue, seeing similar behavior on an Olimex ESP32-PoE-ISO board: mDNS works immediately after boot, fails 30 seconds or so later. Problem does not occur when using Ethernet, only WiFi. This board uses an ESP32-WROOM-32E module.