esphome/issues #811: Strobe effect doesn't work with binary light

woju, 2019-11-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Debian 10, pip3, 1.14.0

**ESP (ESP32/ESP8266, Board/Sonoff):**

`d1_mini`, but it doesn't matter

**Affected component:**

https://esphome.io/components/light/binary.html
https://esphome.io/components/light/index.html#strobe-effect

**Description of problem:**
Strobe effect does not work with binary lights. #720 was closed prematurely, because esphome/esphome#749 did not fix the underlying issue (and I cannot comment on either of those, because at the time of writing those are locked).

The issue is because when strobe is supposed to turn output off, it does `set_state(true)`:
https://github.com/esphome/esphome/blob/dev/esphome/components/light/base_light_effects.h#L106

And in turn this causes the output to be on:
https://github.com/esphome/esphome/blob/dev/esphome/components/light/light_color_values.h#L173
https://github.com/esphome/esphome/blob/dev/esphome/components/binary/light/binary_light_output.h#L22

**Problem-relevant YAML-configuration entries:**
```yaml
# also see #720, equally applicable
esphome:
  name: ...
  platform: ESP8266
  board: d1_mini

light:
  - platform: binary
    id: light_id
    name: ...
    output: gpio_d2
    effects:
      - strobe:
          name: Strobe
          colors:
            - state: True
              duration: 100ms
            - state: False
              duration: 100ms

output:
  - platform: gpio
    id: gpio_d2
    pin: D2

mqtt:
  on_message:
    topic: ...
    then:
      - light.control:
          id: light_id
          state: !lambda |-
            return x != "off";
          effect: !lambda |-
            return x == "0" ? "None" : "Strobe";

```

**Logs (if applicable):**

Nothing really wrong:
```
[D][light:264]: 'XFD' Setting:
[D][light:273]:   State: OFF
[D][light:326]:   Effect: 'None'
[D][light:264]: 'XFD' Setting:
[D][light:273]:   State: ON
[D][light:326]:   Effect: 'Strobe'
```

**Additional information and things you've tried:**

cc @MaximumSU

tomlut, 2019-12-01:
Just discovered I have the same issue with an ESP32.

```
light:
  - platform: binary
    name: "MB Red LED"
    id: light_red_led
    output: mb_red_led
    effects:
      - strobe:
          name: blink_slow
          colors:
            - state: true
              duration: 1s
            - state: false
              duration: 1s

output:
  - platform: gpio
    id: mb_red_led
    pin:
      number: GPIO26
```
```
[14:34:03][I][app:100]: ESPHome version 1.14.3 compiled on Dec  1 2019, 14:28:48
[14:34:03][C][wifi:415]: WiFi:
[14:34:03][C][wifi:283]:   SSID: 

[redacted]
[14:34:03][C][wifi:284]:   IP Address: 10.1.1.90
[14:34:03][C][wifi:286]:   BSSID: 

[redacted]
[14:34:03][C][wifi:287]:   Hostname: 'master_bed_dht'
[14:34:03][C][wifi:291]:   Signal strength: -51 dB ▂▄▆█
[14:34:03][C][wifi:295]:   Channel: 1
[14:34:03][C][wifi:296]:   Subnet: 255.255.255.0
[14:34:03][C][wifi:297]:   Gateway: 10.1.1.1
[14:34:03][C][wifi:298]:   DNS1: 0.0.0.0
[14:34:03][C][wifi:299]:   DNS2: 0.0.0.0
[14:34:03][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Master Bed Lamp Switch'
[14:34:03][C][gpio.binary_sensor:016]:   Pin: GPIO25 (Mode: INPUT_PULLUP, INVERTED)
[14:34:03][C][gpio.output:010]: GPIO Binary Output:
[14:34:03][C][gpio.output:011]:   Pin: GPIO26 (Mode: OUTPUT)
[14:34:04][C][gpio.output:010]: GPIO Binary Output:
[14:34:04][C][gpio.output:011]:   Pin: GPIO18 (Mode: OUTPUT)
[14:34:04][C][gpio.output:010]: GPIO Binary Output:
[14:34:04][C][gpio.output:011]:   Pin: GPIO19 (Mode: OUTPUT)
[14:34:04][C][logger:175]: Logger:
[14:34:04][C][logger:176]:   Level: DEBUG
[14:34:04][C][logger:177]:   Log Baud Rate: 115200
[14:34:04][C][logger:178]:   Hardware UART: UART0
[14:34:04][C][light:177]: Light 'MB Red LED'
[14:34:04][C][light:177]: Light 'MB Green LED'
[14:34:04][C][light:177]: Light 'MB Blue LED'
[14:34:04][C][status:034]: Status Binary Sensor 'Master Bedroom DHT Status'
[14:34:04][C][status:034]:   Device Class: 'connectivity'
[14:34:04][C][dht:017]: DHT:
[14:34:04][C][dht:018]:   Pin: GPIO27 (Mode: INPUT)
[14:34:04][C][dht:024]:   Model: DHT22 (or equivalent)
[14:34:04][C][dht:027]:   Update Interval: 15.0s
[14:34:04][C][dht:029]:   Temperature 'Master Bedroom Temperature'
[14:34:04][C][dht:029]:     Unit of Measurement: '°C'
[14:34:04][C][dht:029]:     Accuracy Decimals: 1
[14:34:04][C][dht:029]:     Icon: 'mdi:thermometer'
[14:34:04][C][dht:030]:   Humidity 'Master Bedroom Humidity'
[14:34:04][C][dht:030]:     Unit of Measurement: '%'
[14:34:04][C][dht:030]:     Accuracy Decimals: 0
[14:34:04][C][dht:030]:     Icon: 'mdi:water-percent'
[14:34:04][C][restart:021]: Restart Switch 'Master Bed DHT Restart'
[14:34:04][C][restart:021]:   Icon: 'mdi:restart'
[14:34:04][C][ota:029]: Over-The-Air Updates:
[14:34:04][C][ota:030]:   Address: 10.1.1.90:3232
[14:34:04][C][ota:032]:   Using Password.
[14:34:04][C][api:095]: API Server:
[14:34:04][C][api:096]:   Address: 10.1.1.90:6053
[14:34:04][C][wifi_signal.sensor:009]: WiFi Signal 'Master Bedroom DHT WiFi Signal'
[14:34:04][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[14:34:04][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[14:34:04][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[14:34:09][D][dht:048]: Got Temperature=19.4°C Humidity=55.5%
[14:34:13][D][light:264]: 'MB Red LED' Setting:
[14:34:13][D][light:326]:   Effect: 'blink_slow'
[14:34:17][D][light:264]: 'MB Red LED' Setting:
[14:34:17][D][light:326]:   Effect: 'None'
```

The light works (can be switched on and off) and the effect is selectable in the UI but does nothing (the light stays on).

Seems there was a previous issue too : https://github.com/esphome/issues/issues/720

Fabi0San, 2020-10-08:
v1.15.2 still reproduces this issue despite the related fix to issue #720 being part of it.

kquinsland, 2021-02-01:
I can confirm this is still an issue w/ `v1.15.3`

```yaml
light:
  - id: light_red
    platform: binary
    name: "Red (binary)"
    output: out_red
    restore_mode: ALWAYS_OFF
    # ESPHome does not support Instant on/off, apparently :/
    #default_transition_length: 0s
    effects:
      - strobe:
```

With the above config, the light does auto register w/ HA over MQTT, but HA does not show the effect drop down like it does on other bulbs. When i manually call the `light.turn_on` service from HA w/ `effect: strobe` in the data, the light only turns on and produces this in the logs:

```
[D][light:265]: 'Red (binary light)' Setting:
[D][light:274]:   State: ON
[D][light:327]:   Effect: 'Strobe'
```

The outcome (calling `turn_on` with `effect: strobe` just turns the lamp on) is still the same even when i explicitly add the true/false + duration settings:

```yaml
light:
  - id: light_red
    platform: binary
    name: "Red (binary light)"
    output: out_red
    restore_mode: ALWAYS_OFF
    # ESPHome does not support Instant on/off, apparently :/
    #default_transition_length: 0s
    effects:
      - strobe:
          colors:
            - state: true
              duration: 1s
            - state: false
              duration: 1s
```

atlflyer, 2021-05-21:
I was able to work around this with a template switch. Posting here in case it's helpful for anyone else.

```yaml
globals:
  - id: my_state
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: my_state_f
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: my_state_vf
    type: bool
    restore_value: no
    initial_value: 'false'    
    
light:
  - platform: binary
    id: led_light
    name: "LED Light"
    output: light_output

switch:
  - platform: template
    name: "LED Blink"
    id: led_blink
    lambda: |-
      return id(my_state);
    turn_on_action:
      - lambda: |-
          id(my_state) = true;
      - while: 
          condition: 
            lambda: |-
              return id(my_state);
          then:
            - light.toggle: led_light
            - delay: 1s
    turn_off_action:
      - lambda: |-
          id(my_state) = false;
      - light.turn_off: led_light
  - platform: template
    name: "LED Fast Blink"
    id: led_fast_blink
    lambda: |-
      return id(my_state_f);
    turn_on_action:
      - lambda: |-
          id(my_state_f) = true;
      - while: 
          condition: 
            lambda: |-
              return id(my_state_f);
          then:
            - light.toggle: led_light
            - delay: 500ms
    turn_off_action:
      - lambda: |-
          id(my_state_f) = false;    
      - light.turn_off: led_light
  - platform: template
    name: "LED Very Fast Blink"
    id: led_vf_blink
    lambda: |-
      return id(my_state_vf);
    turn_on_action:
      - lambda: |-
          id(my_state_vf) = true;
      - while: 
          condition: 
            lambda: |-
              return id(my_state_vf);
          then:
            - light.toggle: led_light
            - delay: 150ms
    turn_off_action:
      - lambda: |-
          id(my_state_vf) = false;    
      - light.turn_off: led_light    

output:
  - id: light_output
    platform: gpio
    pin: GPIO2
```

andriej, 2021-07-26:
Quite-a-shame this issue is still here since 2019 - harsh comment just after loosing 1 hour to get LED on d1 mini blinking as light in HA... :-(

Pe-MaKer, 2021-09-10:
Im trying to let my binary light (which is actually the the relay of the power supply of my LED strip) blink by using an effect. Can it be that hard to create or strip down one of the existing effects to just have to set the pulse on/off times?
Since this issue is open now open for almost two years, it seems to be rocket science.

sjwright, 2021-09-10:
I've "solved" this enough for my purposes by using a monochromatic light (instead of binary) with gamma of 1, and connecting it to an output of type slow_pwm. This gives you a dimmer that always flashes at the rate you see here, but with a variable duty cycle.

I use this with period:5min to give variable control to a dumb electric heater.

```
output:
  - platform: slow_pwm
    id: blah
    pin: 
    period: 5sec

light:
  - platform: monochromatic
    name: 
    output: blah
    gamma_correct: 1
    default_transition_length: 0s
```

andriej, 2021-09-10:
Nice workaround, but totally against simplicity for a new (and not only new) users, which is not mentioned by docs that it's not possible to achieve...
So either docs are to be corrected or bug to be fixed

sjwright, 2021-09-11:
I'm so sorry for offending you, I just thought someone would find it useful.

andriej, 2021-09-11:
OT: you did not offend me at any point and I'd probably use your code if I ever find it... 
But to find it you have to try implementing something that's mentioned in the docs, fail totally, come here to report issue, search for similar one and see that there's one - since quite long time without clear reason for that. :-) 
I regret little level of knowledge in coding as I'd fix that right away so many people don't waste their time running that. Maybe it's easier to remove "effect" from binary lights documentation so it doesn't suggest using it

woju, 2021-09-13:
On Fri, Sep 10, 2021 at 09:12:49PM -0700, Andrzej wrote:
> I regret little level of knowledge in coding as I'd fix that right away so
> many people don't waste their time running that.

Please do. In the meantime, I think maintainers of this piece of free software
owe me exactly nothing, not even correctness of the documentation. So I'm not
complaining about perceived lack of timeliness in this issue. I'm still
patiently waiting.

closed 2021-09-14 by jesserockz

jesserockz, 2021-09-14:
Sorry for this issue being not sorted out earlier. I have pushed a fix into the latest ESPHome 2021.9.0b4 beta release release that fixes this.

kquinsland, 2021-09-14:
Nothing to be sorry for. Thank you for getting to it!