esphome/issues #812: Can't use TX and RX (Wemos d1 Mini) as GPIO

HA-TB303, 2019-11-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 0.101.2, esphome 1.14.2
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini

**Affected component:**

Rotary encoder

**Description of problem:**
I have a Wemos D1 mini with 3 rotary encoders connected. I can't use TX and RX as GPIO.
I am not 100% sure this worked before, but as far as I know this worked before. I have a Wemos D1 mini running espeasy with 3 rotary encoders without issues.

Rotary 3 (which is using GPIO1 and GPIO3) is not working, the encoder works. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: rotary01
  ## Woonkamer rotary controllers ingang
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret esphome_wifi_ssid
  password: !secret esphome_wifi_password
  
api:
  password: !secret esphome_api_password

ota:
  password: !secret esphome_ota_password

logger:

sensor:
### Rotary encoder 1
  - platform: rotary_encoder
    name: "Woonkamer 1 Rotary 1"
    id: woonkamer1rotary1
    pin_a: GPIO12
    pin_b: GPIO13
    filters:
      - or:
        - debounce: 0.2s
        - delta: 5
    resolution: 1
    min_value: 1
    max_value: 10
    on_value:
      then:
        - homeassistant.service:
            service: light.turn_on
            data_template:
              entity_id: light.woonkamer_groep
              brightness_pct: "{{ brightness_1 | int }}"
            variables:
              brightness_1: 'return id(woonkamer1rotary1).state * 10;'
       
### Rotary encoder 2
  - platform: rotary_encoder
    name: "Woonkamer 1 Rotary 2"
    id: woonkamer1rotary2
    pin_a: GPIO5
    pin_b: GPIO4
    filters:
      - or:
        - debounce: 0.1s
        - delta: 3
    resolution: 1
    min_value: 1
    max_value: 25
    on_value:
      then:
        - homeassistant.service:
            service: light.turn_on
            data_template:
              entity_id: light.eettafel
              brightness_pct: "{{ brightness_1 | int }}"
            variables:
              brightness_1: 'return id(woonkamer1rotary2).state * 4;'
              
### Rotary encoder 3
  - platform: rotary_encoder
    name: "Woonkamer 1 Rotary 3"
    id: woonkamer1rotary3
    pin_a: GPIO1
    pin_b: GPIO3
    filters:
      - or:
        - debounce: 0.1s
        - delta: 3
    resolution: 1
    min_value: 1
    max_value: 25
    on_value:
      then:
        - homeassistant.service:
            service: light.turn_on
            data_template:
              entity_id: light.booglamp
              brightness_pct: "{{ brightness_1 | int }}"
            variables:
              brightness_1: 'return id(woonkamer1rotary3).state * 4;'

binary_sensor:
### Switch 1
  - platform: gpio
    pin: GPIO14
    id: switch1
    name: "Woonkamer 1 switch 1"        
    filters:
      - invert:   
    on_multi_click:
    - timing:
        - ON for at most 0.4s
        - OFF for at most 0.4s
        - ON for at most 0.4s
        - OFF for at least 0.2s
      then:
        - logger.log: "Double Clicked"
        - homeassistant.service:
            service: input_select.select_next
            data:
              entity_id: input_select.hue_scenes
    - timing:
        - ON for 0.6s to 3s
        - OFF for at least 0.3s
      then:
        - logger.log: "Single Long Clicked"
        - homeassistant.service:
            service: input_boolean.toggle
            data:
              entity_id: input_boolean.radio_woonkamer
    - timing:
        - ON for at most 0.5s
        - OFF for at least 0.4s
      then:
        - logger.log: "Single short Click"
        - homeassistant.service:
            service: light.toggle
            data:
              entity_id: light.woonkamer_groep
            
### Switch 2
  - platform: gpio
    pin: GPIO0
    id: switch2
    name: "Woonkamer 1 switch 2"        
    filters:
      - invert:   
    on_multi_click:
    - timing:
        - ON for at most 0.4s
        - OFF for at most 0.4s
        - ON for at most 0.4s
        - OFF for at least 0.2s
      then:
        - logger.log: "Double Clicked"
        - homeassistant.service:
            service: hue.hue_activate_scene
            data:
              group_name: "Woonkamer groep"
              scene_name: "Film"
    - timing:
        - ON for 0.6s to 3s
        - OFF for at least 0.3s
      then:
        - logger.log: "Single Long Clicked"
        - homeassistant.service:
            service: input_boolean.toggle
            data:
              entity_id: input_boolean.radio_woonkamer
    - timing:
        - ON for at most 0.5s
        - OFF for at least 0.4s
      then:
        - logger.log: "Single short Click"
        - homeassistant.service:
            service: light.toggle
            data:
              entity_id: light.eettafel
              
### Switch 3
  - platform: gpio
    pin: GPIO16
    id: switch3
    name: "Woonkamer 1 switch 3"        
    filters:
      - invert:   
    on_multi_click:
    - timing:
        - ON for at most 0.4s
        - OFF for at most 0.4s
        - ON for at most 0.4s
        - OFF for at least 0.2s
      then:
        - logger.log: "Double Clicked"
        - homeassistant.service:
            service: hue.hue_activate_scene
            data:
              group_name: "Woonkamer groep"
              scene_name: "Film"
    - timing:
        - ON for 0.6s to 3s
        - OFF for at least 0.3s
      then:
        - logger.log: "Single Long Clicked"
        - homeassistant.service:
            service: input_boolean.toggle
            data:
              entity_id: input_boolean.radio_woonkamer
    - timing:
        - ON for at most 0.5s
        - OFF for at least 0.4s
      then:
        - logger.log: "Single short Click"
        - homeassistant.service:
            service: light.toggle
            data:
              entity_id: light.booglamp
```

**Logs (if applicable):**

**Additional information and things you've tried:**

rradar, 2019-11-03:
Does it compile? No logs?

HA-TB303, 2019-11-03:
Yes it compiles, just nothing happens on rotary encoder 3 (the one that uses gpio1 tx and gpio3 rx), so there is no relevant log either.

randybb, 2019-11-03:
Did you try to disable logging to UART? https://esphome.io/components/logger.html

brandond, 2019-11-04:
@randybb has it right, if you want to use the UART GPIOs for something other than logging then you need to tell ESPHome not to log to them.

HA-TB303, 2019-11-04:
I added the logger baud_rate
On bootup:
```
[08:56:57][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Woonkamer 1 switch 1'
[08:56:57][C][gpio.binary_sensor:016]:   Pin: GPIO14 (Mode: INPUT)
[08:56:57][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Woonkamer 1 switch 2'
[08:56:57][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT)
[08:56:57][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Woonkamer 1 switch 3'
[08:56:57][C][gpio.binary_sensor:016]:   Pin: GPIO16 (Mode: INPUT)
[08:56:58][C][logger:175]: Logger:
[08:56:58][C][logger:176]:   Level: DEBUG
[08:56:58][C][logger:177]:   Log Baud Rate: 0
[08:56:58][C][logger:178]:   Hardware UART: UART0
[08:56:58][C][rotary_encoder:115]: Rotary Encoder 'Woonkamer 1 Rotary 1'
[08:56:58][C][rotary_encoder:115]:   Unit of Measurement: 'steps'
[08:56:58][C][rotary_encoder:115]:   Accuracy Decimals: 0
[08:56:58][C][rotary_encoder:115]:   Icon: 'mdi:rotate-right'
[08:56:58][C][rotary_encoder:116]:   Pin A: GPIO12 (Mode: INPUT)
[08:56:58][C][rotary_encoder:117]:   Pin B: GPIO13 (Mode: INPUT)
[08:56:58][C][rotary_encoder:121]:   Resolution: 1 Pulse Per Cycle
[08:56:58][C][rotary_encoder:115]: Rotary Encoder 'Woonkamer 1 Rotary 2'
[08:56:58][C][rotary_encoder:115]:   Unit of Measurement: 'steps'
[08:56:58][C][rotary_encoder:115]:   Accuracy Decimals: 0
[08:56:58][C][rotary_encoder:115]:   Icon: 'mdi:rotate-right'
[08:56:58][C][rotary_encoder:116]:   Pin A: GPIO5 (Mode: INPUT)
[08:56:58][C][rotary_encoder:117]:   Pin B: GPIO4 (Mode: INPUT)
[08:56:58][C][rotary_encoder:121]:   Resolution: 1 Pulse Per Cycle
[08:56:58][C][rotary_encoder:115]: Rotary Encoder 'Woonkamer 1 Rotary 3'
[08:56:58][C][rotary_encoder:115]:   Unit of Measurement: 'steps'
[08:56:58][C][rotary_encoder:115]:   Accuracy Decimals: 0
[08:56:58][C][rotary_encoder:115]:   Icon: 'mdi:rotate-right'
[08:56:58][C][rotary_encoder:116]:   Pin A: GPIO1 (Mode: INPUT)
[08:56:58][C][rotary_encoder:117]:   Pin B: GPIO3 (Mode: INPUT)
[08:56:58][C][rotary_encoder:121]:   Resolution: 1 Pulse Per Cycle
```
But rotary 3 still doesn't work.

HA-TB303, 2019-11-04:
Just to be sure it is not a hardware issue, I will try another wemos d1 mini tonight with a rotary encoder on GPIO1 en 3.

brandond, 2019-11-04:
Since the D1 Mini has a USB-TTL converter on it, those may be holding the pins high. I'm not convinced you can safely use the serial lines for anything else without somehow disabling the existing connections.

HA-TB303, 2019-11-04:
It's not the wemos, it turns out to be my new encoders. I switched those a few days ago because the new ones are a better fit, but it now turns out these do NOT work on GPIO1 and 3, while the old ones do:

Old:
```
[20:54:48][C][logger:175]: Logger:
[20:54:48][C][logger:176]:   Level: VERBOSE
[20:54:48][C][logger:177]:   Log Baud Rate: 0
[20:54:48][C][logger:178]:   Hardware UART: UART0
[20:54:48][C][rotary_encoder:115]: Rotary Encoder 'Woonkamer 1 Rotary 4'
[20:54:48][C][rotary_encoder:115]:   Unit of Measurement: 'steps'
[20:54:48][C][rotary_encoder:115]:   Accuracy Decimals: 0
[20:54:48][C][rotary_encoder:115]:   Icon: 'mdi:rotate-right'
[20:54:48][C][rotary_encoder:116]:   Pin A: GPIO1 (Mode: INPUT)
[20:54:48][C][rotary_encoder:117]:   Pin B: GPIO3 (Mode: INPUT)
[20:54:48][C][rotary_encoder:121]:   Resolution: 1 Pulse Per Cycle
[20:54:48][C][ota:029]: Over-The-Air Updates:
[20:54:48][C][ota:030]:   Address: esp_rotary_2.local:8266
[20:54:48][C][ota:032]:   Using Password.
[20:54:48][C][api:095]: API Server:
[20:54:48][C][api:096]:   Address: esp_rotary_2.local:6053
[20:54:51][V][sensor:013]: 'Woonkamer 1 Rotary 4': Received new state 17.000000
[20:54:51][D][sensor:092]: 'Woonkamer 1 Rotary 4': Sending state 17.00000 steps with 0 decimals of accuracy
[20:54:52][V][sensor:013]: 'Woonkamer 1 Rotary 4': Received new state 16.000000
[20:54:52][D][sensor:092]: 'Woonkamer 1 Rotary 4': Sending state 16.00000 steps with 0 decimals of accuracy
[20:54:52][V][sensor:013]: 'Woonkamer 1 Rotary 4': Received new state 17.000000
[20:54:52][V][sensor:013]: 'Woonkamer 1 Rotary 4': Received new state 18.000000
[20:54:53][D][sensor:092]: 'Woonkamer 1 Rotary 4': Sending state 18.00000 steps with 0 decimals of accuracy
```

Old one: (working on all gpio)
[image]

New one: (not working on gpio1/3, but does work on other gpio)
[image]

I hope someone will be able to tell me the logic of this all before the issue will be rightfully closed :)

OttoWinter, 2019-11-04:
Hard to say - my best guess would be that it's got something to do with the FTDI adapter. Even if the ESP UART interface is disabled, the FTDI adapter is still attached to those pins and can do all sorts of stuff. Probably the pullups/whatever is not working too well with the encoder.

OttoWinter, 2019-11-04:
As the issue is resolved, I'm closing. If someone knows another explanation, please add it here - anyway I don't think this is a software bug.

closed 2019-11-04 by OttoWinter