esphome/issues #814: Increated baud rate since 1.14.0 causing 'Timed out waiting for packet header' on serial uploads

heckler, 2019-11-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- ESPHome version 1.14.0
- Windows 10 version 1809 (build 17763.805)
- Windows System for Linux (WSL) - Ubuntu 18.04.2 LTS
- NodeMCU UART: CP1210x, Silicon Labs, VID 10C4, PID EA60

**ESP (ESP32/ESP8266, Board/Sonoff):**

- ESP8266, NodeMCU V2, generic module from Aliexpress

**Affected component:**

- Upload via serial port is affected by the higher default baud rate
- Relates to [PR-761](https://github.com/esphome/esphome/pull/761)

**Description of problem:**

Some NodeMCU v2 boards from Aliexpress I have here are no longer taking firmware uploads via serial with the new default baud rate of 460800.  The upload fails with the following error message:

`A fatal error occurred: Timed out waiting for packet header`

**Logs (if applicable):**

Attempting to upload with the default baud reate of 460800 results in this error:

```
$ esphome pir_test.yaml upload --upload-port=/dev/ttyS4
INFO Reading configuration pir_test.yaml...
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/ttyS4 write_flash 0x0 pir_test/.pioenvs/pir_test/firmware.bin
esptool.py v2.7
Serial port /dev/ttyS4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:ac:6c:7b
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
```

**Additional information and things you've tried:**

If I manually write the same firmware using esptool.py and setting baud rate to 115200, upload succeeds:

```
$ /home/heckler/.local/bin/esptool.py --before default_reset --after hard_reset --baud 115200 --chip esp8266 --port /dev/ttyS4 write_flash 0x0 pir_test/.pioenvs/pir_test/firmware.bin
esptool.py v2.7
Serial port /dev/ttyS4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:ac:6c:7b
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 417264 bytes to 283230...
Wrote 417264 bytes (283230 compressed) at 0x00000000 in 25.1 seconds (effective 133.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

After uploading I can run `esphome pir_test.yaml logs` and confirm the new firmware booted and ran fine:

```
INFO Successfully connected to pir_test.local
[17:09:35][I][app:100]: ESPHome version 1.14.0 compiled on Nov  3 2019, 16:37:10
[17:09:35][C][status_led:019]: Status LED:
[17:09:35][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT, INVERTED)
[17:09:35][C][wifi:409]: WiFi:
[17:09:35][C][wifi:277]:   SSID: 'QATSI-2'
[17:09:35][C][wifi:278]:   IP Address: 192.168.6.126
[17:09:35][C][wifi:280]:   BSSID: C4:E9:84:BC:D9:A0
[17:09:35][C][wifi:281]:   Hostname: 'pir_test'
[17:09:35][C][wifi:285]:   Signal strength: -55 dB ▂▄▆█
[17:09:35][C][wifi:289]:   Channel: 11
[17:09:35][C][wifi:290]:   Subnet: 255.255.255.0
[17:09:35][C][wifi:291]:   Gateway: 192.168.6.1
[17:09:35][C][wifi:292]:   DNS1: 192.168.6.1
[17:09:35][C][wifi:293]:   DNS2: 8.8.8.8
[17:09:35][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PIR Sensor'
[17:09:35][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[17:09:35][C][gpio.binary_sensor:016]:   Pin: GPIO13 (Mode: INPUT)
[17:09:35][C][uptime.sensor:030]: Uptime Sensor 'PIR Uptime'
[17:09:35][C][uptime.sensor:030]:   Unit of Measurement: 's'
[17:09:35][C][uptime.sensor:030]:   Accuracy Decimals: 0
[17:09:35][C][uptime.sensor:030]:   Icon: 'mdi:timer'
(...)
```

I also tried adding a 10uF electrolytic capacitor between +3.3V and GND and between EN and GND, but it made no difference.  Only succeeded in uploading the firmware with the lower baud rate set in esptool.py

heckler, 2019-11-04:
In case this adds more clarification: I've tried the upload on a different device, a so-called NodeMCU "LoLin"; this one uses a CH34 USB-Serial which reports VID 1A86 and PID 7523.

The error message on upload is slightly different this time:

```
$ esphome pir_test.yaml run --upload-port=/dev/ttyS5
INFO Reading configuration pir_test.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d pir_test
Processing pir_test (board: nodemcuv2; framework: arduino; platform: espressif8266@2.2.3)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.5
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/pir_test/src/main.cpp.o
Linking .pioenvs/pir_test/firmware.elf
Building .pioenvs/pir_test/firmware.bin
Retrieving maximum program size .pioenvs/pir_test/firmware.elf
Checking size .pioenvs/pir_test/firmware.elf
DATA:    [====      ]  44.8% (used 36660 bytes from 81920 bytes)
PROGRAM: [====      ]  39.8% (used 415416 bytes from 1044464 bytes)
Creating BIN file ".pioenvs/pir_test/firmware.bin" using ".pioenvs/pir_test/firmware.elf"
========================================================================================================= [SUCCESS] Took 6.63 seconds ========================================================================================================= INFO Successfully compiled program.
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/ttyS5 write_flash 0x0 pir_test/.pioenvs/pir_test/firmware.bin
esptool.py v2.7
Serial port /dev/ttyS5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 68:c6:3a:80:98:67
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
ERROR Running command failed: (5, 'Input/output error')
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/ttyS5 write_flash 0x0 pir_test/.pioenvs/pir_test/firmware.bin locally.
```

Once again, using esptool.py to write the firmware at 115200 baud work just fine.

OttoWinter, 2019-11-04:
Ugh, I was under the impression 460800 was safe because the esp8266/Arduino guys merged a similar PR saying 460800 works well on all systems.

Apparently not.

I really like the faster upload speed, maybe we can work around it by first trying with 460800 and switching to 115200 if that fails.

nbergont, 2019-11-08:
For information, some USB Serial Adapter are crappy.
See more here :
https://vilimpoc.org/blog/2016/05/04/esptool-usb-serial-adapter-shootout/

brandond, 2019-11-08:
Yeah, just for the record I've had no problems at all with the increased rate using a cp2104 adapter.

OttoWinter, 2019-11-09:
@nbergont Thanks for the link! Good to have some empirical data on this.

Yeah I know there are some crappy FTDIs, however I read in some esp8266/Arduino thread that 460800 is a safe choice even for bad ones - but apparently that was wrong :(

closed 2019-11-14 by OttoWinter