esphome/issues #815: MHZ19 won't connect

dedepene, 2019-11-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hassio add-on
**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU, ESP32
**Affected component:**

https://esphome.io/components/sensor/mhz19.html

**Description of problem:**
The CO2 sensor won't connect on a number of controllers - Nodemcus and ESP32. The very same sensor works fine on just this one NodeMCU with the exacts same settings across the board. Super weird - how do I even begin to troubleshoot?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```
esphome:
  name: office
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "xxx"
  password: "xxx"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

uart:
  rx_pin: GPIO1
  tx_pin: GPIO3
  baud_rate: 9600

sensor:
  - platform: mhz19
    co2:
      name: "Office CO2 Value"
    temperature:
      name: "Office Temperature"
    update_interval: 60s
    automatic_baseline_calibration: false

i2c:
  sda: 4
  scl: 5
  scan: True
  
apds9960:
  address: 0x39
  update_interval: 60s

binary_sensor:
  - platform: apds9960
    direction: UP
    name: "APDS960 Up Movement"
    #filters:
    #  - delayed_off: 1500ms
  - platform: apds9960
    direction: down
    name: "APDS960 Down Movement"
    #filters:
    #  - delayed_off: 1500ms
  - platform: apds9960
    direction: left
    name: "APDS960 Left Movement"
    #filters:
    #  - delayed_off: 1500ms
  - platform: apds9960
    direction: right
    name: "APDS960 Right Movement"
    #filters:
    #  - delayed_off: 1500ms
    
switch:
  - platform: restart
    name: "Office Nodemcu Restart"
```

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
PASTE DEBUG LOG HERE

```
[13:50:08][I][app:100]: ESPHome version 1.14.2 compiled on Nov  4 2019, 12:32:03
[13:50:08][C][wifi:409]: WiFi:
[13:50:08][C][wifi:277]:   SSID: [redacted]
[13:50:08][C][wifi:278]:   IP Address: 192.168.1.144
[13:50:08][C][wifi:280]:   BSSID: [redacted]
[13:50:08][C][wifi:281]:   Hostname: 'office'
[13:50:08][C][wifi:285]:   Signal strength: -58 dB ▂▄▆█
[13:50:08][C][wifi:289]:   Channel: 1
[13:50:08][C][wifi:290]:   Subnet: 255.255.255.0
[13:50:08][C][wifi:291]:   Gateway: 192.168.1.1
[13:50:08][C][wifi:292]:   DNS1: 54.93.173.153
[13:50:08][C][wifi:293]:   DNS2: 46.166.189.68
[13:50:08][C][i2c:028]: I2C Bus:
[13:50:08][C][i2c:029]:   SDA Pin: GPIO4
[13:50:08][C][i2c:030]:   SCL Pin: GPIO5
[13:50:08][C][i2c:031]:   Frequency: 50000 Hz
[13:50:08][I][i2c:033]: Scanning i2c bus for active devices...
[13:50:08][I][i2c:040]: Found i2c device at address 0x39
[13:50:09][C][uart:145]: UART Bus:
[13:50:09][C][uart:147]:   TX Pin: GPIO3
[13:50:09][C][uart:150]:   RX Pin: GPIO1
[13:50:09][C][uart:152]:   Baud Rate: 9600 baud
[13:50:09][C][uart:153]:   Stop bits: 1
[13:50:09][C][uart:157]:   Using software serial
[13:50:09][C][logger:175]: Logger:
[13:50:09][C][logger:176]:   Level: DEBUG
[13:50:09][C][logger:177]:   Log Baud Rate: 115200
[13:50:09][C][logger:178]:   Hardware UART: UART0
[13:50:09][C][mhz19:094]: MH-Z19:
[13:50:09][C][mhz19:095]:   CO2 'Office CO2 Value'
[13:50:09][C][mhz19:095]:     Unit of Measurement: 'ppm'
[13:50:09][C][mhz19:095]:     Accuracy Decimals: 0
[13:50:09][C][mhz19:095]:     Icon: 'mdi:periodic-table-co2'
[13:50:09][C][mhz19:096]:   Temperature 'Office Temperature'
[13:50:09][C][mhz19:096]:     Unit of Measurement: '°C'
[13:50:09][C][mhz19:096]:     Accuracy Decimals: 0
[13:50:09][C][mhz19:096]:     Icon: 'mdi:thermometer'
[13:50:09][C][mhz19:102]:   Automatic baseline calibration disabled on boot
[13:50:09][C][apds9960:124]: APDS9960:
[13:50:09][C][apds9960:125]:   Address: 0x39
[13:50:09][C][apds9960:127]:   Update Interval: 60.0s
[13:50:09][C][restart:021]: Restart Switch 'Office Nodemcu Restart'
[13:50:09][C][restart:021]:   Icon: 'mdi:restart'
[13:50:09][C][ota:029]: Over-The-Air Updates:
[13:50:09][C][ota:030]:   Address: office.local:8266
[13:50:09][C][api:095]: API Server:
[13:50:09][C][api:096]:   Address: office.local:6053
[13:50:20][E][uart:240]: Reading from UART timed out at byte 0!
[13:50:20][W][mhz19:034]: Reading data from MHZ19 failed!
[13:51:20][E][uart:240]: Reading from UART timed out at byte 0!
[13:51:20][W][mhz19:034]: Reading data from MHZ19 failed!
[13:51:31][D][apds9960:282]: Got gesture RIGHT
[13:51:31][D][binary_sensor:036]: 'APDS960 Right Movement': Sending state ON
[13:51:31][D][binary_sensor:036]: 'APDS960 Right Movement': Sending state OFF
```

**Additional information and things you've tried:**
Again, the same sensor, with the exact same settings and wiring will work on just one nodemcu board (Amica), on every other board it fails to connect via UART?!

dedepene, 2019-11-04:
After another round of attempts with more devices the takeaway so far is that only the AMICA branded boards are any good with the exact same config for the MHZ19. No dice with the rest :(

borpin, 2019-11-04:
IME, the MHZ19 (is it a 'B'?) are really fussy with serial timing to the extent of you use logging above VERBOSE it doesn't work. Have you tried with just the UART configured and nothing else?

dedepene, 2019-11-04:
> IME, the MHZ19 (is it a 'B'?) are really fussy with serial timing to the extent of you use logging above VERBOSE it doesn't work. Have you tried with just the UART configured and nothing else?

Yes, it's a "B", never tried anything above "DEBUG" with it anyway. With only UART config it will boot up and just chill - no error messages of any sort. How is this helping with troubleshooting?

OttoWinter, 2019-11-04:
You're using pins GPIO1 and GPIO3 - these are also the pins used for logging and the USB FTDI adapter.

Please use different pins or disable logging over UART with logger->baud_rate.

dedepene, 2019-11-05:
yup...without fail it works just fine on [AMICA branded](https://imgur.com/YPInoh6) NOdeMCUs - just flashed a fresh one. Every other one (LoLin, noname etc.) is a no-go. With the Amica it's just fine on GPIO1 and GPIO3, no need to diasble logging over UART...

OttoWinter, 2019-11-05:
If it works with some NodeMCU boards and not with others, this is probably a hardware issue.

The MHZ19 operates at 5V if I remember correctly - and if the pins you're connecting it to on the ESP have pullups that might cause issues (see also https://github.com/esphome/issues/issues/812#issuecomment-549572377).

I'll leave the issue open for if you find a different solution, but otherwise I'm going to say this is not a software bug.

arusa, 2020-03-06:
I've got the same problem with my AMICA NodeMCU.

```
logger:
  baud_rate: 0

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

sensor:
  - platform: mhz19
    id: mhz19sensor
    co2:
      name: "MH-Z19 CO2 Value"
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 60s
    automatic_baseline_calibration: false
```

I've also tried switching GPIO1 and GPIO3 but I always get this error:

```
[19:43:02][C][uart:140]: UART Bus:
[19:43:02][C][uart:142]:   TX Pin: GPIO1
[19:43:02][C][uart:145]:   RX Pin: GPIO3
[19:43:02][C][uart:147]:   Baud Rate: 9600 baud
[19:43:02][C][uart:148]:   Stop bits: 1
[19:43:02][C][uart:150]:   Using hardware serial interface.
[19:43:02][C][logger:175]: Logger:
[19:43:03][C][logger:176]:   Level: DEBUG
[19:43:03][C][logger:177]:   Log Baud Rate: 0
[19:43:03][C][logger:178]:   Hardware UART: UART0
[19:43:03][C][mhz19:094]: MH-Z19:
[19:43:03][C][mhz19:095]:   CO2 'MH-Z19 CO2 Value'
[19:43:03][C][mhz19:095]:     Unit of Measurement: 'ppm'
[19:43:03][C][mhz19:095]:     Accuracy Decimals: 0
[19:43:03][C][mhz19:095]:     Icon: 'mdi:periodic-table-co2'
[19:43:03][C][mhz19:096]:   Temperature 'MH-Z19 Temperature'
[19:43:03][C][mhz19:096]:     Unit of Measurement: '°C'
[19:43:03][C][mhz19:096]:     Accuracy Decimals: 0
[19:43:03][C][mhz19:096]:     Icon: 'mdi:thermometer'
[19:43:45][E][uart:229]: Reading from UART timed out at byte 0!
[19:43:45][W][mhz19:034]: Reading data from MHZ19 failed!
```

I also tried getting 5v and ground from a different power supply.

I have no idea how to debug such problems and how to rule out hardware problems.

Basanon, 2020-03-15:
I have little bit different issue
Im using ESP32 with MH-Z19B
And weird thing is, ESP32 connects to a WiFi access point within 1 meter from it, if I put it in 5 meters it loses WiFi connection with MH-Z19B attached to it or unable to connect (No MH - no connect issue) 
Changed MH-Z19 sensors - same issue...
Have changed power supplies (2-3A), even used StepUp module to make a little bit higher voltage (5.05v) for MH-Z19

falithurrahman, 2020-04-09:
> If it works with some NodeMCU boards and not with others, this is probably a hardware issue.
> 
> The MHZ19 operates at 5V if I remember correctly - and if the pins you're connecting it to on the ESP have pullups that might cause issues (see also [#812 (comment)](https://github.com/esphome/issues/issues/812#issuecomment-549572377)).
> 
> I'll leave the issue open for if you find a different solution, but otherwise I'm going to say this is not a software bug.

Hi @OttoWinter, I am having identical issue with [this](https://github.com/esphome/issues/issues/812). At that post, you suggest to disable FTDI adapter. Is it okay to avoid FTDI adapter by hardwire the UART from my device to GPIO1 and GPIO3? I mean hardwire is connecting to ESP12E pinout, not to the Wemos D1 pinhead. Thanks

borpin, 2020-04-09:
I had loads of issues with this sensor and ESPHome. Ended up using Tasmota instead and it is rock solid. This was a while ago so things may be different now.

IronButterfly, 2020-05-13:
I have NodeMCU V3 and MH-Z19B. I have tried a lot of different GPIO pins, configurations, etc, but unfortunately, see:

22:52:09][E][uart:230]: Reading from UART timed out at byte 0!
[22:52:09][W][mhz19:034]: Reading data from MHZ19 failed!

Alafalx, 2020-06-06:
I have the same issue, but previously I tested the same hardware with ESPEasy firmware and everything was working fine, so I assume this is not a hw issue (@OttoWinter )

nkitanov, 2020-07-03:
I have such configuration on esp8266 and it works but you have to disable uart logger as it's bound to the same pin.

jcalbert, 2020-11-25:
I also had this problem after leaving a working esphome device unpowered for a long time.  Ultimately the fix was to recalibrate the MHZ-19 by tying the HD pin to ground for 7 seconds.  After restarting the device it worked again.

P.S. Doing this resets the calibration, so the device should be left in fresh air for 20 minutes afterwards.

hpapagaj, 2021-01-13:
GPIO16 && GPIO17 worked for me on a board like this: 

[image]

closed 2021-05-28 by stale[bot]