esphome/issues #816: Unable to reach ESP32 modules with v1.14.2

DasBootU96, 2019-11-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 0.101.2 with ESPHome addon 1.14.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 - esp32doit-devkit-v1
**Affected component:**

mdns - https://esphome.io/guides/faq.html?highlight=mdns

**Description of problem:**
After uploading 1.14.2 to ESP32 (DoIt DevKit v1.0) modules, ESPHome Dashboard shows module as offline, I can no longer reach its webpage or upload new builds and there is no mdns entry for it in [Chrome’s mdns browser app](https://chrome.google.com/webstore/detail/mdns-browser/kipighjpklofchgbdgclfaoccdlghidp) (running on same subnet) and it shows offline in ESPHome’s config page. Current workaround is using fixed IPs.

After uploading 1.14.2 to ESP8266 (Wemos D1 mini or ESP01) modules have no problem and I can see an mdns entry.

Existing ESP32 modules still running 1.13.6 have no problem and I can see an mdns entry.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32test
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "WIFIAP"
  password: "password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Fallback Hotspot"
    password: "wGmReCrulAVP"

captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:

time:
  - platform: homeassistant
    id: homeassistant_time
    
web_server:
  port: 80

status_led:
  pin:
    number: GPIO2

sensor:
  - platform: uptime
    name: Uptime Sensor

  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

binary_sensor:
  - platform: status
    name: "Status"

text_sensor:
  - platform: wifi_info
[... 236 lines cut ...]
[13:41:42][V][sensor:013]: 'WiFi Signal Sensor': Received new state -48.000000
[13:41:42][D][sensor:092]: 'WiFi Signal Sensor': Sending state -48.00000 dB with 0 decimals of accuracy
[13:41:50][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=300132 (now=310132)
[13:42:00][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=310132 (now=320132)
[13:42:10][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=320132 (now=330132)
[13:42:12][VV][scheduler:131]: Running interval 'update' with interval=60000 last_execution=271295 (now=331297)
[13:42:12][V][sensor:013]: 'Uptime Sensor': Received new state 331.299988
[13:42:12][D][sensor:092]: 'Uptime Sensor': Sending state 331.29999 s with 0 decimals of accuracy
[13:42:20][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=330132 (now=340139)
[13:42:30][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=340132 (now=350137)
[13:42:40][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=350132 (now=360136)
[13:42:42][VV][scheduler:131]: Running interval 'update' with interval=60000 last_execution=301959 (now=361960)
[13:42:42][V][sensor:013]: 'WiFi Signal Sensor': Received new state -48.000000
[13:42:42][D][sensor:092]: 'WiFi Signal Sensor': Sending state -48.00000 dB with 0 decimals of accuracy
[13:42:50][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=360132 (now=370132)
[13:43:00][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=370132 (now=380132)
[13:43:10][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=380132 (now=390132)
[13:43:12][VV][scheduler:131]: Running interval 'update' with interval=60000 last_execution=331295 (now=391298)
[13:43:12][V][sensor:013]: 'Uptime Sensor': Received new state 391.300995
[13:43:12][D][sensor:092]: 'Uptime Sensor': Sending state 391.30099 s with 0 decimals of accuracy
```

**Additional information and things you've tried:**

Using fixed IPs.

OttoWinter, 2019-11-04:
Thanks for providing detailed logs and the note about mDNS.

It sounds reasonable that mDNS might be the problem here. If it doesn't work, nothing will connect.

I remember two code changes that could cause this:

 1. arduino framework version was updated. Please try the same with `arduino_version: 1.0.2` in your core-config to use the esp-idf version from 1.13.6

 2. The mDNS initialization timing changed - If I remember correctly it's now initialized after the WiFi interface is started. Maybe that has something to do with it, though I'm using ESP32s with mDNS everywhere and have not had this problem.

DasBootU96, 2019-11-04:
OK, it is just weird and some others have reported similar issues [here ](https://community.home-assistant.io/t/esp32-mdns-issue-after-upgrading-esphome-to-1-14-0/145896) and [here](https://community.home-assistant.io/t/cosmetic-problems-after-the-latest-ppdate-in-esphome/145930).

About this log entry:

`[13:36:44][V][wifi_esp32:341]: Event: Got IP static_ip=10.0.30.112 gateway=10.0.30.1`

It mentions **static_ip**, but DHCP is used and is therefore confusing to me. Is it a red herring?

I should add that if I refresh Chrome’s mdns browser app often enough, I do see the v1.14.2 ESP32 mdns entries albeit very briefly whereas the v1.14.2 ESP8266 mdns entries and v1.13.6 ESP32 mdns entries are always visible.

OttoWinter, 2019-11-05:
> It mentions static_ip, but DHCP is used and is therefore confusing to me. Is it a red herring?

No. After DHCP negotiation it's essentially the same as using a static IP - the log line shows that it got a valid IP from the DHCP server.

> I should add that if I refresh Chrome’s mdns browser app often enough, I do see the v1.14.2 ESP32 mdns entries albeit very briefly

That's odd. The mDNS records have a TTL of around 2min, so if the extension should leave them there for at least 2min no matter what happens

DasBootU96, 2019-11-05:
Added arduino_version:
```yaml
esphome:
  name: esp32test
  platform: ESP32
  board: esp32doit-devkit-v1
  arduino_version: 1.0.2
```

but no change in resolving the hostname and no mdns entry in the mdns browser app. Also tried another brand [keyesstudio ESP32](https://wiki.keyestudio.com/KS0413_keyestudio_ESP32_Core_Board), but same behaviour as the DoIt board.

Are there other options to try?

grzeg8102, 2019-11-06:
I'm experiencing similar issues which also started with 1.14.0, no improvement with 1.14.2. I posted this as issue #824 . 
When I open the debug window from esphome dashboard (docker) - esphome is not able to connect immediately at the first attempt. It used to work like this and was rock-solid with 1.13.6. Now, it needed 4 retries. Finally connected, but struggled. 
Also when I launch esphome dashboard initially I see my ESP nodes with a red status for 15-20 seconds and then it changes to green. It used to work a lot faster in the previous software.

DasBootU96, 2019-11-06:
Not sure if this will help, but I use a [Mikrotik 962UiGS-5HacT2HnT](https://mikrotik.com/product/RB962UiGS-5HacT2HnT) capable of packet sniffing. This is a Wireshark capture from two ESP32 nodes [ESPHome Issue 816.zip](https://github.com/esphome/issues/files/3813650/ESPHome.Issue.816.zip).

```
Node1 - 10.0.30.110 - 3C:71:BF:F1:EA:14 - ESPHome 1.13.6
Esp32test - 10.0.30.112 - 24:6F:28:29:B9:30 - ESPHome 1.14.2

hassio - 10.0.30.10
```

Mdns for both appear the same, but have never dealt with mdns before.

OttoWinter, 2019-11-06:
Thanks for supplying detailed packet logs @DasBootU96! I had a look at them with wireshark and could not find anything.

 - both .110 and .112 are initially sending 3  mDNS packets announcing their presence (all of which arrive intact).
 - They both periodically retransmit their original mDNS packet as cache flush packets - all fine there too.

What's missing is any device sending mDNS requests to the device - if HA or some other ESPHome client is on the network and trying to connect, they would be sending mDNS queries to which the ESP would respond. But that might be due to filtering the PCAP file.

I tried an ESP32 with my own setup here and can't reproduce the issue.

You also tried the `arduino_version: 1.0.2` already and that didn't give any results - so there are two things left that could explain it:

 - A change in WiFi connection logic [here](https://github.com/esphome/esphome/blame/dev/esphome/components/wifi/wifi_component_esp32.cpp#L170) - highly unlikely that it would manifest itself in this way IMO.
 - `captive_portal` being active - Maybe the AP messes up the mDNS source address in some way. I tried reproducing that too, but found no problem. Could you try without captive_portal and wifi AP enabled?

Spirituss, 2019-11-06:
I have simular issue. Possibly, it’s concerned with ble scanner sensor, which is also activated. Now I can’t even update esp32 via OTA.

OttoWinter, 2019-11-06:
@Spirituss That's a different issue -> https://github.com/esphome/issues/issues/824

DasBootU96, 2019-11-07:
Tried without captive_portal and wifi AP enabled like so:

```yaml
wifi:
  ssid: "WIFIAP"
  password: "password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Fallback Hotspot"
    password: "wGmReCrulAVP"

# captive_portal:
```

No difference. Also, a few existing ESP32 nodes were upgraded from 1.13.6 to 1.14.2 and do not have fallback hotspot since as this is a new feature, but do have the same problem with mdns resolving.

Previous Wireshark captures were filtered to two ESP32 MAC addresses only, I've attached new captures. Captured starts, then in ESPHome Dashboard, clicked 'show logs' for these two nodes:
```
Node1 - 10.0.30.110 - 3C:71:BF:F1:EA:14 - ESPHome 1.13.6
Esp32test - 10.0.30.112 - 24:6F:28:29:B9:30 - ESPHome 1.14.2
```
Once ESPHome logs show, Wireshark capture was stopped. For node Esp32test it took many retries and eventually showed logs after more than 2 minutes. For Node1 it took no retries and log showed after seconds.

Here are Wireshark logs:
[Capture node1 + esp32test - ESPHome Issue 816.zip](https://github.com/esphome/issues/files/3820805/Capture.node1.%2B.esp32test.-.ESPHome.Issue.816.zip).

Packet filtered on hassio (Pi4) and all ESPHome node MAC addreses only. You may notice other, excluded traffic. Of note are:
10.0.30.99 - W10VM with Wireshark + Chrome webpage opened Hassio front-end
10.0.30.1 - Pfsense router
10.0.30.254 - Mikrotik Wifi AP

Hope this helps.

mikkaat, 2019-11-08:
please excuse my lack of technical skill....

I have been playing around with some esp32s devices i have here and have noticed that when i create a new esp yaml (1.14.2) and load it onto a n esp32s using the board: nodemcu-32s value, i can access the board perfectly and it will not disconnect. I have also found that after  configuring as follows, the board still works perfectly.. 

`# MiFlora and Irrigation 2
esphome:
  name: esp32_mi_flora
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 0s
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32 Mi Flora Fallback Hotspot"
    password: "rhhChAowFRwO"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

#esp32_ble_tracker:
#  scan_interval: 10800s`

HOWEVER, when i uncomment the `esp32_ble_tracker:` line, the board will dail with the mDNS issue. I then have to reflash the module with the bin file for it to work again.

Hope this makes sense.

Gooman-rus, 2019-11-08:
I have the same issue: after updating to 1.14.2 from 1.13.6 I have issue with devices: can't connect to them in order to read logs or for the OTA updates. This forces me to downgrading to 1.13.6.

OttoWinter, 2019-11-08:
@mikkaat @Gooman-rus Please confirm what you're facing is really an mDNS issue and not a more general disconnect issue. Please post what symptoms you're seeing, especially with @mikkaat I believe it is a case of https://github.com/esphome/issues/issues/824

Gooman-rus, 2019-11-08:
> @mikkaat @Gooman-rus Please confirm what you're facing is really an mDNS issue and not a more general disconnect issue. Please post what symptoms you're seeing, especially with @mikkaat I believe it is a case of #824

I don't use BLE tracker or other Bluetooth related things. My hardware is ESP32 and Waveshare E-ink display. I use this display in order to show data from homassistant. After updating to 1.14.2 I could not connect to my device via "show logs" button. It said that it can't determine IP of my device.
After restart my device stopped to show any data from homeassistant. Also it appeared as Offline in ESPHome page.
Also after updating to 1.14.2 ESPHome stopped to connect (via "show logs" button) to smart socket Blitzwolf with firmware v. 1.13.6. But homeassistant stil had an ability to receive data from the socket.

Now I used a homeassistant snapshot in order to downgrade ESPHome to 1.13.6. Also it forced me to manually install 1.13.6 firmware to my ESP32. And not it works very well.

DasBootU96, 2019-11-10:
**Update:** 
Used an old Netgear WNDR3700v1 and installed DD-WRT in gateway mode. Moved the Hassio (Pi4) wired connection to it, changed WiFi SSID + password to the same as I was using, then disabled to original WiFi SSID of the same name.

Now mdns entries for ESPHome ESP32 nodes v1.14.2 are all visible and **remain visible** in Chrome's mdns browser. In the ESPHome Dashboard, show logs using the .local name works fine and I can update new builds to ESP32 nodes v1.14.2 using the .local name without issues or delays. 

However, opening any ESPHome node (v1.13.6 or v1.14.2 ESP8266/ESP32) webpage does not work via mdns <nodename>.local (tried Chrome and Opera while on the same subnet as the nodes and Hassio), open webpage via its IP is fine. Also a few ESP32 nodes v1.14.2 have become unavailable which is easily resolved by deleting and adding again.

Had the same webpage .local issue with the mDNS_Web_Server.ino Arduino example sketch on any of Espressif's ESP32 Arduino cores v1.0.1 through v1.0.4 regardless if I used ```MDNS.addService("http", "tcp", 80);``` or ```MDNS.addService("_http", "_tcp", 80);```. 

FWIW: the original setup used Mikrotik 962UiGS-5HacT2HnT connected to a VLAN trunk and the Wifi connections for IOT Wifi and regular Wifi traffic were separated on different VLANs with their own SSIDs. PFsense takes care of routing. Wired Hassio (Pi4) was connected to the same VLAN as the IOT Wifi network. All worked fine until the ESPHome  v1.14.2 mdns resolving issues.

It seems ESPHome ESP32 nodes v1.14.2 and VLANs don't play well toghether, but unsure what to exactly make of all this yet... Hope it helps.

mikkaat, 2019-11-12:
> @mikkaat @Gooman-rus Please confirm what you're facing is really an mDNS issue and not a more general disconnect issue. Please post what symptoms you're seeing, especially with @mikkaat I believe it is a case of #824

I apologise, if you can supply guidance on how to post the symptoms, i will, otherwise, as an update, i have reverted back to 1.13.6 and the device is not experiencing any dropping issues at all.

OttoWinter, 2019-11-12:
@DasBootU96 Thanks for the update. Yeah it might be due to the router. First assumption would of course be that an update changed the mDNS format and the router does not accept that format. But the packet captures you supplied looked fine - all valid mDNS packets. Also there were no major changes to [esp-idf mDNS](https://github.com/espressif/esp-idf/tree/master/components/mdns) lately, so I don't think it's related to that.

If it's the router and VLAN I would just say it's an issue with some network component in the chain not being happy with mDNS format. Since mDNS is sadly quite flaky anyway (from the protocol itself), I don't think there's much that can be done.

OttoWinter, 2019-11-12:
I'm going to close this issue because I think it's due to some other/upstream issue. If there's more clear evidence showing that ESPHome is involved please re-open or create a new issue linking here.

closed 2019-11-12 by OttoWinter

fribse, 2019-11-12:
It is disturbing that this issue is closed, I don't see it solved at all.
I have quite a few ESPHome devices running SONOFF Basic, POW2, RF 433 Bridge, and quite a few of self-built ESP8266 with relays, and a lot of other stuff.
I moved to 1.14 and immediately got problems, the devices was shown as disconnected in HA, so I started giving them fixed IP's as the theory for me was that mDNS was failing. That made it a bit better, still flakey in HA, showing as offline, but going to view logs presented the logs.
The devices seems to reboot sporadically, or at least loosing connection sporadically.
I upgraded to 1.14.2 as soon as it was out, and saw some improvement, but no way as stable as 1.13.6, which led me to the conclusion that I needed to roll back to 1.13.6. I did see an improvement in mDNS, but not everything was present.
I seem to recall that there is a problem with AsyncTCP and certain kernels in ESP8266?
Fortunately I have backups so I didn't loose too much.
Going back to 1.13.6 made it ROCKSOLID again.

Infrastructure: 80 sqm house in two floors, AP on each floor (Fritz!Box and Fritz!Repeater), running both 2,4 GHz and 5 GHz. HA is running on a RPi4 4 GB. Switches are Cisco SG200-24, and Cisco Sg200-8.

OttoWinter, 2019-11-12:
> SONOFF Basic, POW2, RF 433 Bridge, and quite a few of self-built ESP8266

all those are ESP8266s. This issue refers to ESP32s. That's also part of the issue with resolving this - people post their symptoms all across several threads, and it gets really hard to understand what's related to what issue etc.

You're probably looking for https://github.com/esphome/issues/issues/838

brandond, 2019-11-12:
>the original setup used Mikrotik 962UiGS-5HacT2HnT connected to a VLAN trunk and the Wifi connections for IOT Wifi and regular Wifi traffic were separated via a tag configured in Mikrotiks config on different VLANs with their own SSIDs. PFsense takes care of routing.

The M in MDNS stands for multicast. It doesn't route across subnets, so if the devices are on a different vlan and subnet than your server it wont work unless you run a MDNS repeater with an interface on each network.

That doesn't explain the drops, so there may be other issues, but I would definitely try to troubleshoot MDNS separately from the connectivity issues.

DasBootU96, 2019-11-12:
Thanks, I'm aware mdns does not route across subnets. 

To clarify: Hassio, all ESP nodes and the system used to run Chromes mdns browser app are all on the **same** IOT VLAN subnet. 

Yet I experience missing mdns entries for ESP32 v1.14.x, while ESP8266 v1.14.x and ESP32 v1.13.6 node mdns entries are always visible.

mikkaat, 2019-11-12:
> I'm going to close this issue because I think it's due to some other/upstream issue. If there's more clear evidence showing that ESPHome is involved please re-open or create a new issue linking here.

Can you advise which issue these should be posted against? From my (completely NON technical) perspective, I was originally running devices on 1.13.6, all working perfectly, I upgraded to 1.14.2, failing and dropping continuously, downgraded to 1.13.6, all working perfectly again. The only thing that has changed is the esphome version. If this is not esphome, what is it?

OttoWinter, 2019-11-12:
@mikkaat From your initial report, it's related to esp32_ble_tracker. See https://github.com/esphome/issues/issues/824 for that.

@DasBootU96 You see, from my perspective what I see is mDNS fails suddenly with 1.14.x, but there was no change to mDNS in that version - the only change was the SDK version, which you've ruled out by setting `arduino_version` to the old default.

And from the packet captures (which is the best truth we're going to get on this issue), there are also no visible issues - if there were, we would see them there.

Next, some of the other things mentioned here have also been ruled out. You don't have BLE enabled (as in @mikkaat's issue), you don't have an ESP8266 (which is using totally different code) as with @fribse.

Adding to that, I can't replicate this issue. I've tried a bunch of setups with ESP32s and mDNS names with various names, but all work just fine.

The only conclusion I can come to is that mDNS is not working well together with your router.

DasBootU96, 2019-11-12:
So I just changed Mikrotiks wireless interface setting ```Multicast Helper: full``` and now all mdns entries (yes, ESP32 with v1.14.2 too) are visible, 'show logs' work without any delay, I can upload new builds without problem and all ESP nodes show online in the dashboard. 

Here is what the [Mikrotik manual says](https://wiki.mikrotik.com/wiki/Manual:Interface/Wireless) about **```Multicast Helper```**:
```
multicast-helper (default \| disabled \| full; Default: default)

When set to full multicast packets will be sent with unicast destination MAC address, resolving multicast problem on wireless link. 
This option should be enabled only on access point, clients should be configured in station-bridge mode. Available starting from v5.15.

* disabled - disables the helper and sends multicast packets with multicast destination MAC addresses
* full - all multicast packet mac address are changed to unicast mac addresses prior sending them out
* default - default choice that currently is set to disabled. Value can be changed in future releases.
```

Previously its was left as default which was: ```default```...and according to the manual that means it was disabled. For completeness sake, I was running routerOS 6.44.3 when this issue showed up, this evening upgraded to 6.45.7, then finally started messing with routerOS Multicast settings.

Thanks @OttoWinter and all others, yet I still do not know exactly why this change is required for v1.14.x while v1.13.6 was working fine without this change. Then again I'm not a hardcore networking guy.

Hope this helps others with similar issues.

mikkaat, 2019-11-12:
@OttoWinter **thank you**, I have been running those settings for about 3 hours now with no drops.