esphome/issues #817: pzem004t component not working

rradar, 2019-11-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- pip3, 1.14.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

- nodeMCU (esp8266)

**Affected component:**

- pzem004t

**Description of problem:**

The pzem004t integration doesn't work. I'm using the pzem004 v2 with software serial connected to a nodeMCU. Exact hardware combination works with espurna so a hardware fault can be foreclosed.

Reported already by @renat85 in #813 but he failed to fill the template.

**Problem-relevant YAML-configuration entries:**

```yaml
uart:
  tx_pin: D1
  rx_pin: D2
  baud_rate: 9600
  stop_bits: 2

sensor:
  - platform: pzem004t
    current:
      name: "${upper_devicename} Current"
    voltage:
      name: "${upper_devicename} Voltage"
    power:
      name: "${upper_devicename} Power"
      id: pzem_power
    update_interval: 10s
```

**Logs (if applicable):**

Up to `VERBOSE` logging their will me only information for this component for the initial setup:

```yaml
[13:14:43][C][uart:145]: UART Bus:
[13:14:43][C][uart:147]:   TX Pin: GPIO5
[13:14:43][C][uart:150]:   RX Pin: GPIO4
[13:14:43][C][uart:152]:   Baud Rate: 9600 baud
[13:14:43][C][uart:153]:   Stop bits: 2
[13:14:43][C][uart:157]:   Using software serial
```

```yaml
[13:14:43][C][pzem004t:102]: PZEM004T:
[13:14:43][C][pzem004t:103]: Voltage 'PZEM Node Voltage'
[13:14:43][C][pzem004t:103]:   Unit of Measurement: 'V'
[13:14:43][C][pzem004t:103]:   Accuracy Decimals: 1
[13:14:43][C][pzem004t:103]:   Icon: 'mdi:flash'
[13:14:43][C][pzem004t:104]: Current 'PZEM Node Current'
[13:14:43][C][pzem004t:104]:   Unit of Measurement: 'A'
[13:14:43][C][pzem004t:104]:   Accuracy Decimals: 2
[13:14:43][C][pzem004t:104]:   Icon: 'mdi:flash'
[13:14:43][C][pzem004t:105]: Power 'PZEM Node Power'
[13:14:43][C][pzem004t:105]:   Unit of Measurement: 'W'
[13:14:43][C][pzem004t:105]:   Accuracy Decimals: 0
[13:14:43][C][pzem004t:105]:   Icon: 'mdi:flash'
```

Only activating a `VERBOSE_LOGGING` gives some repeating output for the uart component (still nothing about pzem004t):

```yaml
[13:14:48][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=3689 (now=13690)
[13:14:52]pm open,type:0 0
[13:14:52][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=7619 (now=17619)
[13:14:52][VV][uart:184]:     Wrote 0b10110100 (0xB4)
[13:14:52][VV][uart:184]:     Wrote 0b11000000 (0xC0)
[13:14:52][VV][uart:184]:     Wrote 0b10101000 (0xA8)
[13:14:52][VV][uart:184]:     Wrote 0b00000001 (0x01)
[13:14:52][VV][uart:184]:     Wrote 0b00000001 (0x01)
[13:14:52][VV][uart:184]:     Wrote 0b00000000 (0x00)
[13:14:52][VV][uart:184]:     Wrote 0b00011110 (0x1E)
[13:14:58][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=13689 (now=23690)
[13:15:01][VV][api.service:250]: on_ping_request: PingRequest {}
[13:15:01][VV][api.service:035]: send_ping_response: PingResponse {}
[13:15:02][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=17619 (now=27623)
[13:15:02][VV][uart:184]:     Wrote 0b10110100 (0xB4)
[13:15:02][VV][uart:184]:     Wrote 0b11000000 (0xC0)
[13:15:02][VV][uart:184]:     Wrote 0b10101000 (0xA8)
[13:15:02][VV][uart:184]:     Wrote 0b00000001 (0x01)
[13:15:02][VV][uart:184]:     Wrote 0b00000001 (0x01)
[13:15:02][VV][uart:184]:     Wrote 0b00000000 (0x00)
[13:15:02][VV][uart:184]:     Wrote 0b00011110 (0x1E)
[13:15:08][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=23689 (now=33689)
[13:15:12][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=27619 (now=37623)
[13:15:12][VV][uart:184]:     Wrote 0b10110100 (0xB4)
[13:15:12][VV][uart:184]:     Wrote 0b11000000 (0xC0)
[13:15:12][VV][uart:184]:     Wrote 0b10101000 (0xA8)
[13:15:12][VV][uart:184]:     Wrote 0b00000001 (0x01)
[13:15:12][VV][uart:184]:     Wrote 0b00000001 (0x01)
[13:15:12][VV][uart:184]:     Wrote 0b00000000 (0x00)
[13:15:12][VV][uart:184]:     Wrote 0b00011110 (0x1E)
[13:15:16][VV][api.service:250]: on_ping_request: PingRequest {}
[13:15:16][VV][api.service:035]: send_ping_response: PingResponse {}
[13:15:18][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=33689 (now=43689)
[13:15:20][VV][scheduler:131]: Running interval 'update' with interval=60000 last_execution=4294952361 (now=45071)
[13:15:20][V][sensor:013]: 'PZEM Node Uptime': Received new state 45.074001
[13:15:20][D][sensor:092]: 'PZEM Node Uptime': Sending state 45.07400 s with 0 decimals of accuracy
[13:15:20][VV][api.service:120]: send_sensor_state_response: SensorStateResponse {
[13:15:20]  key: 3667435239
[13:15:20]  state: 45.074
[13:15:20]}
[13:15:22][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=37619 (now=47621)
[13:15:22][VV][uart:184]:     Wrote 0b10110100 (0xB4)
[13:15:22][VV][uart:184]:     Wrote 0b11000000 (0xC0)
[13:15:22][VV][uart:184]:     Wrote 0b10101000 (0xA8)
[13:15:22][VV][uart:184]:     Wrote 0b00000001 (0x01)
[13:15:22][VV][uart:184]:     Wrote 0b00000001 (0x01)
[13:15:22][VV][uart:184]:     Wrote 0b00000000 (0x00)
[13:15:22][VV][uart:184]:     Wrote 0b00011110 (0x1E)
[13:15:25][VV][scheduler:131]: Running interval 'update' with interval=60000 last_execution=4294957666 (now=50375)
```

**Additional information and things you've tried:**

A pzem004t v3 with the `pzemac` component works perfectly!

renat85, 2019-11-04:
Confirm. V1 version not work too.

brandond, 2019-11-04:
I can't recall if the intent was for the component to support both V1/V2 and V3, but the V3 uses a completely different pseudo-modbus protocol.

rradar, 2019-11-04:
@brandond technically spoken **pzem004t v1 and v2 are the same** for esphome and should be used with the `pzem004t component`. This is the component which is not working and this issue is about.

For **pzem004t v3** the `pzemac component` is used and this works since the release (1.14) just fine for me.

OttoWinter, 2019-11-04:
The command sent looks fine to me based on the datasheet.

Might be that the PZEM004T do *not* respond to set-address requests, not sure.

I was under the impression that the pzem004t integration worked when it got merged, but I might have missed some comments.

sergeolkhovik, 2019-11-07:
Gentlemen, I'm very upset as I was waiting for this release and after several tests I can confirm that my v2.0 device doesn't work with 1.14.2 either.
I tried both software serial (D5, D6, like https://primalcortex.wordpress.com/2019/07/06/measuring-home-energy-consumption-with-the-pzem004t-and-esp8266/) and even disabled logger to utilize hardware uart. My board is Chineese Wemos D1.
Tested PZEM device both with Arduino and SoftSerial (https://www.nn-digital.com/en/blog/2019/11/04/example-of-the-pzem-004t-v2-v2-0-interfacing-program-using-arduino/), and with ESPEasy (https://www.letscontrolit.com/forum/viewtopic.php?p=21383#p21383) - both work really nice.
So the problem is in ESPHome, unfortunately.
Can anyone tell me how to test Otto's fix?

BTW configuration is really simple, something like this:

```
uart:
  id: uartpz
  tx_pin: D6
  rx_pin: D5
  baud_rate: 9600
  stop_bits: 2

sensor:
  - platform: pzem004t
    uart_id: uartpz
    current:
      name: "PZEM-004T Current"
    voltage:
      name: "PZEM-004T Voltage"
    power:
      name: "PZEM-004T Power"
    update_interval: 16s
```

michelebossa, 2019-11-12:
Hi @OttoWinter,
I have tried your commit using the option "esphome_version": "try-fix-pzem004t" and new i can read the value from my PZEM004T V2 but using a stop_bits 1.

My configuration:
```
uart:
  tx_pin: RX
  rx_pin: TX
  baud_rate: 9600
  stop_bits: 1
  
sensor:
  - platform: pzem004t
    current:
      name: ${plug_name}_Amperage
      icon: mdi:current-ac
      unit_of_measurement: A
    voltage:
      name: ${plug_name}_Voltage
      icon: mdi:flash-circle
      unit_of_measurement: V
    power:
      name: ${plug_name}_Wattage
      icon: mdi:flash-outline
      unit_of_measurement: W
      id: ${plug_name}_Wattage
    update_interval: 20s 
```

[image]

I'm looking the possibility to read the total energy stored into PZEM.
 
Read the energy
Send command: B3 C0 A8 01 01 00 1D
Reply data: A3 01 86 9F 00 00 C9
Note: Reply energy data is D1D2D3 = 01 86 9F, converts 01 86 9F to decimal is 99999, so the accumulated power is 99999Wh.

Mybe we can add a sensor and the case to retrieve the value.

```
      case 0xA3: { // Energy Response
        uint16_t energy = (uint16_t(resp[1]) << 16) | (uint16_t(resp[2]) << 8 | (uint16_t(resp[3]) << 0);
        if (this->energy_sensor_ != nullptr)
          this->energy_sensor_->publish_state(energy);
        ESP_LOGD(TAG, "Got Energy %u Wh", energy);
        this->write_state_(DONE);
        break;
      } 

```
I'm not an expert of GIT and i didn't now how made the commit into a your branch or made my own to test the code. Can you help me? 
My repository [https://github.com/cavaliere78/esphome](url)

Thanks.

marckamil, 2019-11-12:
In my PZEM004T V3 I had to also set stop_bit to 1 and it works now :)

OttoWinter, 2019-11-12:
@michelebossa Thanks for trying the fix. For energy field please see https://github.com/esphome/feature-requests/issues/457

renat85, 2019-11-12:
For v1 version not work.

closed 2019-11-12 by OttoWinter

AndreyHeaven, 2019-11-18:
nothing changed :-(
still not working

sergeolkhovik, 2019-11-18:
Hi, just finished tests with **v2** and can say that now all works for me (even with soft-serial). The only note - stop_bits should be 1, not 2.
Here's my full config if this is interesting:

```
esphome:
  name: d9
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain

# Enable logging
logger:
  #level: VERY_VERBOSE

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

uart:
  id: uartpz
  tx_pin: D6
  rx_pin: D5
  baud_rate: 9600
  #stop_bits: 2
  stop_bits: 1

sensor:
  # Sensors with general information.
  # Uptime sensor.
  - platform: uptime
    name: D9 Uptime
    # WiFi Signal sensor.
  - platform: wifi_signal
    name: D9 WiFi Signal
    update_interval: 60s
  - platform: pzem004t
    uart_id: uartpz
    current:
      name: "PZEM-004T Current"
    voltage:
      name: "PZEM-004T Voltage"
    power:
      name: "PZEM-004T Power"
    update_interval: 16s

binary_sensor:
  - platform: status
    name: D9 Status

text_sensor:
  # Text sensors with general information.
  # Expose ESPHome version as sensor.
  - platform: version
    name: D9 ESPHome Version
    # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: D9 IP address

switch:
  - platform: restart
    name: D9 Restart

status_led:
  pin:
    number: D4
    inverted: yes
```

Some screenshots:

[image]
[image]
[image]

Connected power-saving lamp.