esphome/issues #819: Temperature from sensor htu21d, bme280 is not displayed in Home Assistant

empenoso, 2019-11-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
WeMos D1 mini

**Affected component:**
https://esphome.io/components/sensor/htu21d.html
https://esphome.io/components/sensor/bme280.html

**Description of problem:**
I have two WeMos D1 mini boards. HTU21D Temperature+Humidity Sensor is connected to one board. BME280 Temperature+Pressure+Humidity Sensor is connected to another board.
In the logs I see the temperature value, but it is not in the web interface. There is no temperature value in Home Assistant.

**Problem-relevant YAML-configuration entries:**
htu21d:
```substitutions:
  devicename: printer_3d_room
  upper_devicename: Printer 3D

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "xxx"
  password: "xxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "3D Printer Room Fallback Hotspot"
    password: "xxxx"

captive_portal:

web_server:
  port: 80
  css_url: http://192.168.15.10:8123/local/webserver-v1.min.css 
  js_url: http://192.168.15.10:8123/local/webserver-v1.min.js

# Enable Home Assistant API
api:
 
# Enable OTA Access
ota:
 
# Enable verbose logging over serial
logger:

i2c:
  sda: D2
  scl: D1
  scan: True

sensor:
  - platform: htu21d
    temperature:
      name: "Температура в серверной"
    humidity:
      name: "Влажность в серверной"
    update_interval: 60s

  - platform: uptime
    id: uptime_sec
  - platform: wifi_signal
    name: ${upper_devicename} WiFi Signal
    update_interval: 900s

text_sensor:
[... 272 lines cut ...]
[21:43:15][D][text_sensor:015]: 'Temperature_329 Uptime': Sending state '0d 0h 11m 46s'
[21:43:42][D][bme280.sensor:197]: Got temperature=23.5°C pressure=985.6hPa humidity=19.8%
[21:43:43][D][sensor:092]: 'uptime_sec': Sending state 766.48401 s with 0 decimals of accuracy
[21:44:42][D][bme280.sensor:197]: Got temperature=23.5°C pressure=985.7hPa humidity=19.8%
[21:44:43][D][sensor:092]: 'uptime_sec': Sending state 826.48297 s with 0 decimals of accuracy
[21:44:50][D][text_sensor:015]: 'Temperature_329 Wifi Strength': Sending state 'Excellent'
[21:45:08][D][text_sensor:015]: 'Temperature_329 Uptime': Sending state '0d 0h 13m 46s'
[21:45:42][D][bme280.sensor:197]: Got temperature=23.5°C pressure=985.7hPa humidity=19.8%
[21:45:43][D][sensor:092]: 'uptime_sec': Sending state 886.48199 s with 0 decimals of accuracy
[21:46:42][D][bme280.sensor:197]: Got temperature=23.5°C pressure=985.6hPa humidity=19.8%
[21:46:42][D][sensor:092]: 'Температура 329': Sending state 23.47400 °C with 2 decimals of accuracy
[21:46:42][D][sensor:092]: 'Атм. давление 329': Sending state 985.64551 hPa with 1 decimals of accuracy
[21:46:42][D][sensor:092]: 'Влажность 329': Sending state 19.75723 % with 1 decimals of accuracy
[21:46:43][D][sensor:092]: 'uptime_sec': Sending state 946.48102 s with 0 decimals of accuracy
[21:47:01][D][text_sensor:015]: 'Temperature_329 Uptime': Sending state '0d 0h 15m 46s'
[21:47:42][D][bme280.sensor:197]: Got temperature=23.5°C pressure=985.6hPa humidity=19.8%
[21:47:43][D][sensor:092]: 'uptime_sec': Sending state 1006.48798 s with 0 decimals of accuracy
[21:48:42][D][bme280.sensor:197]: Got temperature=23.5°C pressure=985.7hPa humidity=19.8%
[21:48:43][D][sensor:092]: 'uptime_sec': Sending state 1066.48096 s with 0 decimals of accuracy
[21:48:54][D][text_sensor:015]: 'Temperature_329 Uptime': Sending state '0d 0h 17m 46s'
```

**Additional information and things you've tried:**
[image]
[image]

OttoWinter, 2019-11-04:
I'm having a hard time understanding what you mean.

What temperature value exactly is missing, the bme280 or htu21d one? What logs and what screenshots am I looking at (which of the two configs)? Could you try to minimize your config to the least amount of items that still exhibits this behavior?

empenoso, 2019-11-05:
I found a pattern of error.
If I use the Russian name, then the temperature disappears from the interface (web interface and Home Assistant). If English - then there is no problem. yaml is below.

[image]
[image]

printer_3d_room.yaml:
```
substitutions:
  devicename: printer_3d_room
  upper_devicename: Printer 3D

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "хххх"
  password: "ххххх"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "3D Printer Room Fallback Hotspot"
    password: "ххххх"

captive_portal:

web_server:
  port: 80
  css_url: http://192.168.15.10:8123/local/webserver-v1.min.css #не отображаются состояния из-за отсутсвия сети или доступа к сайту разработчика https://github.com/esphome/issues/issues/648
  js_url: http://192.168.15.10:8123/local/webserver-v1.min.js

# Enable Home Assistant API
api:
 
# Enable OTA Access
ota:
 
# Enable verbose logging over serial
logger:

i2c:
  sda: D2
  scl: D1
  scan: True

sensor:
  - platform: htu21d
    temperature:
      name: "Температура в серверной" #temperature
    humidity:
      name: "Влажность в серверной" #humidity 
    update_interval: 60s
```

empenoso, 2019-11-05:
Temperature is read in the logs, but not in the interface:

```
[D][htu21d:052]: Got Temperature=24.3°C Humidity=21.6%
[D][sensor:092]: 'Температура в серверной': Sending state 24.32175 °C with 1 decimals of accuracy
[D][sensor:092]: 'Влажность в серверной': Sending state 21.59552 % with 1 decimals of accuracy
[D][htu21d:052]: Got Temperature=24.3°C Humidity=21.7%
[D][sensor:092]: 'Температура в серверной': Sending state 24.28957 °C with 1 decimals of accuracy
[D][sensor:092]: 'Влажность в серверной': Sending state 21.69470 % with 1 decimals of accuracy
[D][htu21d:052]: Got Temperature=24.3°C Humidity=22.0%
[D][sensor:092]: 'Температура в серверной': Sending state 24.30030 °C with 1 decimals of accuracy
[D][sensor:092]: 'Влажность в серверной': Sending state 21.98462 % with 1 decimals of accuracy
```

OttoWinter, 2019-11-05:
Ah, I see what's going on.

The way the unique id of each sensor is generated is by taking it's name, converting it to lowercase and stripping all non-ASCII letters (because Home Assistant doesn't like those/didn't like those in the past).

I don't see a way to fix this, for two reasons:
 - Fixing it would require storing a full NFKD table in the firmware, which can get huge and uses a lot of flash.
 - Fixing it would also change the unique IDs for these entities, invalidating the ESPHome entities for a lot of existing users.

I think Home Assistant had this same issue in the past too - and the suggested workaround was to add a short ASCII prefix to the name. It's not pretty but probably the best possible solution.

closed 2020-08-07 by stale[bot]