esphome/issues #820: Automation stopped working since 1.14.x

HA-TB303, 2019-11-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 0.101.2  esphome 1.14.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini

**Affected component:**

Rotary encoder

**Description of problem:**

Automation not working anymore since 1.14.x

This automation dimmed the lights just fine before 1.14, but since 1.14 rotating left or rights results in the light being switched off.

I am not a coder, so be gentle please ;)

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
### Rotary encoder 1
  - platform: rotary_encoder
    name: "Woonkamer 1 Rotary 1"
    id: woonkamer1rotary1
    pin_a: GPIO12
    pin_b: GPIO13
    filters:
      - or:
        - debounce: 0.2s
        - delta: 5
    resolution: 1
    min_value: 1
    max_value: 10
    on_value:
      then:
        - homeassistant.service:
            service: light.turn_on
            data_template:
              entity_id: light.woonkamer_groep
              brightness_pct: "{{ brightness_1 | int }}"
            variables:
              brightness_1: 'return id(woonkamer1rotary1).state * 10;'
```

**Logs (if applicable):**

```
[20:37:21][V][sensor:013]: 'Woonkamer 1 Rotary 1': Received new state 2.000000
[20:37:21][VV][sensor.filter:013]: Filter(0x3fff1c2c)::input(2.000000)
[20:37:21][VV][sensor.filter:013]: Filter(0x3fff16b4)::input(2.000000)
[20:37:21][VV][scheduler:025]: set_timeout(name='debounce', timeout=200)
[20:37:21][VV][sensor.filter:013]: Filter(0x3fff19d4)::input(2.000000)
[20:37:21][VV][scheduler:131]: Running timeout 'debounce' with interval=200 last_execution=78767 (now=78973)
[20:37:21][VV][sensor.filter:023]: Filter(0x3fff16b4)::output(2.000000) -> 0x3fff1c44
[20:37:21][VV][sensor.filter:013]: Filter(0x3fff1c44)::input(2.000000)
[20:37:21][VV][sensor.filter:020]: Filter(0x3fff1c2c)::output(2.000000) -> SENSOR
[20:37:21][D][sensor:092]: 'Woonkamer 1 Rotary 1': Sending state 2.00000 steps with 0 decimals of accuracy
[20:37:21][VV][api.service:160]: send_homeassistant_service_response: HomeassistantServiceResponse {
  service: 'light.turn_on'
  data_template: HomeassistantServiceMap {
  key: 'entity_id'
  value: 'light.woonkamer_groep'
}
  data_template: HomeassistantServiceMap {
  key: 'brightness_pct'
  value: '{{ brightness_1 | int }}'
}
  variables: HomeassistantServiceMap {
  key: 'brightness_1'
  value: 'return id(woonkamer1rotary1).state * 10;'
}
  is_event: NO
}
```

OttoWinter, 2019-11-04:
Not sure if something has changed semantically, but there's definitely an issue with your YAML: You need to declare the variable as a `!lambda`. Otherwise, the string will be sent as a literal string (as seen in logs `return ...`)

HA-TB303, 2019-11-05:
Strange because this worked before 1.14 100%.

Now i have to figure out how to do this with lamda.

OttoWinter, 2019-11-05:
Yeah, I see now that it has changed semantics.

As for how to do it with lambdas: You just need to add `!lambda` after the `brightness_1:` like this `brightness_1: !lambda`

HA-TB303, 2019-11-05:
Thanks Otto, that fixed it!

closed 2019-11-05 by OttoWinter