esphome/issues #821: Cannot build ESP32 based image - stops at linking - Crashes HA

borpin, 2019-11-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, Docker, VM, HA 0.101.2

ESPHome add-in V1.14.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

WEMOS D32

**Affected component:**

Any

**Description of problem:**
ESP32 based item never complies. Sits at linker and eventually crashes HA.

ESP8266 seems fine.

Worked fine with 1.14.0-dev

**Problem-relevant YAML-configuration entries:**
```yaml
<<: !include common/esphome.yaml
<<: !include common/wifi.yaml
<<: !include common/mqtt.yaml

esp32_ble_tracker:

sensor:
  - platform: xiaomi_lywsdcgq
    mac_address: 4c:65:a8:d8:de:98
    temperature:
      name: "Xiaomi Temperature 98"
      state_topic: emon/xiaomi4c:65:a8:d8:de:98/temperature
      filters:
        - throttle: 90s
        - offset: -0.65
        - sliding_window_moving_average:
            window_size: 11
            send_every: 1
    humidity:
      name: "Xiaomi Humidity 98"
      state_topic: emon/xiaomi4c:65:a8:d8:de:98/humidity
      filters:
        - sliding_window_moving_average:
            window_size: 5
            send_every: 1
        - throttle: 90s
    battery_level:
      name: "Xiaomi Battery 98"
      state_topic: emon/xiaomi4c:65:a8:d8:de:98/battery
```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/esp32_ble.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp32_ble
Processing esp32_ble (board: lolin_d32; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncMqttClient-esphome> 0.8.3
|   |-- <AsyncTCP> 1.1.1
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/esp32_ble/.pioenvs/esp32_ble/src/main.cpp.o
Linking /data/esp32_ble/.pioenvs/esp32_ble/firmware.elf

```

**Additional information and things you've tried:**

Taken sensor out, rebooted server, clean build.

OttoWinter, 2019-11-04:
1. Define "crashes HA" please. Does it crash the whole system or just the addon? If it's the whole system, you're probably out of memory. Please check memory usage while compiling (with something like htop)

2. Might be due to newer esp-idf core that has gotten a bit bigger. So please also try manually using the old core with `arduino_version: 1.0.2` in the esphome: section

borpin, 2019-11-05:
It seems to be the linker rather than the compiler

> 1. Define "crashes HA" please.

I get a `connection lost` message and cannot access any part of the HA system.

> 1. Please check memory usage while compiling

It get pretty high but does not seem to max out.

> 2\. So please also try manually using the old core

Memory use is *significantly* less when linking using the old core.

I have also noticed that the linker may well finish even though HA has restarted (process is still running).  It would be good if there was always a link to the 'latest' bin in the dashboard for each item. I looked for it, but could not work out where it was!

OttoWinter, 2019-11-05:
> It seems to be the linker rather than the compiler

True, though it's both part of the toolchain.

> It get pretty high but does not seem to max out.

The fact that the whole system goes down though points to something is exhausting resources - with memory being the most probable culprit. Maybe also docker plays in here limiting the amount of memory available (depends on host system)

> Memory use is significantly less when linking using the old core.

Ok, good to know. But that shows that it's just new code getting added increasing the SDK size, and thus increasing the amount of memory the linker needs to allocate.

>  It would be good if there was always a link to the 'latest' bin in the dashboard for each item.

There's only so many links that can be put on there without hurting usability. And I think the correct workflow is using the compile button and then pressing download. Otherwise you don't know if it's really the latest config or some previous one.

As there's nothing that can really be done about this, I think we should close this issue.

borpin, 2019-11-05:
Probably so. The memory use is about 40% greater!

The link would be useful as it allows the bin to be backed up without a second build.  An extra link to 'last bin' would fit here 😀 

[image]

OttoWinter, 2019-11-06:
I really don't think this justifies adding a whole new button to the interface.

Plus, as mentioned, the behavior of the button would be hard to define. If you just edited the config and press the button, you will get the binary from the previous config - and that is very bad UX.

I'm going to close this as an external/insufficient resources issue.

closed 2019-11-06 by OttoWinter