esphome/issues #822: 1.14.2 nodemcu doesn't connect with GPIO high

andriej, 2019-11-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
docker (non-hassio)

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 / nodemcu v2

**Affected component:**
n/a

**Description of problem:**
I have nodemcu v2 with connected two digital sensors to two GPIOs. Since 1.14.2 it can't boot while one of it's sensors is HIGH state.

Binary sensor with GPIO14 - when it's HIGH esp won't connect to MQTT/boot, second after input changes it starts like nothing happened.
Im 99% sure it wasn't like that in 1.13.6 which was my previous release.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  platform: ESP8266
  board: nodemcuv2
  devicename: klucznik
  upper_devicename: "Klucznik"
  static_ip: <redacted>
<<: !include common.yaml

logger:

switch:
  - platform: restart
    name: "Klucznik: restart"
    
sensor:   
  - platform: wifi_signal
    name: "Sygnał WiFi: Klucznik"
    update_interval: 60s

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    name: "Zamek: górny"
    device_class: lock
    filters:
    - delayed_on: 3000ms
    - delayed_off: 3000ms
    
  - platform: gpio
    pin: 
      number: GPIO12
      mode: INPUT_PULLUP
      inverted: True
    name: "Zamek: dolny"
    device_class: lock
    filters:
    - delayed_on: 3000ms
    - delayed_off: 3000ms

```

**Logs (if applicable):**
n/a

**Additional information and things you've tried:**

OttoWinter, 2019-11-04:
I highly doubt this has to do with GPIO12.

Please double check it really has to do with GPIO2 - and explain how you checked (what is different between the two).

Also, logs please.

Please also see https://github.com/esphome/issues/issues/773

andriej, 2019-11-04:
Simply - after OTA I see debug log, which is empty.
I was waiting for node to re-appear for few minutes.

After that when I come down where it's installed and change the lock state (as it's checking the lock-state only) to open, node connects immidiately after on.

I don't have any other oppoturnity to debug it other than logs from console right now.
Didn't have any problems with it earlier (meaning 1.13.6) and was just upgrading everything to 1.14.2 when problem appearead. Downgrade solved it.

OttoWinter, 2019-11-04:
Yeah but without logs there's no way to help here. Please try to access the device via USB and check what it's printing while the issue occurs.

andriej, 2019-11-17:
I'm stll not able to get any serial output on failed nodemcu - today I've reflashed some old tasmota on sonoff basic to newest 1.14.2 and it's not putting anything on serial. Reflashing it with the same code (without captive portal) but 1.13.6 made it working from first try.

closed 2020-08-07 by stale[bot]