esphome/issues #823: Xiaomi LYWSD02 does not show battery status

sermayoral, 2019-11-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):Hass.io**

Home Assistat 0.101.2
**ESP (ESP32/ESP8266, Board/Sonoff):Everything**

**Affected component:xiaomi_ble**

https://esphome.io/components/sensor/xiaomi_lywsd02.html

**Description of problem:**
Xiaomi LYWSD02 does not show battery status. There is a thought that this device does not send battery status, but it's not true. This information is hidden and it can be get tapping three times in MAC address of Mi Home app

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-11-04:
> This information is hidden and it can be seen tapping three times in MAC address of Mi Home app

That app uses a totally different communication method. ESPHome uses BLE ADV packets, the mi home app uses a full BLE connection.

1. Please check OpenMQTTGateway reports battery data - it also uses BLE adv packets. If it works there, you're right.

2. Attach debug logs with the log level set to `very_verbose`. Those will show the BLE ADV packet payload, and show if such data is sent.

sermayoral, 2019-11-05:
Hi @OttoWinter.

Here is a pastebin of a very_verbolse log: https://pastebin.com/fHKrWcpu

OttoWinter, 2019-11-06:
@sermayoral I looked through the logs, and can say that there's no battery level information to be found there from MAC address `3F:59:C8:63:38:99`.

The only datapoint types are `04` (temperature) and `06` (humidity).

Battery level is probably transmitted through some other mean with newer firmwares (or the packet is sent less often, maybe only when the value changes).

closed 2019-11-06 by OttoWinter