esphome/issues #825: Binary sensor randomly changes to unavailable

grzeg8102, 2019-11-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esphome 1.14.2 in docker. HA 101.1 also docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP DEVKIT V1
**Affected component:**

binary sensor https://esphome.io/components/binary_sensor/gpio.html

**Description of problem:**
Status of binary sensor randomly changes to unavailable for a short period. I'm not sure if it started with 1.14, since the periods where it becomes unavailable are very short and last for a minute or so. But I spotted this after upgrading to 1.14 as I started to experience similar issue with a switch becoming unavailable for hours (issue #824).
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: sensors_tracker
  platform: ESP32
  board: esp32doit-devkit-v1
          
wifi:
  ssid: 'REMOVED'
  password: 'removed'

ota:
  password: 'removed'

# Enable logging
logger:

esp32_ble_tracker:

sensor:
  - platform: xiaomi_hhccjcy01  #xiaomi_miflora
    mac_address: C4:7C:8D:65:FC:33
    temperature:
      name: "MiFlora Trawa Temperature"
    moisture:
      name: "MiFlora Trawa Moisture"
    illuminance:
      name: "MiFlora Trawa Illuminance"
    conductivity:
      name: "MiFlora Trawa Soil Conductivity"
    battery_level:
      name: "MiFlora Trawa Battery Level"

  - platform: xiaomi_hhccjcy01  #xiaomi_miflora
    mac_address: C4:7C:8D:65:FB:D5
    temperature:
      name: "MiFlora Tuje Temperature"
    moisture:
      name: "MiFlora Tuje Moisture"
    illuminance:
      name: "MiFlora Tuje Illuminance"
    conductivity:
      name: "MiFlora Tuje Soil Conductivity"
      
  - platform: xiaomi_lywsdcgq #xiaomi_mijia
    mac_address: 4C:65:A8:DB:DB:A6
    temperature:
      name: "Garaz Temperature"
      filters:
      - sliding_window_moving_average:
          window_size: 3
          send_every: 3
    humidity:
      name: "Garaz Humidity"
    battery_level:
      name: "Garaz Battery Level"

  - platform: xiaomi_lywsd02
    mac_address: E7:2E:01:21:1C:33
    temperature:
      name: "Xiaomi LYWSD02 Temperature"
    humidity:
      name: "Xiaomi LYWSD02 Humidity"
  
  - platform: adc
    pin: GPIO34
    name: "esp32_swiatlo"
    update_interval: 60s
    attenuation: 11db

  - platform: adc
    pin: GPIO35
    name: "esp32_gleba"
    filters:
      - sliding_window_moving_average:
          window_size: 12
          send_every: 12
      - lambda: return abs(3.90 - x);
    update_interval: 5s
    attenuation: 11db
    
  - platform: adc
    pin: GPIO32
    name: "esp32_deszcz"
    update_interval: 60s
    attenuation: 11db

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO23
      mode: INPUT_PULLDOWN
      inverted: True
    name: "Brama garazowa"
    device_class: garage_door
    filters:
      - delayed_on: 50ms
      - delayed_off: 50ms
        
api:  
  
time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 0
        hours: 12
        days_of_week: MON-SUN
        then:
          - switch.toggle: restart_sw

switch:
  - platform: restart
    name: "ESP32 garaz restart"
    id: restart_sw
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

zvldz, 2019-11-06:
Did you check uptime?
The device may be rebooting.

```
sensor:
  - platform: uptime
    name: ESP32 uptime
```

grzeg8102, 2019-11-06:
> Did you check uptime?
> The device may be rebooting.
> 
> ```
> sensor:
>   - platform: uptime
>     name: ESP32 uptime
> ```

No, haven't heard of it earlier. First I'll try to disable BLE tracker to see if it solves the problem as recommended by Otto Winter in another thread and if it doesn't - I'll play with this one. Thanks for the info

OttoWinter, 2019-11-06:
I'm not sure what the base issue is here.

Do you mean that the status sensor showing "unavailable" is the issue? From the picture it's hard to see if it's "unavailable" or "on".

Status binary sensor should only show on/off, unavailable would be an invalid state. But I think it's showing on, in which case this would just be a symptom for the device not connecting well, and a duplicate of https://github.com/esphome/issues/issues/824

grzeg8102, 2019-11-06:
> I'm not sure what the base issue is here.
> 
> Do you mean that the status sensor showing "unavailable" is the issue? From the picture it's hard to see if it's "unavailable" or "on".
Yes, what I meant is that the sensor randomly changes status to unavailable. This lasts for a couple of seconds and then it gets back to on/off depending what the door/window sensor's state.
I marked these thin lines with these ugly red arrows (sorry) to illustrate this better.

This may be again related to the BLE tracker issue. In the morning I removed the BLE tracker from the software and during 5 hours the sensor wasn't unavailable even a single time. So let me observe in a longer time frame and come back with observations.

grzeg8102, 2019-11-06:
And I can already say this has nothing to do with BLE tracker. This afternoon the sensor changed to unavailable twice already. It lasts a few seconds and gets back to normal. I also checked status history directly on a database and saw that on 13.6 it also behaved this way. So this isn't new.

@OttoWinter  - is it normal that esphome entities randomly change their state to unavailable for a couple of seconds during the day? My WiFi signal range is great as the ESP device is in the next room and in logs the signal strength is shown as maximum

OttoWinter, 2019-11-07:
@grzeg8102 
> is it normal that esphome entities randomly change their state to unavailable for a couple of seconds during the day?
Depends on a couple of factors - wifi coverage and what type of module it is. Couple of unavailables can happen if you don't have the best coverage. Especially the latter part (ESP module itself) can be a big influence, I've got ESPs running near-identical software next to each other with way different network availability.

grzeg8102, 2019-11-11:
ok, so this has probably something to do with my ESP32 as the WiFi coverage in place where the ESP32 is located is very good, almost maximal. 
Having the binary switch shown in HA as unavailable 3 or 4 times a day isn't a big deal for me, but if you manage to workaround it somehow then it'd be great.

OttoWinter, 2019-11-12:
3/4 times a day is not good but I would say it's possible - with that kind of disconnection rate it's also hard to pinpoint what the issue really is and depends on a lot of factors.

Unless it's much more often than that I will close this.

closed 2019-11-12 by OttoWinter