esphome/issues #826: No sensor updates after 49 days uptime

jvanderneutstulen, 2019-11-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip with Home Assistant

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**Affected component:**
Multiple, observed the issue with dallas and htu21d.

**Description of problem:**
After 49 days esphome does not send sensors updates to homeassistant. The API works, HA is still connected to the sensor. Also I can reboot the sensor with the restart switch. After reboot the sensor
functions normally.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: lux3
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "SSID"
  password: "secret"

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: "secret"

ota:
  password: "secret"

dallas:
  - pin: GPIO15
    update_interval: 3s

sensor:
  - platform: dallas
    address: 0xE30417C47758FF28
    name: "Temperature Zone 2"
    accuracy_decimals: 2
```

**Logs (if applicable):**

I don't see any logging from the `update` function.

**Additional information and things you've tried:**
49 days uptime smells like uint32 overflow. 

`uint32_t Scheduler::millis_()` in `core/scheduler.cpp` is suspicious as it
checks `last_millis_`, but this variable is never updated, so I doubt the
arithmetic with  `millis_major_` is working as intended.

OttoWinter, 2019-11-06:
Please note what version the device was running while the issue happened.

The code you mentioned has only been there since 1.14 - in 1.13 the scheduled functions were still per-component and the code was very different.

I've tested the new scheduler against weird conditions (rolling over millis) etc rigorously, so I don't think the new code has major issues anymore.

jvanderneutstulen, 2019-11-06:
The dallas sensor is/was running on 1.14.0-dev (commit `bd0be41064937399b31909faace3923880e8ada5`)

jvanderneutstulen, 2019-11-06:
When I create a test config with an uptime sensor with the proposed change, and add a large offset (2^32 - 100000) to millis(), the scheduler keeps working. But I get an extra update() run when the counter wraps. Probably not a huge deal for most components, but maybe some component does not like this.

```
[14:16:23][VV][scheduler:124]: Running interval 'update' with interval=10000 last_execution=4294945214 (now=4294955221)
[14:16:23][V][sensor:013]: 'Uptime Sensor': Received new state 87.925003
[14:16:23][D][sensor:092]: 'Uptime Sensor': Sending state 87.92500 s with 0 decimals of accuracy
[14:16:33][VV][scheduler:124]: Running interval 'update' with interval=10000 last_execution=4294955214 (now=4294965214)
[14:16:33][V][sensor:013]: 'Uptime Sensor': Received new state 97.917999
[14:16:33][D][sensor:092]: 'Uptime Sensor': Sending state 97.91800 s with 0 decimals of accuracy
[14:16:35][D][scheduler:206]: Incrementing scheduler major
[14:16:35][VV][scheduler:124]: Running interval 'update' with interval=10000 last_execution=4294965214 (now=100006)
[14:16:35][V][sensor:013]: 'Uptime Sensor': Received new state 100.005997
[14:16:35][D][sensor:092]: 'Uptime Sensor': Sending state 100.00600 s with 0 decimals of accuracy
[14:16:43][VV][scheduler:124]: Running interval 'update' with interval=10000 last_execution=97918 (now=107918)
[14:16:43][V][sensor:013]: 'Uptime Sensor': Received new state 107.917999
[14:16:43][D][sensor:092]: 'Uptime Sensor': Sending state 107.91800 s with 0 decimals of accuracy
[14:16:53][VV][scheduler:124]: Running interval 'update' with interval=10000 last_execution=107918 (now=117924)
[14:16:53][V][sensor:013]: 'Uptime Sensor': Received new state 117.924004
[14:16:53][D][sensor:092]: 'Uptime Sensor': Sending state 117.92400 s with 0 decimals of accuracy
```

OttoWinter, 2019-11-06:
Hmm, I can't replicate that second issue.

I tried with the following config:

```yaml
sensor:
  - platform: uptime
    name: Uptime
    update_interval: 10s
```

```
[15:13:54][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=4294927276 (now=4294937546)
...
[15:14:04][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=4294937276 (now=4294947276)
...
[15:14:14][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=4294947276 (now=4294957276)
...
[15:14:24][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=4294957276 (now=4294967276)
...
[15:14:34][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=4294967276 (now=9980)
...
[15:14:44][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=9980 (now=19983)
```

And with this patch: (subtraction is same as addition by UINT_MAX-value)
```cpp
const uint32_t now = millis() - 30000U;
```

jvanderneutstulen, 2019-11-06:
I didn't simulate the wrap correctly. It works fine now.

closed 2019-11-06 by OttoWinter