esphome/issues #827: pzemac uart error (Invalid stop bits: Integration requested stop_bits 2 but you have 1!)

rradar, 2019-11-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- pip3

**ESP (ESP32/ESP8266, Board/Sonoff):**

- esp8285 and pzem004t v3

**Affected component:**

- pzemac

**Description of problem:**

Everything works like expected but the log prints a error message regarding stop bits.

**Problem-relevant YAML-configuration entries:**

```yaml
uart:
  tx_pin: 14
  rx_pin: 13
  baud_rate: 9600
  stop_bits: 2

sensor:
  - platform: pzemac
#    energy:
#      name: "${upper_devicename} Energy"
    frequency:
      name: "${upper_devicename} Frequency"
      unit_of_measurement: hz
    voltage:
      name: "${upper_devicename} Voltage"
    current:
      name: "${upper_devicename} Current"
    power_factor:
      name: "${upper_devicename} Powerfactor"
    power:
      name: "${upper_devicename} Power"
      id: pzem_power
    update_interval: 5s
```

**Logs (if applicable):**

```yaml
[E][uart:388]:   Invalid stop bits: Integration requested stop_bits 2 but you have 1!
```

**Additional information and things you've tried:**

I also tested with
```yaml
uart:
  tx_pin: 14
  rx_pin: 13
  baud_rate: 9600
#  stop_bits: 2
```
and
```yaml
uart:
  tx_pin: 14
  rx_pin: 13
  baud_rate: 9600
  stop_bits: 1
```

but the error keeps showing up

OttoWinter, 2019-11-07:
Can't reproduce, I copied the exact config you had above (added upper_devicename substitution too) and don't see that message in the logs.

Do you have multiple UART buses defined in your config? Can you also make sure the firmware you compiled is running on the device? Thanks!

rradar, 2019-11-09:
I double checked and have the exact config running.

The only other setting I have regarding uart is that the logger is disabled (actually their is no reason because I use software serial and not the hardware one...):

```
logger:
  baud_rate: 0
```

OttoWinter, 2019-11-09:
Ok, then I don't know what it could be - tried with baud_rate: 0 too here but couldn't replicate.

Could you try opening up the `<NODE_NAME>/main.cpp` file and copying all the lines related to `uart` (a Ctrl+F/grep for uart is probably enough)

rradar, 2019-11-11:
All uart related lines I could find in the main.cpp

```
using namespace uart;
uart::UARTComponent *uart_uartcomponent;
```

```
 // logger:
  //   baud_rate: 0
  //   logs: {}
  //   esp8266_store_log_strings_in_flash: true
  //   tx_buffer_size: 512
  //   id: logger_logger
  //   level: DEBUG
  //   hardware_uart: UART0
  logger_logger = new logger::Logger(0, 512, logger::UART_SELECTION_UART0);
  logger_logger->pre_setup();
  App.register_component(logger_logger);
```

```
  // uart:
  //   tx_pin: 14
  //   rx_pin: 13
  //   baud_rate: 9600
  //   stop_bits: 1
  //   id: uart_uartcomponent
  uart_uartcomponent = new uart::UARTComponent();
  App.register_component(uart_uartcomponent);
```

```
  // modbus:
  //   uart_id: uart_uartcomponent
  //   id: modbus_modbus
  modbus_modbus = new modbus::Modbus();
  App.register_component(modbus_modbus);
  App.register_component(my_time);
  uart_uartcomponent->set_baud_rate(9600);
  uart_uartcomponent->set_tx_pin(14);
  uart_uartcomponent->set_rx_pin(13);
  uart_uartcomponent->set_stop_bits(1);
```

```
 pzemac_pzemac->set_parent(modbus_modbus);
  pzemac_pzemac->set_address(0x01);
  modbus_modbus->register_device(pzemac_pzemac);
  wifi_info_ssidwifiinfo = new wifi_info::SSIDWiFiInfo();
  App.register_component(wifi_info_ssidwifiinfo);
  modbus_modbus->set_uart_parent(uart_uartcomponent);
```

OttoWinter, 2019-11-12:
The issue is here:

```cpp
uart_uartcomponent->set_stop_bits(1);
```

But I don't understand how that can happen if you have `stop_bits: 2` in your config.

If I type in `stop_bits: 2` I get the following generated cpp:

```cpp
uart_uartcomponent->set_stop_bits(2);
```

I hate to say it but it looks an awful lot like user error to me - please make absolutely sure the config you're compiling contains exactly these lines:

```yaml
uart:
  tx_pin: 14
  rx_pin: 13
  baud_rate: 9600
  stop_bits: 2
```

rradar, 2019-11-17:
I created a new config from the scratch and the **error isn't printed anymore** when I set:

```yaml
uart:
  tx_pin: 14
  rx_pin: 13
  baud_rate: 9600
  stop_bits: 2
```
but downside from this configuration is that no readings from the pzem004t v3 happen anymore. All pzem related sensors have a `unavailable`.

Log:

```yaml
[17:11:17][I][app:100]: ESPHome version 1.14.2 compiled on Nov 17 2019, 17:09:49
[17:11:17][C][wifi:409]: WiFi:
[17:11:17][C][wifi:277]:   SSID: '...'
[17:11:17][C][wifi:278]:   IP Address: ....
[17:11:17][C][wifi:280]:   BSSID: ...
[17:11:17][C][wifi:281]:   Hostname: 'mains'
[17:11:17][C][wifi:285]:   Signal strength: -14 dB ▂▄▆█
[17:11:17][V][wifi:287]:   Priority: 0.0
[17:11:17][C][wifi:289]:   Channel: ...
[17:11:17][C][wifi:290]:   Subnet: ...
[17:11:17][C][wifi:291]:   Gateway: ...
[17:11:17][C][wifi:292]:   DNS1: ...
[17:11:17][C][wifi:293]:   DNS2: (IP unset)
[17:11:17][V][sensor:013]: 'Mains Total Daily Energy': Received new state 0.000000
[17:11:17][D][sensor:092]: 'Mains Total Daily Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[17:11:17][C][uart:145]: UART Bus:
[17:11:17][C][uart:147]:   TX Pin: GPIO14
[17:11:17][C][uart:150]:   RX Pin: GPIO13
[17:11:17][C][uart:152]:   Baud Rate: 9600 baud
[17:11:17][C][uart:153]:   Stop bits: 2
[17:11:17][C][uart:157]:   Using software serial
[17:11:17][C][modbus:096]: Modbus:
[17:11:17][C][uptime.sensor:030]: Uptime Sensor 'Mains Uptime'
[17:11:17][C][uptime.sensor:030]:   Unit of Measurement: 's'
[17:11:17][C][uptime.sensor:030]:   Accuracy Decimals: 0
[17:11:17][C][uptime.sensor:030]:   Icon: 'mdi:timer'
[17:11:17][V][uptime.sensor:030]:   Unique ID: '60019485dd1e-uptime'
[17:11:17][C][template.sensor:021]: Template Sensor 'Mains Uptime Days'
[17:11:17][C][template.sensor:021]:   Unit of Measurement: 'days'
[17:11:17][C][template.sensor:021]:   Accuracy Decimals: 1
[17:11:17][C][template.sensor:021]:   Icon: 'mdi:clock-start'
[17:11:17][C][template.sensor:022]:   Update Interval: 60.0s
[17:11:17][C][template.text_sensor:020]: Template Sensor 'Mains Hardware'
[17:11:17][C][template.text_sensor:020]:   Icon: 'mdi:tag'
[17:11:17][C][logger:175]: Logger:
[17:11:17][C][logger:176]:   Level: VERBOSE
[17:11:17][C][logger:177]:   Log Baud Rate: 0
[17:11:17][C][logger:178]:   Hardware UART: UART0
[17:11:17][C][status:034]: Status Binary Sensor 'Mains Status'
[17:11:17][C][status:034]:   Device Class: 'connectivity'
[17:11:17][C][pzemac:061]: PZEMAC:
[17:11:17][C][pzemac:062]:   Address: 0x01
[17:11:17][C][pzemac:063]: Voltage 'Mains Voltage'
[17:11:17][C][pzemac:063]:   Unit of Measurement: 'V'
[17:11:17][C][pzemac:063]:   Accuracy Decimals: 1
[17:11:17][C][pzemac:063]:   Icon: 'mdi:flash'
[17:11:17][C][pzemac:064]: Current 'Mains Current'
[17:11:17][C][pzemac:064]:   Unit of Measurement: 'A'
[17:11:17][C][pzemac:064]:   Accuracy Decimals: 3
[17:11:17][C][pzemac:064]:   Icon: 'mdi:current-ac'
[17:11:17][C][pzemac:065]: Power 'Mains Power'
[17:11:17][C][pzemac:065]:   Unit of Measurement: 'W'
[17:11:17][C][pzemac:065]:   Accuracy Decimals: 1
[17:11:17][C][pzemac:065]:   Icon: 'mdi:power'
[17:11:17][C][pzemac:066]: Frequency 'Mains Frequency'
[17:11:17][C][pzemac:066]:   Unit of Measurement: 'hz'
[... 161 lines cut ...]
[17:16:08][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[17:16:08][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[17:16:08][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[17:16:08][V][wifi_signal.sensor:009]:   Unique ID: '60019485dd1e-wifisignal'
[17:16:08][C][wifi_info:009]: WifiInfo IPAddress 'Mains IP'
[17:16:08][C][wifi_info:009]:   Icon: 'mdi:ip-network'
[17:16:08][V][wifi_info:009]:   Unique ID: '60019485dd1e-wifiinfo-ip'
[17:16:08][C][wifi_info:010]: WifiInfo SSID 'Mains SSID'
[17:16:08][C][wifi_info:010]:   Icon: 'mdi:access-point-network'
[17:16:08][V][wifi_info:010]:   Unique ID: '60019485dd1e-wifiinfo-ssid'
[17:16:08][C][wifi_info:011]: WifiInfo BSSID 'Mains BSSID'
[17:16:08][C][wifi_info:011]:   Icon: 'mdi:access-point-network'
[17:16:08][V][wifi_info:011]:   Unique ID: '60019485dd1e-wifiinfo-bssid'
[17:16:12][D][pzemac:046]: PZEM AC: V=235.2 V, I=3.093 A, P=577.7 W, F=49.9 Hz, PF=0.79
[17:16:12][V][sensor:013]: 'Mains Voltage': Received new state 235.199997
[17:16:12][D][sensor:092]: 'Mains Voltage': Sending state 235.20000 V with 1 decimals of accuracy
[17:16:12][V][sensor:013]: 'Mains Current': Received new state 3.093000
[17:16:12][D][sensor:092]: 'Mains Current': Sending state 3.09300 A with 3 decimals of accuracy
[17:16:12][V][sensor:013]: 'Mains Power': Received new state 577.700012
[17:16:12][D][sensor:092]: 'Mains Power': Sending state 577.70001 W with 1 decimals of accuracy
[17:16:12][V][sensor:013]: 'Mains Total Daily Energy': Received new state 0.805089
[17:16:12][D][sensor:092]: 'Mains Total Daily Energy': Sending state 0.00081 kWh with 3 decimals of accuracy
[17:16:12][V][sensor:013]: 'Mains Energy Meter': Received new state 3.382913
[17:16:12][D][sensor:092]: 'Mains Energy Meter': Sending state 2349.00342 kWh with 3 decimals of accuracy
```

MariusLaurinaitis, 2019-11-18:
Hi, do I need a PZEM soldering resistor (Pzem 004t V1.0)? And what do you use ESP8266 or ESP32? I try everything with the Wemos d1 mini, nothing can be seen.

brandond, 2019-11-18:
@MariusLaurinaitis please don't use this issue for unrelated Q/A. Start a new issue or ask on the HA forums.

sendorm, 2019-11-19:
I am using the integration with `stop_bits: 1`. Won't work if I set to 2. 
The integration gives the error `Invalid stop bits: Integration requested stop_bits 2 but you have 1!` as described. I propose a change to the integration's default values. Which should be 1. 
My pzems are V3 with esphome 1.4.3.

MariusLaurinaitis, 2019-11-19:
> @MariusLaurinaitis please don't use this issue for unrelated Q/A. Start a new issue or ask on the HA forums.

Thank you, my fault

kcinIvan, 2019-12-11:
I lost two nights before I found this solution!
#  stop_bits: 2

CarlosGS, 2020-08-08:
> I lost two nights before I found this solution!
> `# stop_bits: 2`

THANK YOU!! :sob:

CarlosGS, 2020-12-06:
I'm going to submit a documentation patch removing that line from the
example

closed 2020-12-21 by jesserockz