esphome/issues #828: Template Binary Sensor Unavailable after boot

pinkywafer, 2019-11-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Proxmox VM on Intel i7

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff RF Bridge (ESP8266)

**Affected component:**

https://esphome.io/components/binary_sensor/homeassistant.html

**Description of problem:**
Template binary sensors no longer return a state at startup

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sonoff_rf_bridge
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_psk
  manual_ip:
    static_ip: REDACTED
    gateway: REDACTED
    subnet: REDACTED

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: remote_receiver
    name: Motion
    device_class: motion
    rc_switch_raw:
      code: 'REDACTED'
      protocol: 6
    filters:
      delayed_off: 5s
  
  - platform: remote_receiver
    id: door_open
    internal: true
    rc_switch_raw:
      code: 'REDACTED'
      protocol: 6
    filters:
      delayed_off: 100ms
      
  - platform: remote_receiver
    id: door_closed
    internal: true
    rc_switch_raw:
      code: 'REDACTED'
      protocol: 6
    filters:
      delayed_off: 100ms
      
  - platform: template
    name: Door
    device_class: door
    lambda: |-
      if (id(door_open).state) {
        return true;
      } else if (id(door_closed).state) {
        return false;
      } else {
        return {};
      }

remote_receiver:
  pin: 
    number: 4
    mode: INPUT_PULLUP
    inverted: True
  dump: rc_switch
  tolerance: 70
  filter: 3us
  idle: 3ms
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
When the device first connects, the `Motion` sensor returns either on or off,
BUT
the `door` template sensor returns *unavailable* until either the `door_open` or `door_closed` sensors has been triggered. (`door_open` and `door_closed` are on only for a brief period when opening or closing, and do not remain 'on')

On a previous version of ESP home, the last known state was restored in Home Assistant, and if no state was stored, Home assistant would show false (ie. closed).  I appreciate that the state is not known by the device, but the last known Home Assistant state being restored was very helpful for this case.

I believe this was caused by https://github.com/home-assistant/home-assistant/pull/27638#  see #426 

While current behaviour is ideal for most binary sensors, in the case of the door / windows sensors here, the initial assumed state being reported in Home Assistant is very helpful.

* (ideally) Is there any way to pull the last known value from home assistant at boot
* or is it possible to restore previous operation for template binary sensors?
* or is it possible to have an option to allow Home Assistant fall back to previous state if no value is returned?
* or failing that, can a template sensor have an `initial_value` option added, so that the initial value can be set in the setup of the sensor? currently, the only way to set the initial state seems to be using `esphome.on_boot` and then publishing a chosen initial state.

randybb, 2019-11-08:
I am having the same problem with remote (platform: homeassistant) binary sensors. Based on log esphome receives the status information from HA, but the sensor is updated only after state changes.

deslizador27, 2019-11-11:
Los nuevos controles rf no funciona

xppx99, 2019-11-11:
Hello, I have the exact same problem. 
I use the sonoff RF to receive a lot of signals (doorbell, mailbox, doors, windows, movement sensors) and in this case the problem is minimized since once for example the doorbell rings and the sensor is activated, in HA the sensor changes from unavailable to ON (and then OFF after some time). 

The main problem is that I also use it to control 5 PowerPlugs (RF), and in this case I loose the ability to control them, since the switch entities became unavailable... 

What I see is that a lot of entities became "unavailable" randomly.
I've tried to reboot the sonoff RF but nothing changes. The thinks that temporary fix the problem is removing the Sonoff RF ESPhome integration, and add it again. 
Sometimes a specific sensor is unavaible, and after some time it fixes itself. 

This was taken now: 
[image]

All grey entities are unavailable. This is a problem for me.

closed 2019-11-12 by OttoWinter