esphome/issues #829: State change delays

brandond, 2019-11-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip in venv

**ESP (ESP32/ESP8266, Board/Sonoff):**
esp8266 / sonoff T1 3-gang clone

**Affected component:**
on_click + light.toggle

**Description of problem:**
I'm using a GPIO binary input with an on_click trigger to toggle the state of a binary light. The input functions as a momentary switch. One out of every four or five taps on the switch will not cause the binary light to toggle until 4-5 seconds later. I see the input go ON or OFF immediately when I press or release the button, but the light toggle just doesn't happen until much later than it should.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: gpio
    id: button0
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_click:
      then:
        light.toggle: relay0

output:
  - platform: gpio
    id: relay0_output
    pin: GPIO12

light:
  - platform: binary
    name: "Touch ${id} Light 1"
    id: relay0
    output: relay0_output
```

**Logs (if applicable):**
```
[14:49:15][D][binary_sensor:036]: 'button0': Sending state ON
[14:49:15][D][binary_sensor:036]: 'button0': Sending state OFF
--- DELAY HERE ---
[14:49:18][D][light:264]: 'Touch 00 Light 1' Setting:
[14:49:18][D][light:273]:   State: ON

[14:49:18][D][binary_sensor:036]: 'button0': Sending state ON
[14:49:18][D][binary_sensor:036]: 'button0': Sending state OFF
[14:49:18][D][light:264]: 'Touch 00 Light 1' Setting:
[14:49:18][D][light:273]:   State: OFF

[14:49:19][D][binary_sensor:036]: 'button0': Sending state ON
[14:49:19][D][binary_sensor:036]: 'button0': Sending state OFF
[14:49:19][D][light:264]: 'Touch 00 Light 1' Setting:
[14:49:19][D][light:273]:   State: ON

[14:49:20][D][binary_sensor:036]: 'button0': Sending state ON
[14:49:21][D][binary_sensor:036]: 'button0': Sending state OFF
[14:49:21][D][light:264]: 'Touch 00 Light 1' Setting:
[14:49:21][D][light:273]:   State: OFF

[14:49:22][D][binary_sensor:036]: 'button0': Sending state ON
[14:49:22][D][binary_sensor:036]: 'button0': Sending state OFF
--- DELAY HERE ---
[14:49:27][D][light:264]: 'Touch 00 Light 1' Setting:
[14:49:27][D][light:273]:   State: ON

[14:49:27][D][binary_sensor:036]: 'button0': Sending state ON
[14:49:27][D][binary_sensor:036]: 'button0': Sending state OFF
[14:49:27][D][light:264]: 'Touch 00 Light 1' Setting:
[14:49:27][D][light:273]:   State: OFF

[14:49:28][D][binary_sensor:036]: 'button0': Sending state ON
[14:49:28][D][binary_sensor:036]: 'button0': Sending state OFF
--- DELAY HERE ---
[14:49:32][D][light:264]: 'Touch 00 Light 1' Setting:
[14:49:32][D][light:273]:   State: ON

[14:49:32][D][binary_sensor:036]: 'button0': Sending state ON
[14:49:32][D][binary_sensor:036]: 'button0': Sending state OFF
[14:49:32][D][light:264]: 'Touch 00 Light 1' Setting:
[14:49:32][D][light:273]:   State: OFF
```

brandond, 2019-11-07:
I should note that I only noticed this because I'm not actually using the output on the device to switch things; I'm just using the light as a trigger for automations in HA. The local light seems to always switch immediately - I can hear the relay click - but when there's a delay, neither the logs nor the light show the change until a few seconds later.

So, this is probably a wifi or mqtt issue? It's just odd since I've never had any sort of delay like this with any other device.

brandond, 2019-11-07:
Did some packet captures and this does appear to be some sort of network issue. wireshark shows frequent spurious retransmits, which would usually indicate that the acks are getting dropped. The device is about 8 feet and one wall from the AP, and shows an RSSI of -53 dB, so it shouldn't be due to poor coverage.

```
    1 0.000000000   10.0.1.197 → 10.0.1.27    MQTT 133 Publish Message [touch00/debug]
    2 0.053456714    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=80 Win=63784 Len=0
    3 0.335716558   10.0.1.197 → 10.0.1.27    MQTT 134 Publish Message [touch00/debug]
    4 0.335742450    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=160 Win=63784 Len=0
    5 0.348041964   10.0.1.197 → 10.0.1.27    MQTT 180 Publish Message [touch00/debug], Publish Message [touch00/debug]
    6 0.348067495    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=286 Win=63784 Len=0
    7 0.358216284   10.0.1.197 → 10.0.1.27    MQTT 108 Publish Message [touch00/light/touch_00_light_1/state]
    8 0.358242730    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=340 Win=63784 Len=0
    9 2.207194405   10.0.1.197 → 10.0.1.27    MQTT 133 Publish Message [touch00/debug]
   10 2.207236442    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=419 Win=63784 Len=0
   11 2.540025893   10.0.1.197 → 10.0.1.27    MQTT 134 Publish Message [touch00/debug]
   12 2.540050617    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=499 Win=63784 Len=0
   13 3.788308299   10.0.1.197 → 10.0.1.27    MQTT 134 [TCP Spurious Retransmission] , Publish Message [touch00/debug]
   14 3.788338944    10.0.1.27 → 10.0.1.197   TCP 66 [TCP Dup ACK 12#1] 1883 → 56758 [ACK] Seq=1 Ack=499 Win=63784 Len=0 SLE=419 SRE=499
   15 3.795003044   10.0.1.197 → 10.0.1.27    MQTT 236 Publish Message [touch00/debug], Publish Message [touch00/debug], Publish Message [touch00/light/touch_00_light_1/state]
   16 3.795026849    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=681 Win=63784 Len=0
   17 4.449286503   10.0.1.197 → 10.0.1.27    MQTT 133 Publish Message [touch00/debug]
   18 4.449316807    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=760 Win=63784 Len=0
   19 4.768020587   10.0.1.197 → 10.0.1.27    MQTT 134 Publish Message [touch00/debug]
   20 4.768055422    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=840 Win=63784 Len=0
   21 5.811554865   10.0.1.197 → 10.0.1.27    MQTT 134 [TCP Spurious Retransmission] , Publish Message [touch00/debug]
   22 5.811598567    10.0.1.27 → 10.0.1.197   TCP 66 [TCP Dup ACK 20#1] 1883 → 56758 [ACK] Seq=1 Ack=840 Win=63784 Len=0 SLE=760 SRE=840
   23 5.822271029   10.0.1.197 → 10.0.1.27    MQTT 234 Publish Message [touch00/debug], Publish Message [touch00/debug], Publish Message [touch00/light/touch_00_light_1/state]
   24 5.822295911    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=1020 Win=63784 Len=0
   25 8.557263398   10.0.1.197 → 10.0.1.27    MQTT 133 Publish Message [touch00/debug]
   26 8.557287964    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=1099 Win=63784 Len=0
   27 8.834934913   10.0.1.197 → 10.0.1.27    MQTT 60 Ping Request
   28 8.834961186    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=1 Ack=1101 Win=63784 Len=0
   29 8.835232802    10.0.1.27 → 10.0.1.197   MQTT 56 Ping Response
   30 8.891941717   10.0.1.197 → 10.0.1.27    MQTT 134 Publish Message [touch00/debug]
   31 8.891979584    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=3 Ack=1181 Win=63784 Len=0
   32 8.903079238   10.0.1.197 → 10.0.1.27    MQTT 181 Publish Message [touch00/debug], Publish Message [touch00/debug]
   33 8.903103659    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=3 Ack=1308 Win=63784 Len=0
   34 10.360390216   10.0.1.197 → 10.0.1.27    MQTT 181 [TCP Spurious Retransmission] , Publish Message [touch00/debug], Publish Message [touch00/debug]
   35 10.360417040    10.0.1.27 → 10.0.1.197   TCP 66 [TCP Dup ACK 33#1] 1883 → 56758 [ACK] Seq=3 Ack=1308 Win=63784 Len=0 SLE=1181 SRE=1308
   36 10.368345767   10.0.1.197 → 10.0.1.27    MQTT 109 Publish Message [touch00/light/touch_00_light_1/state]
   37 10.368380592    10.0.1.27 → 10.0.1.197   TCP 54 1883 → 56758 [ACK] Seq=3 Ack=1363 Win=63784 Len=0
```

OttoWinter, 2019-11-07:
That is interesting. My guess is it's actually on the broker side (or TCP) if it works instantly locally.

Why? The "turning loght on/off" log message is created before the output is switched. So the delay must be in mqtt/wifi as you noted.

Next, mqtt log messages are not buffered. If it can't be sent at the moment, the message gets thrown away. - Any delay there must thus be in TCP/IP or broker side.

I don't see a good way of debugging this. If you can, I would recommend trying one of two things:
- Capture TCP fragments with Wireshark - especially the timing there would be interesting.
- Create a quick mqtt broker on your dev machine and make the ESP connect to that. If it works better there it's probably the mqtt broker.

OttoWinter, 2019-11-07:
Thanks for the packet logs! Yeah looks like an issue in TCP/IP. Could you try `arduino_version: 2.4.2`?

On a sidenote, I wonder what TCP retransmission strategy lwip is using for unacked packets. I've had some problems with poor default strategies on some other wifi-based systems for streaming a webcam feed.

brandond, 2019-11-08:
Seems to be better under 2.4.2. In a quick test I saw 1 retransmit in 100 packets, as opposed one in 10 or so with 2.5.2. I'll let it run for a while and see if it gets better.

Turning down logging to INFO seems to have helped as well, but that might just be because it's got less data to send without logging all the state changes.

closed 2020-12-05 by stale[bot]