esphome/issues #830: Unable to flash Sonoff 4CH

drthanwho, 2019-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 0.100.1
**ESP (ESP32/ESP8266, Board/Sonoff):**
 
Sonoff 4CH 
**Affected component:**

**Description of problem:**
Unable to flash a sonoff 4CH with esphome. Connected everything as per the documentation but cannot communicate with the esp chip. Seems like it isn't in programming mode but unsure how to proceed

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
Log from Esphome-flasher:
“Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: Timed out waiting for packet header”
```

**Additional information and things you've tried:**

-tried powering sonoff off serial adapter and nodemcu pins  in case of not enough power (common ground between adapter and nodemcu)
-tried swapping TX/RX pins
-not holding down GPIO0 does boot board on default firmware and connects to wifi so power should be fine from serial adapter as well
- soldered cable from GPIO0 to GND

OttoWinter, 2019-11-08:
Duplicate of https://github.com/esphome/issues/issues/814

closed 2019-11-08 by OttoWinter

drthanwho, 2019-11-08:
Tried following command on esptool (twice for crossing/non-crossing TX/RX): 

`C:\Python>esptool.py --before default_reset --after hard_reset --baud 115200 --c
hip esp8266 write_flash 0x0 sonoff_4ch.bin`

But still getting same error as before :(

OttoWinter, 2019-11-09:
Ok, then it is not a duplicate of #814.

However, if it still shows up when you're using esptool.py directly, that means there's something wrong with a) your setup or b) esptool - and not a software bug in ESPHome.

As for the issue, I think you went through all the usual troubleshooting steps, only thing I would recommend to check is what COM port the ESP is connected to. Check by going into the device manager windows panel.

drthanwho, 2019-11-09:
I don't think it's #814 either as I saw people having that issue were still able to see on esptool output that chip was recognised and got a MAC address. I can't even get that.

I'll have to try a different serial adapter maybe. Although I've successfully flashed NODEMCUs, ESP01s and Sonoff Basics with it so far. 

And also used esptool previous to backup the firmware from an ESP32 cam module and flash ESPHOME into it. 

Keep thinking the board might be faulty but then doesn't explain it working normally with default firmware.

COM port is definitely correct. Have already checked on device manager. Thanks for suggesting it.

I've literally run out of ideas here.

I did read about people that flashed it while connected to mains to power the chip and then had RX TX connected to adapter and computer, however I'm not sure whether I should really try that approach?

OttoWinter, 2019-11-12:
> I did read about people that flashed it while connected to mains to power the chip and then had RX TX connected to adapter and computer, however I'm not sure whether I should really try that approach?

I don't recommend that. However one problem could be that the device is not getting enough power, I usually use a bench power supply set at 3.3V for this.

drthanwho, 2019-11-12:
> I don't recommend that. However one problem could be that the device is not getting enough power, I usually use a bench power supply set at 3.3V for this.

I did try off a nodemcu 3.3 pin. Is that not enough power if power was the issue here? Should I find a dedicated 3.3V power supply to try instead?